oliexdev/openScale #130: Disable measurements in the overview

woj-tek, 2018-01-05:
Currently all measurements types are enabled which impose the scale (so bone mass in [kg] would be around dozens and then weight in 60-90 [kg]) which results in 5 almost straight lines.
In chart view we can enable/disable particular measurements types which gives better information.

oliexdev, 2018-01-05:
You can completely disable a measurement in the settings->mesurements.

closed 2018-01-05 by oliexdev

woj-tek, 2018-01-05:
Hmm, but I do want to track those measurements and disabling them there removes them completely from tracking. This is only about disabling showing particular measurements in overview graph (they can still be listed in text form below).

oliexdev, 2018-01-05:
Ok I understand but how will you do it? Add more buttons like in the graph view? Add more options in the settings and to overload it?

woj-tek, 2018-01-05:
Hmm, option in preference could clutter the preferences, but in respect to #59 there could be sub-items in the preferences for each measurement type:
```
Muscle mass:
   -> use [kg] or [%]
   -> display in overview tab
```

Alternatively, as the tap on measurement in overview list is already assigned to the assesment view (small scale with the information about the norm) there could be a long tap to show/hide in the graph, but that would require more discoverability (maybe a mark that it's shown in the grap in the form of a small line similar to the one in the graph?)

oliexdev, 2018-01-05:
Yeah they are some ideas like in #59 but it needs to be also practicable to programme it ;)
That's currently the point I could hack something but I would brake my idea. The idea is If we had one spot in the source code for the measurements, the other views or fragments should be dynamically changed.

woj-tek, 2018-01-08:
But the above idea wouldn't break the idea. Haven't perused the code so I'm not sure what is the base (kg or %) but assuming you are using kg for weight and % for other metrics then it should be possible to still provide two fields (or configuration what metric type is entered) and either dynamically calculate the value while storing it or dynamically calculate it while retrieving. (I'm aware that this would all more code, and keeping DRY in mind and wanting to have it as general as possible this could be tricky :-) )

woj-tek, 2018-03-11:
Could we re-open the issue to 'have it on the table'?

For the moment (because I don't have a weight that gives more measurements except for the plain weight) I've disabled all measurements) and that kinda solved my issue (*), but in the future it would be nice to have that sort of customizability.

And there are also #224 and #225 which are quite similar to this one.

(*) - there is still `body mas index` displayed all the time even though I've disabled everything ( #130)