oliexdev/openScale #132: "Normal" weight vs "Normal" bmi

comcamot, 2018-01-05:
I understand that in EvaluationSheet.java one defines the values of "normal" parameters. For BMI one uses age and sex instead of the simplified version which considers normal everything between 18.5 and 24.99.
For weight, however, one just assumes that normal BMI is between 20 and 25 (for men) and from it and the heigth it derives the normal weight.  For ages>40 or ages<20 this leads to inconsistencies (one could be overweight according to weight and underweight according to BMI). The assumed variation of "normal" BMI with age is probably  too agressive. Probably one should set the correct BMI limits first and from them derive the correct weight limits. 
This is the typical behaviour  (from  [https://www.smartbmicalculator.com/])
[image]

oliexdev, 2018-01-05:
Sorry I don't get you point. Currently the evaluation is based on your sex and age. The corresponding ranges are used from scale company or see [here](http://www.bernhard-gaul.de/BMI-Rechner/bmi-rechner.php). So its not basically wrong. Ok maybe this formula is out dated.
What you are referring is a new calculation of the BMI so called "Smart Body Mass Index".
Well this formula could be better I don't know. Do you want to use another calculation?
But its like the calculation for the body mass, body water or body fat calculation, look [here](https://github.com/oliexdev/openScale/wiki/Body-metric-estimations) to see how many different formula are available (and this is only a small selection).
You will find a lot of different formula for calculating the BMI if you search for scientific papers for this research field.

comcamot, 2018-01-05:
No, I do not think it is worth to use a different BMI formula or body fat estimators (I saw that many of them are already implemented). My concern was about what is considered "normal". For BMI one  now uses the tables you [mention](http://www.bernhard-gaul.de/BMI-Rechner/bmi-rechner.php), it is OK (or use any other table, as for instance, the standard table independent on age and sex at [WHO](http://www.euro.who.int/en/health-topics/disease-prevention/nutrition/a-healthy-lifestyle/body-mass-index-bmi)). However, I understand that in the present version of EvaluationSheet.java, the "normal" weight is derived from the "normal" BMI, but in this case the range is taken independent on age, 20 < BMI <25 (in method `evaluateWeight(float weight)`)
`

        if (evalUser.isMale()) {
            lowLimit = body_height_squared * 20.0f;
            highLimit = body_height_squared * 25.0f;
        } else {
            lowLimit = body_height_squared * 19.0f;
            highLimit = body_height_squared * 24.0f;
        }
`
which, according to the tables  [here](http://www.bernhard-gaul.de/BMI-Rechner/bmi-rechner.php), would only be valid for people from 18 to 24.

oliexdev, 2018-01-06:
Ok I think I understand what you mean. The problem is if the BMI is out of range, the weight evaluation will still evaluate. So I could set the evaluation for the weight to none if the BMI was out of range? Or do you have a better evaluation formula for the weight which also include the sex and age?

oliexdev, 2018-01-12:
I am closing this issue here for now. If you have an idea please reply to this issue again.

closed 2018-01-12 by oliexdev

Olangu, 2019-02-02:
WHO defines the normal value between 18.5 and 24.9.
Maybe honor this standard or change the name?

http://www.euro.who.int/en/health-topics/disease-prevention/nutrition/a-healthy-lifestyle/body-mass-index-bmi