oliexdev/openScale #133: Translation participation?

AxiK101, 2018-01-06:
I wanna translate this app to greek language. Is there any way like crowdin for example? Thanks

oliexdev, 2018-01-06:
I don't have crowdin you have to translate the string.xml into your language, see the [wiki](https://github.com/oliexdev/openScale/wiki/Frequently-Asked-Questions-(FAQ)#why-is-my-language-xyz-is-missing-or-incomplete)

AxiK101, 2018-01-06:
Can you help me a bit? What do i translate? The black text? What software do i need? A wordpad?

AxiK101, 2018-01-06:
[image]

oliexdev, 2018-01-06:
Yes translate the black text with wordpad or another text editor.

erijo, 2018-01-06:
@oliexdev: perhaps you can ask for hosting on https://weblate.org/en/hosting/ ? It would make it easier for translators to contribute.

AxiK101, 2018-01-07:
Thanks, i will do that

oliexdev, 2018-01-15:
@erijo I filled out the web form for the free weblate hosting service but it will take time up to one month.

@AxiK101 How is your translation going?

oliexdev, 2018-01-19:
I guess @AxiK101 has vanished, so I will close this issue. Please comment here if you come back or create a PR.

closed 2018-01-19 by oliexdev

erijo, 2018-02-14:
@AxiK101: you can now use https://hosted.weblate.org/projects/openscale/ to do the translation.