oliexdev/openScale #134: App crash with RT-Libra-B

mmangard, 2018-01-07:
Hello, 
I just tried openScale  with the latest development build (v1.7-beta-7d735146 ) and a "RT-Libra-B".
After a successful connection the app displays some values which differ from the display and crashes afterwards. The issues is 100% reproduce able. I use Android version 6.0.1. How can I help? If you want I could do some tests with a development build which includes some debug-traces.

oliexdev, 2018-01-07:
Yeah would be great if you could print the stacktrace so we can see in which source code line the crash happens.

mmangard, 2018-01-07:
How can I provide you the stacktrace? Do I need an  Android Development Environment for that (which I do not have)?

oliexdev, 2018-01-07:
Do you have any programming experiences? You can install android studio and compile the source code.

Or see https://www.utest.com/articles/how-to-get-android-crash-logs-by-using-adb-commands
Or see here https://f-droid.org/de/docs/Getting_logcat_messages_after_crash/

mmangard, 2018-01-14:
See my log/stackstrace [measurelog.txt](https://github.com/oliexdev/openScale/files/1629246/measurelog.txt)

oliexdev, 2018-01-14:
Did you create an user in openScale? And what is the name of the user?

mmangard, 2018-01-14:
I created a user with the name "mm".  In order to double check my settings i opened the settings and the app crashed again. After a restart of the app I could do measurements.

In order to force the problem again I reinstalled the app.  After the initial start the app shows the user-account dialog. After pressing the add button the app crashed with the following stacktrace. 
[settingslog.txt](https://github.com/oliexdev/openScale/files/1629439/settingslog.txt)

For me it seems that these crashes are somehow releated.

oliexdev, 2018-01-14:
ok for first problem I think I know the problem can please try to use a name with at least 3 characters. 

> In order to force the problem again I reinstalled the app. After the initial start the app shows the user-account dialog. After pressing the add button the app crashed with the following stacktrace.
> settingslog.txt

That is strange. Can you exactly tell me what you entered in the user-account dialog? Currently I can't reproduce this error.

mmangard, 2018-01-14:
I can reproduce the crash with a 2 letter username by doing the following steps:

1) delete all existing user 
2) exit the app
3) start the app  and create the following user
[image]

if I choose a username with 3 letters I can create the account without a crash.

oliexdev, 2018-01-14:
In the latest development version you have to enter an username with at least 3 characters.

But does it crash again with your scale when you have a longer username?

mmangard, 2018-01-14:
no it does not crash with a username with at least 3 characters and I can perform a measurement. 
The measurement only results in the weight (water, and other values are missing). 
Is this a known issue?

oliexdev, 2018-01-14:
yes see [here](https://github.com/oliexdev/openScale/wiki/Frequently-Asked-Questions-(FAQ)#how-do-i-get-on-my-sbf70-scale-the-body-metrics-water-muscle-fat-percentage)

closed 2018-01-14 by oliexdev

oliexdev, 2018-01-14:
btw which tool did you use to get the stacktraces? And did you need root permission for that?

mmangard, 2018-01-15:
first I tried "aLogcat" but somehow I could not get any logs so I installed Android Studio and compiled the sources.