oliexdev/openScale #138: Connection to SBF75

Maerzn, 2018-01-11:
Since I was not able to connect to my 'SBF75'-Scale, I wanted to ask, whether there is a trick to connect the scale to the app.
In the device-list, the scale is listed as 'SBF75 [Device not supported]' (in german).
With the proprietary app, the scale is found and works flawlessly.

I'm using android 7.1 on a Z5C

Thanks for the help!

oliexdev, 2018-01-12:
Hi Maerzn which version of openScale did you use? Did you try out the latest development or beta version of openScale?

Maerzn, 2018-01-12:
I used the PlayStore-Version (1.6.1/21).
Due to the question I tried the latest build (auto. build 1.7: 9f76724d)
It is able to connect to the scale, and get my weight along with all other measurements, too.
Seems I was too hasty... when will this be supported in the main release?
Trying to initialize the scale failed (I tried both scale-, app-, and both reset), though. It only worked in the proprietary app.

By the way, thanks for the development of the app!

oliexdev, 2018-01-12:
I am currently working on releasing the new openScale version. :wink: 
So the next version will be released soon but it will a incremental release which means that not all users will see this new release instantly on GooglePlay.

Dododappere, 2018-01-14:
@Maerzn , what do you mean exactly with initialization? I don't see any initialization function in the HealthForYou app....

Maerzn, 2018-01-14:
On first startup (with reset scale), the app asks some data for the users profile, which is later stored in the scale. It is used by the scale to calculate BMI and also display the initials when a weight is recognized as a user.
This allows the scale to save measurements assigned to different users. 
The function only shows up with a completly reset scale and (in my case) no app data on the phone.

Dododappere, 2018-01-15:
Ah ok, I understand. Well, the code is there to do it but it might not be working properly...I'll have a look this evening and test with my SilverCrest scale.

Dododappere, 2018-01-15:
Hi @Maerzn I just tested with my Silvercrest scale and it is working fine as far as I can see. So what I did was:
1. Factory reset the scale by pressing the button at the back of the scale until it says 'del'
2. Do a fresh install of the latest openScale app
3. Enter my personal details like name, height etc. Make sure your firstname has at least 3 characters.
4. Pair the scale
5. Connect to the scale by pressing the bluetooth icon
6. The scale is activated and the first 'reference' measurement can be taken. My initials are shown on the display of the scale now.
7. After the measurement is done the data appears in the app including my correct body composition data

Can you try it also? What is indeed not very clear is that after creating the user on the scale, you are supposed to take a 'reference measurement'. The display of the scale is activated but since it doesn't have backlighting it is not very obvious.

In order to improve the User Experience it would be better if there was a UI prompt for the user saying that a profile is created and that he needs to step on the scale for the first reference measurement....

@oliexdev , can we add this UI improvement?

oliexdev, 2018-01-15:
@Dododappere a toast message should be already shown when you have to step on the scale. Saying that a user profile is created is easy. I could change that.

Dododappere, 2018-01-15:
You are right! It is there....but even I didn't notice it....perhaps some
more 'in your face' would do it?

Maerzn, 2018-01-15:
With the latest revision (c1f247c) , i get following error after creating my 'account':
Build version: 1.7 (beta)-dev_c1f247cb 
Current date: 2018-01-16 00:18:14 
Device: Sony E5823 
 
Stack trace:  
java.lang.IndexOutOfBoundsException: Index: 0, Size: 0
 at java.util.ArrayList.get(ArrayList.java:411)
 at com.health.openscale.gui.activities.UserSettingsActivity$onClickListenerOk.onClick(UserSettingsActivity.java:327)
 at android.view.View.performClick(View.java:5685)
 at android.view.View$PerformClick.run(View.java:22481)
 at android.os.Handler.handleCallback(Handler.java:751)
 at android.os.Handler.dispatchMessage(Handler.java:95)
 at android.os.Looper.loop(Looper.java:241)
 at android.app.ActivityThread.main(ActivityThread.java:6274)
 at java.lang.reflect.Method.invoke(Native Method)
 at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:886)
 at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:776)

User actions: 
2018-01-16 00:17:34: MainActivity resumed
2018-01-16 00:17:34: MainActivity paused
2018-01-16 00:17:34: UserSettingsActivity created
2018-01-16 00:17:34: UserSettingsActivity resumed

It only happens with wiped app data.
Nevertheless, after deleting and recreating he user, the initialisation works (@Dododappere )!
Without the recreation, the scale is not initialized.

oliexdev, 2018-01-16:
@Maerzn I think I found a bug which results in this unexpected behaviour. The user add method was an async process which I have now removed. Please try again with the latest development version.

Maerzn, 2018-01-16:
works perfectly now!

closed 2018-01-19 by oliexdev