oliexdev/openScale #140: Connection to YunChen Smart Scale

pkscwc, 2018-01-14:
Dear Olie,

I have recently purchased this Yunchen Brand smart scale. 
Can you add it as one of supported scales in your program.

I have tried using some hcitool and GATTtool commands to get relevant handles and UUIDs for this scale.

It is based on Texas Instrument chip.

Output of GATTTool commands like Primary  &  char-desc-hnd are given below:

Result of GATTtool log : command "primary"

attrhandle:0x0001,endgrp handle:0x0007 uuid:00001800-0000-1000-8000-00805f9b34fb
attrhandle:0x0008,endgrp handle:0x000a uuid:0000180a-0000-1000-8000-00805f9b34fb
attrhandle:0x000b,endgrp handle:0x000e uuid:00010203-0405-0607-0809-0a0b0c0d1912
attrhandle:0x000f,endgrp handle:0x0013 uuid:0000ffaa-0000-1000-8000-00805f9b34fb
attrhandle:0x0014,endgrp handle:0x001c uuid:0000fff0-0000-1000-8000-00805f9b34fb

Result of GATTtool log : command "char-desc"

handle:0x0001,uuid:00002800-0000-1000-8000-00805f9b34fb
handle:0x0002,uuid:00002803-0000-1000-8000-00805f9b34fb
handle:0x0003,uuid:00002a00-0000-1000-8000-00805f9b34fb
handle:0x0004,uuid:00002803-0000-1000-8000-00805f9b34fb
handle:0x0005,uuid:00002a01-0000-1000-8000-00805f9b34fb
handle:0x0006,uuid:00002803-0000-1000-8000-00805f9b34fb
handle:0x0007,uuid:00002a04-0000-1000-8000-00805f9b34fb
handle:0x0008,uuid:00002800-0000-1000-8000-00805f9b34fb
handle:0x0009,uuid:00002803-0000-1000-8000-00805f9b34fb
handle:0x000a,uuid:00002a50-0000-1000-8000-00805f9b34fb
handle:0x000b,uuid:00002800-0000-1000-8000-00805f9b34fb
handle:0x000c,uuid:00002803-0000-1000-8000-00805f9b34fb
handle:0x000d,uuid:00010203-0405-0607-0809-0a0b0c0d2b12
handle:0x000e,uuid:00002901-0000-1000-8000-00805f9b34fb
handle:0x000f,uuid:00002800-0000-1000-8000-00805f9b34fb
handle:0x0010,uuid:00002803-0000-1000-8000-00805f9b34fb
handle:0x0011,uuid:0000ffab-0000-1000-8000-00805f9b34fb
handle:0x0012,uuid:00002902-0000-1000-8000-00805f9b34fb
handle:0x0013,uuid:00002901-0000-1000-8000-00805f9b34fb
handle:0x0014,uuid:00002800-0000-1000-8000-00805f9b34fb
handle:0x0015,uuid:00002803-0000-1000-8000-00805f9b34fb
handle:0x0016,uuid:0000fff4-0000-1000-8000-00805f9b34fb
handle:0x0017,uuid:00002902-0000-1000-8000-00805f9b34fb
handle:0x0018,uuid:00002901-0000-1000-8000-00805f9b34fb
handle:0x0019,uuid:00002803-0000-1000-8000-00805f9b34fb
handle:0x001a,uuid:0000fff1-0000-1000-8000-00805f9b34fb
handle:0x001b,uuid:00002902-0000-1000-8000-00805f9b34fb
handle:0x001c,uuid:00002901-0000-1000-8000-00805f9b34fb

The values received from each handle above through command "char-read-hnd" in above order are given under:

0018
12
59756e4368656e0000000000000000000000
02
0000
02
140028000000e803
0a18
02
028a2466820100
12190d0c0b0a09080706050403020100
16
00
4f5441
aaff
14
0000000000000000000000000000000000000000
0000
436d64205257
f0ff
10
0000000000000000000000000000000000000000
0000
53707020524f
04
0000000000000000000000000000000000000000
0000
53707020574f

I have browsed through files hosted at openScale/android_app/app/src/main/java/com/health/openscale/core/bluetooth/

and found that four of the java files use these UUIDs.

Like: 
BluetoothBeurerBF700_800 and BluetoothSanitasSbf70 use 

UUIDs of 00002A00, 01, 04 and 50. 

BluetoothDigooDGSO38H and BluetoothExcelvanCF369BLE use UUIDs of 0000FFF0, 1 and 4. 

I hope it helps.

oliexdev, 2018-01-14:
Hi @pkscwc 

I can't find this scale on the internet. :thinking: 
Do you have a link to the vendors website?

Did you read the [wiki](https://github.com/oliexdev/openScale/wiki/How-to-reverse-engineer-a-Bluetooth-4.x-scale) about how to reverse engineer a Bluetooth scale?
Maybe [this](https://github.com/oliexdev/openScale/wiki/Xiaomi-Bluetooth-Mi-Scale) helps a bit too how you do it on a real device.

You have to analyse the incoming and outcoming values so we know which byte is used for what.

Some UUIDs are standard UUIDs but that doesn't mean that the data format are the same (but sure it could be possible).

I don't own this scale so its difficult to help in this stage. You have to do analyse the byte structure and the commands which are send from the your scale.

Btw. please use the Markdown code tags for posting source code or formatted output messages.

Best regards
oli

pkscwc, 2018-01-14:
I have some hci snoop logs.

oliexdev, 2018-01-14:
then try to analyse it with wireshark as described in the wiki.

pkscwc, 2018-01-14:
OK. What data exactly may help us.

pkscwc, 2018-01-14:
I have uploaded the blue pixel ble scan log having UUID s in it.

Dododappere, 2018-01-14:
@pkscwc if your scale uses the FFF1 characteristic I would try to make a copy of the Digoo scale handler code and configure OpenScale to accept your scale. With a bit of luck it will work....if it doesn't you need to do reverse engineering of the hci snoop logs to see what is going on.

pkscwc, 2018-01-14:
Ok. Thanks. I will put in required efforts from my side also.

pkscwc, 2018-01-17:
Hi Olie,

I have done some work today on wireshark and ble snoop.

Perhaps, got the values required.

Hope, it helps.

pkscwc, 2018-01-17:
Hi Olie,

I have attached the ble log file also.

oliexdev, 2018-01-17:
@pkscwc you didn't attached anything.  Also you have to analyse the protocol to find weight and other data in the Bluetooth data packages.

pkscwc, 2018-01-18:
Please check. I can see attachments sent in the email with desired
details...

erijo, 2018-01-18:
I didn't get the attachment either. Please see https://help.github.com/articles/file-attachments-on-issues-and-pull-requests/ for how to attach files.

pkscwc, 2018-01-18:
> I didn't get the attachment either. Please see https://help.github.com/
> articles/file-attachments-on-issues-and-pull-requests/ for how to attach
> files.

OK thanks.

pkscwc, 2018-01-20:
I have submitted it as a new issue with relevant files. Thanks.

pkscwc, 2018-01-20:
[YunchenLog.docx](https://github.com/oliexdev/openScale/files/1648948/YunchenLog.docx)
[YunChen_only_filter.log](https://github.com/oliexdev/openScale/files/1648949/YunChen_only_filter.log)

oliexdev, 2018-01-21:
@pkscwc 
thanks for the files but I need more information.

1. Please provide a link to the vendors app
2. Please provide a link to the product page of your YunChen scale.
3. I need a photo of your scale for the gallery (same angle like the others)
4. I need the complete Bluetooth log file. Your `YunChen_only_filter.log` file filtered to much out. The commands from the vendors App are missing.

pkscwc, 2018-01-21:
In a minute.

pkscwc, 2018-01-21:
https://www.amazon.in/gp/product/B071J756ZB/ref=oh_aui_detailpage_o06_s00?ie=UTF8&psc=1

pkscwc, 2018-01-21:
Vendors other products
https://www.amazon.in/Health-Personal-Care-Hesley-Inc/s?ie=UTF8&field-brandtextbin=Hesley%20Inc&page=1&rh=n%3A1350384031

pkscwc, 2018-01-21:
Vendor's app known as KeepFit
https://play.google.com/store/apps/details?id=com.beikang&hl=en

pkscwc, 2018-01-21:
The complete log. I just realized that I have filtered the commands from localhost to remote. This is the complete log containing all the bluetooth devices nearby to my mobile. Lot of garbage. 

[BtSnoop_17Jan2018.log](https://github.com/oliexdev/openScale/files/1649674/BtSnoop_17Jan2018.log)

pkscwc, 2018-01-21:
hope it helps.

pkscwc, 2018-01-21:
In this file, I have tried to include both. scale to mobile and mobile to scale.
[scale_to_mobile_all.log](https://github.com/oliexdev/openScale/files/1649681/scale_to_mobile_all.log)

pkscwc, 2018-01-21:
Another, filtered but having both way traffic log

[scale_to_PERX_all.log](https://github.com/oliexdev/openScale/files/1649699/scale_to_PERX_all.log)

oliexdev, 2018-01-21:
@pkscwc 
I implemented a first try for Hesley scale. Please try the latest dev version of openScale.

Some thinks that are not clear yet:

1. The scale is sending around 149 (:open_mouth: !) times the same command bytes `a5:01:2c:ab:50:5a:29`  to the scale. After every command the scale send sometimes mostly nothing or the weight with the body metrics.
2. Currently I don't know what the magic bytes `a5:01:2c:ab:50:5a:29` are. Maybe the date and time are in these byte!?
3. Which date and exact time did you weight in the above log files?
4. What did you enter in the app for your body height and your age?

Do you receive any data from the scale with openScale? A toast message should popup with some hex bytes. Please post that here.

pkscwc, 2018-01-21:
Dear Olie,

I have noticed the  magic bytes a5:01:2c:ab:50:5a:29 and tried my level best to connect with them date and time formats. So far, there is nothing to link with them date and time. The exact date and time of each packet is logged and can be seen in Wireshark. 

[image]

pkscwc, 2018-01-21:
another one from today.
[image]

pkscwc, 2018-01-21:
This cannot be a data or time string as it is same for both days i.e. 17th and 21st January. Date & time should be somewhere else.

oliexdev, 2018-01-21:
@pkscwc It was just an idea but please give clear feedback what is working and what not otherwise we don't get any further in the reverse engineering process.

pkscwc, 2018-01-21:
An image of the scale:

[image]

pkscwc, 2018-01-21:
Dear Olie,

I am sorry. But, I don't know how to install developer version of openscale on my mobile. Is there any apk to download? I cannot find anything helpful.

with regards,

pkscwc, 2018-01-21:
Ok.
Got it, I have to install Android-Studio. 
Please give me some time.

oliexdev, 2018-01-21:
Or use this APK file https://github.com/oliexdev/openScale/releases/tag/travis-dev-build

pkscwc, 2018-01-21:
Install openScale-dev-build.apk . Yes found this. Thanks.

pkscwc, 2018-01-21:
In Bluetooth logs, it is recognised as Yunchen only.

pkscwc, 2018-01-21:
App installed without any error. 
It found my scale as Yunchen.
After that, "trying to connect to Yunchen".
And then " no bluetooth devices found".

I noticed that app bluetooth search timeout is rather short. Scale remains active for twice more time.

pkscwc, 2018-01-21:
So, no readings for now.

pkscwc, 2018-01-22:
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

pkscwc, 2018-01-22:
> Or use this APK file https://github.com/oliexdev/
> openScale/releases/tag/travis-dev-build

I used the apk file. It connects but no data comes back from scale to app.

Can we change the magic number and try.

pkscwc, 2018-01-22:
nRF Connect, 2018-01-22
YunChen (78:9C:E7:0C:1F:F3)
I	14:49:22.839	[Server] Server started
V	14:49:22.858	Device Information (0x180A)
- PnP ID [R] (0x2A50)
Unknown Service (00010203-0405-0607-0809-0a0b0c0d1912)
- Unknown Characteristic [N R WNR] (00010203-0405-0607-0809-0a0b0c0d2b12)
   Characteristic User Description (0x2901)
   Client Characteristic Configuration (0x2902)
Unknown Service (0000ffaa-0000-1000-8000-00805f9b34fb)
- Unknown Characteristic [N WNR] (0000ffab-0000-1000-8000-00805f9b34fb)
   Characteristic User Description (0x2901)
   Client Characteristic Configuration (0x2902)
Unknown Service (0000fff0-0000-1000-8000-00805f9b34fb)
- Unknown Characteristic [N] (0000fff4-0000-1000-8000-00805f9b34fb)
   Characteristic User Description (0x2901)
   Client Characteristic Configuration (0x2902)
- Unknown Characteristic [WNR] (0000fff1-0000-1000-8000-00805f9b34fb)
   Characteristic User Description (0x2901)
V	14:49:23.169	Connecting to 78:9C:E7:0C:1F:F3...
D	14:49:23.169	gatt = device.connectGatt(autoConnect = false, TRANSPORT_LE)
D	14:49:23.515	[Server callback] Connection state changed with status: 0 and new state: CONNECTED (2)
I	14:49:23.515	[Server] Device with address 78:9C:E7:0C:1F:F3 connected
V	14:49:23.515	[Server] Creating server connection...
D	14:49:23.515	server.connect(device, autoConnect = false)
D	14:49:23.521	[Callback] Connection state changed with status: 0 and new state: CONNECTED (2)
I	14:49:23.521	Connected to 78:9C:E7:0C:1F:F3
D	14:49:23.524	wait(1600ms)
D	14:49:23.545	[Broadcast] Action received: android.bluetooth.device.action.ACL_CONNECTED
V	14:49:25.131	Discovering services...
D	14:49:25.131	gatt.discoverServices()
D	14:49:25.158	[Callback] Services discovered with status: 0
I	14:49:25.159	Services discovered
V	14:49:25.186	Generic Access (0x1800)
- Device Name [N R] (0x2A00)
- Appearance [R] (0x2A01)
- Peripheral Preferred Connection Parameters [R] (0x2A04)
Device Information (0x180A)
- PnP ID [R] (0x2A50)
Unknown Service (00010203-0405-0607-0809-0a0b0c0d1912)
- Unknown Characteristic [N R WNR] (00010203-0405-0607-0809-0a0b0c0d2b12)
   Characteristic User Description (0x2901)
Unknown Service (0000ffaa-0000-1000-8000-00805f9b34fb)
- Unknown Characteristic [N WNR] (0000ffab-0000-1000-8000-00805f9b34fb)
   Client Characteristic Configuration (0x2902)
   Characteristic User Description (0x2901)
Unknown Service (0000fff0-0000-1000-8000-00805f9b34fb)
- Unknown Characteristic [N] (0000fff4-0000-1000-8000-00805f9b34fb)
   Client Characteristic Configuration (0x2902)
   Characteristic User Description (0x2901)
- Unknown Characteristic [WNR] (0000fff1-0000-1000-8000-00805f9b34fb)
   Client Characteristic Configuration (0x2902)
   Characteristic User Description (0x2901)
D	14:49:29.620	[Server callback] Connection state changed with status: 0 and new state: DISCONNECTED (0)
I	14:49:29.620	[Server] Device disconnected
D	14:49:29.667	[Callback] Connection state changed with status: 8 and new state: DISCONNECTED (0)
E	14:49:29.667	Error 8 (0x8): GATT CONN TIMEOUT
I	14:49:29.667	Disconnected
D	14:49:29.814	[Broadcast] Action received: android.bluetooth.device.action.ACL_DISCONNECTED

pkscwc, 2018-01-22:
Log by nRF logger

oliexdev, 2018-01-22:
please fork the project and edit the `BluetoothHesley.java` file in Android Studio. In that file you can see the magic bytes which you can change.
I can't help you much anymore because you have to do some kind of try and error analyses with your scale.

pkscwc, 2018-01-24:
Hi Olie,

App now gets connected with scale. But it is not showing any measurement
yet. Can we increase the bluetooth time-out of the app. It time-outs before
the scale.

Any suggestion.

oliexdev, 2018-01-26:
@pkscwc 
It has nothing to do with the Bluetooth time-out. Please create some image with the  [BLE Scanner](https://play.google.com/store/apps/details?id=com.macdom.ble.blescanner). Then I can check again of the right services and characteristic. But as I said you have to experiment with your scale.

pkscwc, 2018-01-28:
Dear Olie,

I would like to express my sincere thanks for all the encouragements and actually writing the BluetoothHesley.java file for users of Hesley scale. 

Yes, it worked. 

I studied the logs in wireshark as suggested by you and tweaked the java file. The scale starts broadcasting the data as soon as it is activated.  So, the use of magic bytes are still a mystery. 

Relevant files are attached with screenshot. 
[image]
[BluetoothHesley_worked_java.txt](https://github.com/oliexdev/openScale/files/1671314/BluetoothHesley_worked_java.txt)

pkscwc, 2018-01-28:
Remaining data like fat etc is not present, because scale did not broadcast it.

oliexdev, 2018-01-28:
I implemented your changes into the master repository. Use the latest development version of openScale.

I guess to get the remaining data we have to figure out what the magic bytes are. Until then I will close this issue. If you find a solution create a new issue or PR or reply to this issue.

closed 2018-01-28 by oliexdev

pkscwc, 2018-01-30:
Dear Olie,

It looks like that I made a mistake. It is not getting connected. I only read my manually entered data. I have updated some information in the java file. Particularly the data test part, data [1] = 0x01 should be 0xc1. I have altogether removed it. But, still it is not working. Although, in the log file, it clearly shows that scale is returning full data (including fat etc ) after receiving the magic bytes. 

Your expert opinion is required. 
[BluetoothHesley.java.txt](https://github.com/oliexdev/openScale/files/1678688/BluetoothHesley.java.txt)
[os30jan8640.log](https://github.com/oliexdev/openScale/files/1678691/os30jan8640.log)

pkscwc, 2018-01-30:
Further, the toast commands are not creating any message window. What to do?

oliexdev, 2018-01-31:
I have seen your post but I don't know either.

oliexdev, 2018-01-31:
@pkscwc I changed the implemented Bluetooth search algorithm in openScale which use now the new Android API so please test it with the latest development version.

pkscwc, 2018-02-12:
Hi,

Can you give me some idea for debugging the app. It is not picking up the
reading sent by scale. Scale is surely sending the readings as seen in BLE
log.

Bye

oliexdev, 2018-02-12:
@pkscwc  for debugging see https://developer.android.com/studio/debug/index.html

I will remove this try in the master repository until we get working version.

For reference the latest implementation in the mater repository was [BluetoothHesley.java](https://github.com/oliexdev/openScale/blob/master/android_app/app/src/main/java/com/health/openscale/core/bluetooth/BluetoothHesley.java).

pkscwc, 2018-02-13:
Please check in the new issue. I have uploaded the screenshots and working
java file.