oliexdev/openScale #143: SBF75 - Not properly switching users?

Skyrid3r, 2018-01-14:
I'm currently on the dev version as that's the only version currently that works with the SBF75, and it works like a charm for a single user. But with multiple users it's having some difficulties. Regardless on what 'profile' I set the app to, according to the SBF75 it keeps showing the first added user, rather than my own.

So For example, I have 2 users called Demo 1 and Demo 2.

Sometimes its saving the data on the proper account (Demo 2 on the app, Demo 1 on the device, but writes the data to Demo 2 on the app). 

And sometimes using Demo 2 on the app, showing Demo 1 on the device but writes the data to Demo 1 on the app instead.

The SBF75 also starts with no name, as such i have to stand on it to get a clue which name it is choosing. Not sure if this is related to the device itself, or the name is not being transferred correctly from the app to the device to show the proper name.

Any clue to why this is happening?

oliexdev, 2018-01-14:
Did you enabled the smart user assignment in the Bluetooth settings?
Did you use your scale with the vendors app before? Are the user name in the vendors app and in openScale the same?

For the device itself I don't really now the behaviour because I don't own one.

RishiMehta123, 2018-01-15:
Hi, sorry for jumping on this thread but my issue sounds similar. I've started using my new Mi Scale 2 with the OpenScale app. I have set up 2 users.  Even when I have selected a particular user (e.g. my wife), if I use it straight after (first changing the user selection), both the readings get assigned under the first user! I can edit various other fields of an existing reading, but there doesn't seem to be a way to reassign 'user' within the reading. So I can't correct the misassigned (incorrect) readings. Please advise. Thanks

oliexdev, 2018-01-15:
@RishiMehta123 Can you please answer my questions above as well?

The problem might be that you use enabled smart user assignment then your measurements will automatically sort between the users. It depends on the initial weight (in the user dialog) or at the latest measurement. The user which have the closet measurement are assigned. If you turn that option off all measurements are assigned to the current selected user.

True currently you can't move measurements between users but a good idea. Need to consider it to implement it.

Skyrid3r, 2018-01-15:
smart user assignment was off, left it off as well as I figured it would use the weight to get an estimated user. Hence I wrote this report :). 

As for the official app, not sure. I removed it and I reset the device so I can fully use openScale. The scale itself appears to be using a 3 letter name, rather than the name used in the app.

oliexdev, 2018-01-15:
if you had disabled the smart user assignment, all measurements should add to the current selected user and this also don't work?

Skyrid3r, 2018-01-15:
That depends.. it has a mind on its own. Sometimes its saving correctly, and the other time its saving under the wrong username.

Even if I choose "Demo 2" as selected username on the app, the device shows "Demo 1", but only after I step on it to make its calculations. For some reason no name is showing when I turn the device on, though it shows Bluetooth is connected.

oliexdev, 2018-01-15:
did you delete all user on the scale with the vendors app before you use openScale?

Skyrid3r, 2018-01-16:
I didn't. I removed the app and hard reset the device.

Dododappere, 2018-01-17:
@Skyrid3r the scale uses the weight of the last measurement to recognize the user and therefore it will only display the initials (on the scale) after the measurement. Connecting via BT to the scale does not influence this at all and it will not automatically 'select a user on the scale'. So what you are seeing is 'normal' behavior.

However, using the scale with multiple accounts should work correctly in the sense that measurements should be assigned to the correct user. When the app connects to the scale, the following happens:

1. The list of users of the scale is requested
2. The App compares each user of the scale with the current user in the app
3. If the initials of the scale user (3 chars) is matches the first 3 characters of the app user AND they have the were born in the same year, then that scale user ID is chosen by the app.
4. It then requests the scale to give the measurements for that user ID.
5. Measurements are retrieved and stored.

So as long as the initials/year match the user in the app, it should work.

@oliexdev , I see one potential issue in the code. In the method 'parseScaleData' (line 538 in BluetoothSanitasSbf70) the ScaleMeasurement object is created and all values are set. However, the userID field is not being set. I see that this is later on corrected in the method 'addScaleData' but if smartUserAsssign is TRUE, strange things may happen. Wouldn't it be better to set userID immediately when we create the ScaleMeasurement object. We know exactly which user the measurement is for: the current user!

oliexdev, 2018-01-17:
@Dododappere if you only get measurement from the user which you have previously set or if in the measurement the user id is encoded then sure set the user id. Please create a PR.
The smart user will work if we don't know the users (user id = -1) and the measurements are coming mixed up from the scale then openScale decides by the incoming weight to which user it will be assigned. Like for example the Mi scale don't save on the scale any user id. To support multi user you have to do it like that.

closed 2018-01-19 by oliexdev

Skyrid3r, 2018-01-19:
Wait, closed? How come :p. I haven't attempted to reproduce the issue by using the official app yet and so another reset + using 3 character username.

oliexdev, 2018-01-19:
use the latest dev version of openScale and test it and give feedback here. 
@Dododappere had send already a PR, that's why I closed it. You need to test faster :wink:

Dododappere, 2018-01-19:
Uh, did I? That was for setting the measurement unit....

I'll try to find some time this weekend to do a PR for setting the userID...