oliexdev/openScale #145: Import export questions

ivvv, 2018-01-14:
I have installed v1.7 (beta) to give it a try. I am on Android N (7.0).

I  successfully imported a CSV file with no column names, with the columns ordered following the Play Store version 1.6 export: 
```
date time,weight, fat, water, muscle, waist,bone,hip,lbw,comment. 
```
And the import seems ok, I see all the data correctly in the app. As I did not provide waist, hip and lbw, their ordering is irrelevant in this case, these were all zeros.

Then I exported the table and I see the column names in the csv, but the ordering of the columns seems arbitrary to me: 
```
"bone","comment","dateTime","fat","hip","lbw","muscle","waist","water","weight"
```
How the user would now the ordering of the csv columns if she wants to import data from another scale application?

erijo, 2018-01-14:
As long as the file to import contains the header line (i.e. the line with the column names) the columns can be in any order and with only the valid columns given (i.e. you would not have to include the waist, hip and lbw column).

If the file does not contain the header line, then the order of the columns matter. For full reference, see [getOldStyleHeaders() method](https://github.com/oliexdev/openScale/blob/6247e19a3d839d5beca22fcc8984a64331f8637f/android_app/app/src/main/java/com/health/openscale/core/utils/CsvHelper.java#L35). E.g. a minimal CSV could contain only the columns dateTime, weight, fat, water and comment or include all columns as in the "From version 1.6 up to 1.7" variant. (Please note that this support was added today, so unless you have installed the [dev build](https://github.com/oliexdev/openScale/releases/tag/travis-dev-build) you may not have that support yet.)

oliexdev, 2018-01-15:
As already @erijo wrote you don't have to know the ordering. You are free to order the column as long you have the correct column header with the new openScale version.

closed 2018-01-15 by oliexdev