oliexdev/openScale #146: Suggestion for improved chart

ivvv, 2018-01-14:
Using v1.7 (beta) on Android 7.0.

The chart measurements are displayed with their values. This makes it difficult to see the general trend when there are many data points. The chart is too crowded, see the screenshot.

[image]

I would suggest to remove the value labels from the chart for the individual measurements: the y-axis provides the approximate values, e.g via the grid lines, and they are also available in the table if the actual value is important.

erijo, 2018-01-15:
This can be set in the settings under chart -> Label on data point.

oliexdev, 2018-01-15:
you can even add a regression line into the graph :wink:

closed 2018-01-15 by oliexdev