oliexdev/openScale #147: Bug report: graph time index is broken when current year changes (newbie-friendly?)

piotr-yuxuan, 2018-01-15:
Hiya!

First, thanks a lot for your libre / free app :-) Since January 1st I've noticed there is a bug is the overview graph. We're in 2018, data points from the current year are correctly displayed like "3 days ago", but data points from December 2017 are said to be "343 days ago" or something similar.

I guess it's a pretty easy bug to fix. If you're OK with that, and if you leave me some time for it I'd love to work on it and fix it. It's not a critical bug and the app works well otherwise. If I would fix it, I would only fix the computation, nothing else.

Waiting for your response.
Piotr

erijo, 2018-01-15:
I'm sure @oliexdev would gladly accept contributions, but this particular problem is already fixed in the upcoming release. See #116.

piotr-yuxuan, 2018-01-15:
Indeed, so you can close this issue. Moreover, it would be great if you would label some _ramp up_ tasks so newcomers could still help you guys :-)

oliexdev, 2018-01-15:
sure your are always welcome to contribute. You could for example implement a setting where you can switch between the current theme and a dark theme, see issue #91. You will find a lot of information on the internet about that topic.

You can also support to translate the app into your native language or a in a language which you can speak, see [here](https://github.com/oliexdev/openScale/wiki/Frequently-Asked-Questions-(FAQ)#why-is-my-language-xyz-is-missing-or-incomplete).

closed 2018-01-15 by oliexdev

erijo, 2018-01-16:
Another good way to start is to find something that you think could be improved and fix that. That way you can more easily keep the motivation up.