oliexdev/openScale #151: Beurer BF710

Megachip, 2018-01-16:
Any chance to get support for the BF710?

oliexdev, 2018-01-16:
No as long you reverse engineer your scale on your own. See the FAQ for more information.

closed 2018-01-16 by oliexdev

Megachip, 2018-01-21:
[image]
Hope this will help to start...

pkscwc, 2018-01-22:
Dear Megachip,
the image is not clear.

Can you send it again.

Megachip, 2018-01-22:
@pkscwc *fixed* Sorry, seems to catch the compressed one before. Not yet checked if they use the same characteristics as the BF700/800... hope a little bit. If not, I should try some package dumps next time ;)

Megachip, 2018-01-27:
Seems using the same protocol like other Beurer scales. So it should work with openScale...

Megachip, 2018-04-02:
Anyone had success? Get managed to create a test build but cant install it anywhere :/ (Tested Android 7 and 8)

magaiverpr, 2018-04-02:
Waiting for this too.

This ticket was closed because there is no way to implement?

erijo, 2018-04-14:
If you haven't seen it already, support is available in the latest [dev version](https://github.com/oliexdev/openScale/releases/tag/travis-dev-build).