oliexdev/openScale #153: Sanitas SB70 related questions

PhilH55, 2018-01-18:
Hi, a few questions please:

I've just started to use openScale to get my measurement data from my Sanitas SB70 (I had been using Sanitas HealthCoach app quite successfully (surprising I know!) for a number of months, but recently it stopped communicating; might be related to the Fitbit app recently installed?).

So I've got openScale talking to the scales ok, but it operates slightly differently from HealthCoach. I've tried looking through all the FAQs etc for info, but couldn't find anything that helped.

• It hasn't downloaded the history of measurements that were left in the scale since the last successful sync with HealthCoach, is that correct/expected behaviour?
• The data sync only seems to work at the time I weigh myself; if I try the data sync some time after, it doesn't load the most recent measurement, is that expected?
• I've exported all my data from HealthCoach to Excel. Looking at various comments, it would seem I need to use the latest dev/beta build to be able to successfully import those, is that correct? If so, will that delete the most recent measurements in openScale or do I need to export those first, combine with the HealthCoach data and import it all back in?

Despite the questions, just want to say thanks for developing this app, it looks good and means I don't have to throw away the scales! (Once I've used and got know it a bit more I'll stick a rating on Play store).

oliexdev, 2018-01-19:
Hi @PhilH55 

> • It hasn't downloaded the history of measurements that were left in the scale since the last successful sync with HealthCoach, is that correct/expected behaviour?

Unfortunately, I don't own this scale. So I can only guess.
I mark the history data function in the [wiki](https://github.com/oliexdev/openScale/wiki/Supported-scales-in-openScale) to n/a, because I thought that this scale have no history data function.
But if you said that it works with the original app, it might be that is function is not reverse engineered yet. So if you are right, I have to mark it as `o`.

> • The data sync only seems to work at the time I weigh myself; if I try the data sync some time after, it doesn't load the most recent measurement, is that expected?

I would guess this depends on the question above.

> • I've exported all my data from HealthCoach to Excel. Looking at various comments, it would seem I need to use the latest dev/beta build to be able to successfully import those, is that correct? If [...]

No if you import something, nothing will be deleted. In the beta/dev version your are more flexible with the import, see #145 .

Best regards
oli

lenucksi, 2018-01-21:
I also own that scale, so here are a few answers.

>>     • It hasn't downloaded the history of measurements that were left in the scale since the last successful sync with HealthCoach, is that correct/expected behaviour?

> Unfortunately, I don't own this scale. So I can only guess.
> I mark the history data function in the wiki to n/a, because I thought that this scale have no history data function.
> But if you said that it works with the original app, it might be that this function is not reverse engineered yet. So if you are right, I have to mark it as o.

The standard use case as intended by the vendor app appears to have been the follwing. Sadly the quality of the vendor app leaves quite a bit of room for improvement.
- Register yourself to the scale using the app, creating a user.
- Measure yourself whenever. The scale will store show and the store your measurements.
- Repeat for a number of times, all measurements will be stored.
- At some time, connect to the scale using the app. 
  - The app will fetch all stored data, remove it from the scale / trigger removal of seen entries. 
  - The app imports all fetched measurements and updates the view with it.

However, since the connectivity to the scale using the vendor app only worked occasionally the on scale queue reached its maximum capacity at some point at which the scale refused to do more than standard weight measurements. At this point a reset using the reset button at the bottom was required. Sadly, any attempts to re-register the scale with the vendor app and use it were unsuccessful at.

>    • The data sync only seems to work at the time I weigh myself; if I try the data sync some time after, it doesn't load the most recent measurement, is that expected?

Based on the vendor app behavior the desired use case appears to have included offline-fetching of data, i.e. fetching of stored measurements without activating the scale / activly measuring yourself.

oliexdev, 2018-01-21:
@lenucksi 
thank you for the explanation. Do you know the behaviour of openScale? Does it load the stored measurements if you are not on the scale?

lenucksi, 2018-01-22:
@oliexdev From what I can remember - as the vendor app doesn't do anything anymore here - it was no problem to retrieve measurements from the scale without stepping on it. You just had to be in BTLE range.

PhilH55, 2018-01-22:
Thanks for the responses to date guys. @lenucksi, you are correct, the vendor app (when it worked!) retrieved all past measurements from the scale when you opened the app and were within range; you didn't need to activate the scales by stepping on them.

@oliexdev So far with openScale it hasn't loaded any historical data for me, it just loads the current measurement when I stand on the scale?

Dododappere, 2018-01-23:
Hi @PhilH55 , I did some testing at home but as far as I can see the retrieval of stored measurements is working fine. What I did was:

- Kill the app
- Take 3 measurements a couple of minutes apart
- Open the app
- Connect to the scale but don't stand on it
- All 3 measurements appear as expected

Could you try the same? Should work.

P.S. Make sure HealthCoach isn't running at the same time....or better even, uninstall it if you can.

PhilH55, 2018-01-23:
Hi @Dododappere 

Thanks for following up on this and testing it out, really appreciate it. It's good to know that the app should capture all measurements. I'll do as you suggest and try it out when I get home and see how we get on. I'll keep you posted.

Thanks again.

lenucksi, 2018-01-23:
Hi @Dododappere which OpenScale version did you make this observation with? I could try to reproduce them using my unit if I would know a bit more.

Dododappere, 2018-01-23:
I am using the latest code, compiling it myself...

I did some more investigation and I found some conditions when the test I did could fail:

1. If you are actively doing a measurement while openScale connects, the initialisation process where openScale looks for the right user is interrupted by the active measurements coming in. So openScale doesn't know which user you are...
2. If the first point occurs, openScale will also not request stored measurements because that happens after the initialisation completes
3. If you complete the measurement anyway, openScale will try to determine which user it could be, potentially assigning it to the wrong user. The scale will assigned the measurement to a particular user but openScale doesn't look at that...

So if you want to be sure everything goes ok, wait until openScale connects to the scale, then do a measurement. Stored measurements should then also arrive.

@oliexdev , I'll probably need some help understanding the queueing mechanism to make it more robust.

Dododappere, 2018-01-23:
@lenucksi , let me know your findings! I don't think the version of openScale matters in this case...nothing changed recently, it has been like this for a while

MartinNowack, 2018-01-23:
@Dododappere the race condition could be indeed an issue.
Normally, if you connect with the scale (clicking the button):
* it should read the history data and sort them for the currently selected user and delete the old one. The problem is that there is currently not really a notification if new data is read/added
* if you step on the scale (and no user on the scale is found) it should add you automatically the first time and make the base measurement
* if you step on the scale, another time - new measurements are added

Have a look at #66 to see the changes I added that should give you an idea.

@Dododappere I currently don't have time to look into it.
Can you try the following:
* there is a registering to event changes: can you disable it and postpone it until all history has been read and after that register it (as a later step of the state machine) that should hopefully fix it.

MartinNowack, 2018-01-23:
@PhilH55 Can you change the title? It's the SBF 70 just to make sure this thread can be found. Thanks!

PhilH55, 2018-01-24:
Thanks for looking in to this guys. I didn't have a lot of time to play last night, but I removed the HealthCoach app from the phone, stood on the scales a few times over a ½ hr period then connected openScale. It found and connected to the scales no problem, but downloaded no measurements. It's probably me doing something not quite right.

When I transferred from HealthCoach to openScale, I didn't do a reset on the scales, just matched the user profile names in the app to the scales, do you think this might be a cause of the problem. I'm using the latest version [v1.7(23)] of openScale from Play store.

D-o-m-i-n-i-c, 2018-01-24:
Hi all,
I'm new to Github so apologies if i'm posting this in the wrong place or if this question has been answered elsewhere..
I have a Silvercrest SBF-75 scale and as seems to be the norm the stock "HealthForYou" app has completely stopped communicating with the scale, it worked well for about two weeks but now will not connect to the scale.
I installed OpenScale and it connected instantly (Kudos on that alone!!) , 
My question is that that on OpenScale the only values measured and displayed are weight and BMI, is this because of an account/sync issue with the scale or because i need to configure some sitting in the app? 
Are the %Body fat, %water, bone density etc values calculated on the scale and then transferred to the app when a user account is correctly setup or are metric values transmitted to  the app and calculated in the app?
Is there a procedure I can carry out to set a new account on the scale with OpenScale?

Thanks in advance for any info and great job on the app!

MartinNowack, 2018-01-24:
@PhilH55 It should work like that: You step on the scale. Depending on your weight (plus/minus some minor changes), the scale is guessing it's you. And it has following things saved of you:
* The first three letters of you username (as uppercase)
* your birthday
* your sex
* your height
* your activity level

Currently openScale is just checking for your name and birthday and find you by that.
If it is not working - we should fix that ;)

MartinNowack, 2018-01-24:
@D-o-m-i-n-i-c fat/water/bone density is calculated by the scale. But I'm currently not sure if it is transmitted correctly.

To create a new user:
* Create a new user in the app
* step on the scale
And a base measurement should be done automatically.

Keep in mind, if the scale finds your existing profile - none will be added.

D-o-m-i-n-i-c, 2018-01-24:
@MartinNowack  Thanks for the advice, I had already created a profile with OpenScale but I think the existing accounts from the manufacturer app may have been causing the conflict/issue. I will reset my scale to clear any existing accounts from the manufacturer app this evening and try to sync the OpenScale account then. i will report my findings back.
Thanks again.

D-o-m-i-n-i-c, 2018-01-25:
Hi all,
Just to update, unfortunately I still had no luck getting my scales to measure more than just weight.

I followed the instructions above, using the OpenScale app but the app doesn't seem to set-up the user profile on the scale so even though the bluetooth symbol is displayed to show there's is a connection the scale doesn't detect an user account and just displays "- - - -" on the bottom left of the display.
I have previously rest my scale so there is no user accounts currently stored on it.

I even tried reinstalling and having another go at using the manufacturer's "HealthForYou" app in the hope of getting a user account set up on the scale and then use the same user details to OpenScale and go from there but the the manufacturer's app still wouldn't not connect to the app.

I've tried all the combinations I can think of paring, measuring, resetting etc, unfortunately I'm getting close to accepting defeat and returning the scale to the shop

Nadivar, 2018-01-25:
Hi, I see this app in dark color, How I can change to light color?
Thanks

oliexdev, 2018-01-26:
@D-o-m-i-n-i-c 
I am sorry for your disappointment maybe you get a faulty scale or you still doing something wrong. Read the [FAQ](https://github.com/oliexdev/openScale/wiki/Frequently-Asked-Questions-(FAQ)#how-do-i-get-on-my-sbf70-scale-the-body-metrics-water-muscle-fat-percentage) on this. 

@Dododappere 
The queue mechanism is really easy you have to queue all message that you want to send to the scale because you can't send two Bluetooth command one after another. You have to wait for the response. After you received the response you can send the next Bluetooth command and so on.

lenucksi, 2018-01-27:
I've just attempted another measurement with version 1.7 of Openscale and a SBF70 scale.
I was successfully able to measure weight and fat as well as create a user on the scale and associate the app with it.

Here's what I did:
- Reset the scale.
- Create a user in the app. I've added a few more letters to the username. Before I was using 2, now 4 that were truncated to three on the scale. Using >= letters for the username seems to be very relevant as illustrated above. I guess it would be valid to pad or otherwise enlarge too short usernames to prevent malfunction through too short names in case this is an actual bug. 
- Enable "auto search" and "auto associate measurements with user" settings
- Search and add the scale (if the scale is off, shortly step on it to get it to activate)
- Execute the reference measurement with bare feet.

At this point the scale recognized my name and showed it in the display together with the BT symbol. After measuring my weight (all with the app open) the usual "fat / non-weight metric" animation at over the name started, finished and the scale started the reel displaying the measured metrics. It looks like the non-weight measurements were successfully after the reel finished.

If have to yet test the offline retrieval of stored data but will do so in the coming days.
Once the process is sucessfully debugged it might a valuable target to document a longer part of the designated usage in the wiki. Or embed the wiki into the app.

oliexdev, 2018-01-28:
> Create a user in the app. I've added a few more letters to the username. Before I was using 2, now 4 that were truncated to three on the scale. Using >= letters for the username seems to be very [...]

In the latest version it is already restricted to have a minimum user name with three characters.

> Once the process is sucessfully debugged it might a valuable target to document a longer part of the designated usage in the wiki. Or embed the wiki into the app.

Sure, I already added the reset hint into the wiki but I won't embed the wiki into the app.
But if you have a better explanation, you are welcome to update the wiki text.

lenucksi, 2018-01-30:
> Sure, I already added the reset hint into the wiki but I won't embed the wiki into the app.
> But if you have a better explanation, you are welcome to update the wiki text.

I'd suggest to add a "Help" button in the main menu slider from the left just below the settings button. Have that open the wiki then.

oliexdev, 2018-01-31:
@lenucksi good idea with the help button to link to the FAQ page that would be an option.

closed 2018-01-31 by oliexdev

PhilH55, 2018-02-06:
Just wanted to say thanks to everyone who contributed to my question here. Sorry for slow response, but I've now had the chance to look at this again and managed to get it all working properly. I successfully imported all the previous data from the Sanitas vendor app and after completely resetting the scales, got the app to successfully sync multiple readings from the scales, so success all round!

Again, thanks for the help and it's great to be able to fully use the scales again.

jflesch, 2018-02-09:
I did the reverse-engineering and the implementation for the SBF-70. I think I need to clarify what is supported and what is not. It's actually simple, I didn't have much time to make it, so unless someone extended it, support is minimal:

- Unsupported: It doesn't know how to register a new user in the scale (you still need the HealthCoach app for that).
- Unsupported: It doesn't know how the scale identifies users. Any ID info is ignored. I'm not even actually sure that the scale does identify the user. I woulnd't be surprised if the HealthCoach application actually does that based on the measures.
- Unsupported: It doesn't know how to get the history from the scale.
- Supported: It does know how to read the values sent by the scale when an user steps on it.
- Unsupported: It doesn't know how to purge the history stored in the scale. Sorry for that. I though the last sequence sent to the scale took care of it, but I've recently seen that it doesn't. At some point, when you use OpenScale, the memory of the scale becomes full and it refuses to take any other measurements.

I'll try to fix the last point in the coming weeks. However I probably won't have time to fix any other, sorry :/

jflesch, 2018-02-09:
Update: Sorry I wasn't aware that @MartinNowack completed the support of the SBF-70. Kudos to him :-)

MartinNowack, 2018-02-09:
No, no, kudos to  you @jflesch!! My changes mainly built on your work and effort. Without it - I wouldn't have started but sent the scale back immediately ;)