oliexdev/openScale #154: hardware choice is hard

drandreaskrueger, 2018-01-19:
I can choose freely, great.  But capitalism also sucks - there are so many crappy products out there.  

just want   

* a good scale, which is 
* sufficiently accurate, and which 
* has a good app.

When checking all scales apps, I was *really* happy to find your App "openScale". Great that you go through the effort of freeing that type of hardware, much much appreciated.  Also I assume many of the other apps force us to store stuff in the cloud, and I don't want that.

Some app ratings:  

4.2 openScale  
2.8 Medisana VitaDock+  
2.7 renpho  
2.6 Yolanda (marsboy, AnGeer)
2.5 New iwellness (hamswan?)
2.4 1byone wellness 2.0  
which other apps are out there?

## choose hardware 

Now ... as I have not yet bought the scale ... **which one do you recommend**?

At the moment your wiki looks as if everything is pointing to the [Medisana BS444](https://github.com/oliexdev/openScale/wiki/Supported-scales-in-openScale); because you have a dedicated page for that one (and the only other one, Xiaomi, is not yet fully freed); and the Medisana has no warnings in the table (apart from n/a not applicable -  *why* is that not appilicable here?).

The Medisana gets 3.3 out of 5.0 points from 383 customers ([DE](https://www.amazon.de/dp/B01DPJSOT6/), none in [UK](https://www.amazon.co.uk/dp/B01DPJSOT6)), and [21% disappointed](https://www.amazon.de/product-reviews/B01DPJSOT6/ref=cm_cr_dp_hist_one?filterByStar=one_star) people talk about the drawbacks.

I don't want to **buy** on Amazon - but the ratings (are sometimes manipulated yes, but) help to avoid totally crappy products - and after choosing, then I can buy locally, not from the devil. 

Some (rather randomly selected) scales:

* Marsboy - not many [disappointed](https://www.amazon.co.uk/product-reviews/B01N99NODP/ref=cm_cr_dp_d_hist_1?filterByStar=one_star) - "Amazon's choice"
* AnGeer - very few [disappointed](https://www.amazon.co.uk/product-reviews/B06W54P7M7/ref=acr_dpx_hist_1?filterByStar=one_star) - (actually 87% happy could mean manipulated?)
* Hamswan - [still very few opinions](https://www.amazon.co.uk/dp/B074CGC4LJ), unclear which app to use - "Amazon's choice"
* 1byone - [many many reviews](https://www.amazon.co.uk/product-reviews/B01FHF9DOQ/ref=cm_cr_dp_d_hist_1?filterByStar=one_star) - "Amazon's choice"
* Omron [BF508](https://www.amazon.co.uk/product-reviews/B0033AGBVG/ref=cm_cr_dp_d_hist_1?filterByStar=one_star) and [BF511](https://www.amazon.co.uk/product-reviews/B0033AGBVQ/ref=cm_cr_dp_d_hist_1?filterByStar=one_star) do not mention Bluetooth? 

That handle for the hands ... looks like it makes a lot of sense to measure visceral fat.

**Important:** But if the scale is not supported by your app, I don't want it.

Here are lists of popular scales:

* https://www.amazon.co.uk/s/?keywords=body%20fat%20scales
* https://www.amazon.co.uk/s/?keywords=body%20fat%20scales%20android
* https://www.amazon.co.uk/s/?keywords=body%20composition
* https://www.amazon.co.uk/s/?keywords=smart%20scales%20body%20analyser
* https://www.amazon.co.uk/s/?keywords=digital%20scales%20bluetooth

Please give me an idea, some hints what to look for, or even a direct recommendation.

Thanks a lot,  
Andreas

drandreaskrueger, 2018-01-19:
P.S.: Yes, I know, github issues were originally not intended for such a question. But it is of paramount importance - a bad scale together with your good app ... will probably lead to an overall bad experience, no?  Shall we make this topic into a wiki page?

oliexdev, 2018-01-20:
Hi @drandreaskrueger 

thank your for the compliment. :smile: 

Yeah it is not really easy to find a good scale which is perfect. I wrote some thoughts in issue #96. Maybe it will help you a bit in your decision. 

Well, your popular scales are only popular in UK :wink: 

Yeah is a good idea to write some statement into a wiki page. If you want you can write a draft version.

Best regards
oli

lenucksi, 2018-01-21:
@drandreaskrueger The BF511 actually is one of the better scales.
The fat measurements using the handles deviate quite a bit from the "legs only" version. Also, that type of scale is the only one that was not ridiculed entirely by a medical doctor, although he did recommend the caliper measurement for more precision. The scale itself is well made and reliable.

However, that scale is unluckily totally unconnected. 
Thus, manual hardware instrumentation as illustrated elsewhere by @oliexdev would be required.

drandreaskrueger, 2018-01-21:
Thanks a lot!

I won't have the time to hack a bluetooth scale myself, even though the [description reads really interesting](https://github.com/oliexdev/openScale/wiki/Custom-Bluetooth-Scale).  (I would actually buy a readily hacked & modified high quality scale if someone builds it for me, but I guess that could quickly become pretty expensive?)

Hmmm ... what to do now?  [This reads](https://github.com/oliexdev/openScale/issues/96#issuecomment-343475342) as if I should buy the **Sanitas SBF70** ?

Customer feedback:
* DE 51% (!) [one star](https://www.amazon.de/product-reviews/B00NABN36E/ref=cm_cr_dp_d_hist_1?filterByStar=one_star) ratings, but most seem to have BlueTooth problems with the app. Perhaps with OpenScale that is not even a problem then?
* DE only 21% [five star](https://www.amazon.de/product-reviews/B00NABN36E/ref=cm_cr_dp_d_hist_1?filterByStar=five_star) - but they seem to be very happy; and OpenScale is recommended there too
* [UK similar](https://www.amazon.co.uk/product-reviews/B00NABN36E/ref=cm_cr_dp_d_hist_1?filterByStar=one_star) - but even less full happy customers

So ... judging by the rating, I would have never chosen that thing.

drandreaskrueger, 2018-01-21:
> ...  wiki page. If you want you can write a draft version.

As long as the wiki points here, I would have found it. So what about this temporary solution:

---

https://github.com/oliexdev/openScale/wiki/What-to-buy   

# What to buy   

See these issues to get some ideas what could be a good scale:

* questions: [#96](https://github.com/oliexdev/openScale/issues/96), [#154](https://github.com/oliexdev/openScale/issues/154)
* answers: [#96 Nov 10th 2017](https://github.com/oliexdev/openScale/issues/96#issuecomment-343475342), [#154 Jan 20th 2018](https://github.com/oliexdev/openScale/issues/154#issuecomment-359154463)

---

drandreaskrueger, 2018-01-23:
@oliexdev - how can I send you a PM? Are you on Twitter? email?

MartinNowack, 2018-01-23:
@drandreaskrueger I can highly recommend the SBF70 because I added most of the advanced stuff to this app  ;) #66 (Adding users automatically, reading history, updating time/date of the scale, ... ) There must be some bug (timing issues) in the original app which lets it run well with older android versions but starting with 5.* it gets really bad.
I added many comments to the different parts of the code so it should be easy to read/change.

@oliexdev Can you fix the wiki? The SBF70 has history support. It should also delete old entries if they have been read and for the FAQ section: Just resetting the scale with the button on the back is super easy.

drandreaskrueger, 2018-01-24:
> some bug (timing issues) in the original app 
> which lets it run well with older android versions but starting with 5.* it gets really bad.

oh oops. I am on 7.1.2.  Does that mean I will have problems?

Scale is going to arrive this weekend. Hooray, really looking forward to this.

MartinNowack, 2018-01-24:
@drandreaskrueger Whoops that was misleading - i meant the original software seems to have problems with newer android versions.

drandreaskrueger, 2018-01-25:
Ah, good, I am relieved ;-)

A big thanks for your contribution then.

oliexdev, 2018-01-26:
@MartinNowack I updated the wiki.

@drandreaskrueger I wrote you back but your email address `2018@....` was invalid so I wrote to `mail@....`

closed 2018-01-26 by oliexdev

drandreaskrueger, 2018-01-26:
Thanks a lot.

Yes, got your mail, well done - and repaired the 2018@

Have a great weekend.

Andreas