oliexdev/openScale #158: App crash, parallel use of Amazfit Notify and openScale

chross, 2018-01-22:
Build version: 1.7 
Current date: 2018-01-22 13:12:10 
Device: Xiaomi Redmi 4X 
 
Stack trace:  
java.lang.NullPointerException: Attempt to invoke virtual method 'com.health.openscale.core.utils.Converters$WeightUnit com.health.openscale.core.datatypes.ScaleUser.getScaleUnit()' on a null object reference
	at com.health.openscale.core.OpenScale.addScaleData(OpenScale.java:228)
	at com.health.openscale.core.OpenScale.addScaleData(OpenScale.java:184)
	at com.health.openscale.gui.MainActivity$9.handleMessage(MainActivity.java:379)
	at android.os.Handler.dispatchMessage(Handler.java:102)
	at android.os.Looper.loop(Looper.java:154)
	at android.app.ActivityThread.main(ActivityThread.java:6236)
	at java.lang.reflect.Method.invoke(Native Method)
	at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:891)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:781)

User actions: 
2018-01-22 06:35:53: MainActivity resumed

oliexdev, 2018-01-22:
Do you add an user in openScale?
Is this crash repeatable?
Does it also crash when your Amazfit Notfiy is not running?

chross, 2018-01-29:
Sorry for replying late.
The point was I did not add any user. After adding a user everything was fine. Shouldn't happen, though. Maybe prevent BT connection when there are no users.

erijo, 2018-02-22:
@chross: can you reproduce this issue with the latest dev build (https://github.com/oliexdev/openScale/releases/tag/travis-dev-build)? It can be installed alongside the official version.

chross, 2018-02-26:
The dev build says something along the lines of "No user present. Please add a user unter settings." When trying to connect to BT or add values manually. Working fine.

erijo, 2018-02-26:
Excellent, thanks for verifying.