oliexdev/openScale #159: Add support for 1byone scale

dinki, 2018-01-22:
I'm writing in hopes that you can add 1byone BT scale (https://www.amazon.com/1byone-Bluetooth-Wireless-Bathroom-Visceral/dp/B01FHELB56/)  to your list of supported devices.  I received this as a gift from my wife but it is tied to a proprietary app that leaves much to be desired.

While I am much more of a hacker than a programmer, I am pretty sure that this is possible as someone has written some software that pulls the information from the scale:

https://github.com/ec1oud/qt-ble-weightscale

Further, I was able to use BT Snoop to pull the log where the scale connects and passes the information on to the phone.  You can see the same type of information listed at the above repo here:

https://github.com/ec1oud/qt-ble-weightscale/blob/master/doc/notes.txt

Here's what it looks like:

```
Frame 13693: 32 bytes on wire (256 bits), 32 bytes captured (256 bits)
Bluetooth
Bluetooth HCI H4
Bluetooth HCI Event - LE Meta
    Event Code: LE Meta (0x3e)
    Parameter Total Length: 29
    Sub Event: LE Advertising Report (0x02)
    Num Reports: 1
    Event Type: Connectable Undirected Advertising (0x00)
    Peer Address Type: Public Device Address (0x00)
    BD_ADDR: TexasIns_2a:8d:a6 (f0:c7:7f:2a:8d:a6)
    Data Length: 17
    Advertising Data
        Flags
            Length: 2
            Type: Flags (0x01)
            000. .... = Reserved: 0x0
            ...0 .... = Simultaneous LE and BR/EDR to Same Device Capable (Host): false (0x0)
            .... 0... = Simultaneous LE and BR/EDR to Same Device Capable (Controller): false (0x0)
            .... .1.. = BR/EDR Not Supported: true (0x1)
            .... ..0. = LE General Discoverable Mode: false (0x0)
            .... ...1 = LE Limited Discoverable Mode: true (0x1)
        16-bit Service Class UUIDs (incomplete)
            Length: 3
            Type: 16-bit Service Class UUIDs (incomplete) (0x02)
            UUID 16: Unknown (0xfff0)
        Manufacturer Specific
            Length: 9
            Type: Manufacturer Specific (0xff)
            Company ID: Ericsson Technology Licensing (0x0000)
            Data: f0c77f2a8da6
    RSSI (dB): -60

Frame 13694: 33 bytes on wire (264 bits), 33 bytes captured (264 bits)
Bluetooth
Bluetooth HCI H4
Bluetooth HCI Event - LE Meta
    Event Code: LE Meta (0x3e)
    Parameter Total Length: 30
    Sub Event: LE Advertising Report (0x02)
    Num Reports: 1
    Event Type: Scan Response (0x04)
    Peer Address Type: Public Device Address (0x00)
    BD_ADDR: TexasIns_2a:8d:a6 (f0:c7:7f:2a:8d:a6)
    Data Length: 18
    Advertising Data
        Device Name: Electronic Scale
            Length: 17
            Type: Device Name (0x09)
            Device Name: Electronic Scale
    RSSI (dB): -60

Frame 13721: 14 bytes on wire (112 bits), 14 bytes captured (112 bits)
Bluetooth
Bluetooth HCI H4
Bluetooth HCI ACL Packet
Bluetooth L2CAP Protocol
Bluetooth Attribute Protocol
    Opcode: Write Request (0x12)
        0... .... = Authentication Signature: False
        .0.. .... = Command: False
        ..01 0010 = Method: Write Request (0x12)
    Handle: 0x0015 (Unknown: Unknown: Client Characteristic Configuration)
        [Service UUID: Unknown (0xfff0)]
        [Characteristic UUID: Unknown (0xfff4)]
        [UUID: Client Characteristic Configuration (0x2902)]
    Characteristic Configuration Client: 0x0001, Notification
        0000 0000 0000 00.. = Reseved: 0x0000
        .... .... .... ..0. = Indication: False
        .... .... .... ...1 = Notification: True
    [Response in Frame: 13722]

Frame 13722: 10 bytes on wire (80 bits), 10 bytes captured (80 bits)
Bluetooth
Bluetooth HCI H4
Bluetooth HCI ACL Packet
Bluetooth L2CAP Protocol
Bluetooth Attribute Protocol
    Opcode: Write Response (0x13)
        0... .... = Authentication Signature: False
        .0.. .... = Command: False
        ..01 0011 = Method: Write Response (0x13)
    [Handle: 0x0015 (Unknown: Unknown: Client Characteristic Configuration)]
        [Service UUID: Unknown (0xfff0)]
        [Characteristic UUID: Unknown (0xfff4)]
        [UUID: Client Characteristic Configuration (0x2902)]
    [Request in Frame: 13721]

Frame 1965: 28 bytes on wire (224 bits), 28 bytes captured (224 bits)
Bluetooth
    [Source: TexasIns_2a:8d:a6 (f0:c7:7f:2a:8d:a6)]
    [Destination: AsustekC_7e:22:04 (50:46:5d:7e:22:04)]
Bluetooth HCI H4
Bluetooth HCI ACL Packet
Bluetooth L2CAP Protocol
Bluetooth Attribute Protocol
    Opcode: Handle Value Notification (0x1b)
    Handle: 0x0014 (Unknown: Unknown)
    Value: cf01adaa0404015b1e026c1301cd0599

cf01adaa 0414 015e 1c02 7013 01ca059c
104.4 kg, fat 35%, water 45.8%, muscle 62.4kg, bone 2.8kg

cf01adaa 0406 015b 1e 026d 13 01cd 0599
103.0 kg, fat 34.7%, water 46.1%, muscle 62.1kg, bone 3.0kg, BMR 1433 kcal

0x0406: 1030  kg
0x015b: 347 fat
0x1e: 30 bone
0x026d: 621 muscle
0x13: 19 visceral fat
0x01cd: 461  water
0x0599: 1433  BMR
```

I am certainly willing to do testing on this if someone can decrypt the handshake and add the functionality to openScale.

Let me know what you think!

Thanks

oliexdev, 2018-01-22:
I need more information. What is the Bluetooth device name? You have to do step 1 and 2 in the [wiki](https://github.com/oliexdev/openScale/wiki/How-to-reverse-engineer-a-Bluetooth-4.x-scale)! Thanks to @ec1oud step 3 has been already analysed.

dinki, 2018-01-22:
Here's three files for part one:
82.4 pounds / 37.37601 kg
https://www.dropbox.com/s/exc8lndzddqw8o2/824btsnoop_hci.log?dl=0

92.6 pounds / 42.00265 kg
https://www.dropbox.com/s/5rx6u91psujh7vd/926btsnoop_hci.log?dl=0

87.8 pounds / 39.82541 kg
https://www.dropbox.com/s/xv21yiuclcz179i/878btsnoop_hci.log?dl=0

Here's an Imgur gallery with the information for part two:
https://imgur.com/a/sV8tC

I couldn't figure out how to export this as text.

Please let me know how I can assist further.  Thanks again for your time.  This is a pretty good scale at a really good price so I'd imagine quite a few people would fine this a useful addition of openScale as the 1byOne app is not too good.

Thanks!

dinki, 2018-01-24:
Great news.  Can I do anything to help with anything?  I know having things as images makes it impossible to copy/paste.  Do you guys need me to transcribe any of this?

oliexdev, 2018-01-26:
@dinki 
I implemented a first try. Please try out the latest development version.
Is openScale connected to the scale?
Do you receive anything? (a toast message should be shown)

dinki, 2018-01-26:
Thanks for spending time on this! 

To be clear, it's the APK at https://github.com/oliexdev/openScale/releases/tag/travis-dev-build  right?  Will download and try this evening.

oliexdev, 2018-01-26:
@dinki that is correct

dinki, 2018-01-27:
Just tried and I could not get it to connect. I went to settings Bluetooth and tried selecting device. It went to a white screen and nothing happened. No error message just a white screen. Any ideas?

oliexdev, 2018-01-27:
Then your scale is still connected to the vendors app or not active.
But you have a white screen? At least a message should shown with "Searching for Bluetooth scales"

dinki, 2018-01-27:
I tried again and did notice the searching message for a split second before it went to the white screen.  The 1byone app requires you to open the app before it will attempt to sync so I doubt that it was causing any connection issues.  To be sure, however, I force closed it before trying with openScale.

I'm running v1.7-dev_5dd254f7 (1516984585) with bluetooth options search for scale on startup.  Device Type shows up as '-'

oliexdev, 2018-01-28:
That's weird, because what openScale does when you click on the settings->bluetooth->device is that it only searches for every nearby Bluetooth device and list them in the App with their name and address.
But openScale can only see devices which are not connected to anything else.

Do you see the device in the Android Bluetooth settings?

dinki, 2018-01-29:
I do not see it there.  I tried to scan from Android Bluetooth settings and it showed up as 'Health Scale'.  I tried connecting/pairing but it failed.

I ensured that the bluetooth symbol on the scale was flashing when trying to connect/scan using openScale.  I'm not sure what else I could do.  Any thoughts?  Thanks

oliexdev, 2018-01-31:
Sorry I have no other idea. As I wrote openScale doesn't do anything else beside scanning for active Bluetooth devices and list them if you click in the openScale settings->bluetooth->device. If they can't even been seen there then the scale is hidden or paired somehow?! I don't know if it even possible to disable the scale visibility in Bluetooth 4.x.

As long if you can't fix this problem I will close this issue.

dinki, 2018-01-31:
Understood.  The scan doesn't show any bluetooth devices.  Not just the scale.  Should I be able to see any and all bluetooth and bluetooth le devices I have there or does it somehow limit to only scales?

My phone is a Samsung S4.  I do realize that it is old but it does what I need.  Could this be the problem?

oliexdev, 2018-01-31:
openScale should list all Bluetooth 4.x devices. 

I don't know if you smartphone is the problem. If you can try out another tablet or smartphone, you could test it. But I doubt that because it did work with the vendors app didn't it?

I changed the implemented Bluetooth search algorithm in openScale which use now the new Android API so please test it with the latest development version.

dinki, 2018-01-31:
Thanks for doing this.  I'm at work but just downloaded the dev version.  I went to Device Type and now it is showing nearby BT devices so this looks promising.  Will update when I get home this evening and can try with the scale.

oliexdev, 2018-01-31:
Please download again the latest development version I changed again something.

itsmegb, 2018-01-31:
I have this same scale, but it doesn't display as 'Health Scale' it gets detected as 'ExcelvanCF369BE', but when i check on BluetoothLE Scanner, its device name shows as 'Electronic Scale'.

I wonder if this is to do with device hardware/firmware versions, my scale is has the following:
* Firmware: v1.0
* Hardware: B0002
* Software: wechat&&SPS_v1.0

@dinki What are the values for your scale? shown under device information?

dinki, 2018-02-01:
@oliexdev I have some definite progress to report.  I was able to scan for devices and find my scale.  I went to the main menu and clicked the bluetooth icon and it attempted and successfully connected to the scale.  I weighed myself with the scale connected to the app but it did not download the data.  I did not get any kind of message at all.

@itsmegb I'm not near my scale but I'll see if I can grab that and post it later.

oliexdev, 2018-02-01:
@dinki I tried to analysed @ec1oud source code and implemented his magic bytes and also set on UUID notification on. Please try again the latest development version. Can you install Android Studio and compile the source code? Do you have any programming experience. Would make the much easier.

@itsmegb That is strange that your scale has a different Bluetooth name. Are you sure you have a 1byone scale?

dinki, 2018-02-04:
@oliexdev Unfortunately that did not work either.  Are there any logs that might shed light on what is actually happening?  In regards to being a programmer, I am not.  I'm much more of a hacker than anything.

oliexdev, 2018-02-05:
@dinki if you don't see any toast message then you don't receive any data from the scale. To figure out what is going on is difficult if you don't have the scale. You have to analyse the scale behaviour via a try and error process. Sorry, I can't do that here. Try out @ec1oud qt program and see if that is working. Maybe it has something to do with the magic bytes!? Maybe the key to success is to understand the magic bytes which are send to the scale.

oliexdev, 2018-02-17:
@dinki maybe you don't see any toast because you have the same problem as #187 .
Try to debug the app in Android Studio.

I will remove this try in the master repository until we get working version.

For reference the latest implementation in the mater repository was [BluetoothOneByone.java](https://github.com/oliexdev/openScale/blob/master/android_app/app/src/main/java/com/health/openscale/core/bluetooth/BluetoothOneByone.java).

dinki, 2018-03-05:
I'm finally getting around to looking at this.  Apologies.  Do you have a guide for debugging/building with Android Studio?  I'm pretty technical but I have never done this before so a general guide would help.  Also, what do I need to install?  Is Android Studio a standalone package or do I need to add certain modules etc?

Thanks again for the help.  Just a little push and I'll be happy to try and experiment on this end.

oliexdev, 2018-03-05:
you could set breakpoints or you could just simple add some `Log.d(...)` statements.

See https://developer.android.com/studio/debug/index.html

dinki, 2018-03-05:
Thanks.. so these would go near the bottom where the variables are assigned?  Like around scaleBtData.setWeight(weight);  ?

oliexdev, 2018-03-05:
wherever you want. For example if you want to inspect what is in the weight variable just set there a breakpoint or write directly above the following code snippet:
```
Log.d("MY VARIABLE", "Weight value :" + weight);
scaleBtData.setWeight(weight); 
```

dinki, 2018-03-06:
Thank you for your help and your patience.  So once I make these changes do I need to build the APK and put it on the phone and try a reading?  I'm guessing these values will be sent to a log file somewhere.  How do I find and read this log file?

oliexdev, 2018-03-06:
see https://developer.android.com/training/basics/firstapp/running-app.html how to run the app.
No the values are not send to a log file it will display in the `logcat tab` (see the bottom tabs) in the Android Studio.

dinki, 2018-03-06:
Oh that's really cool.  Thanks.. I'll give it a shot soon.

dinki, 2018-03-09:
@oliexdev I plugged in my phone last night and did not get what was listed in the guide.  I am not running Android v8.  Far from it.  My phone is an old Samsung S4.  It serves me just fine for my day-to-day needs, but perhaps can't be used with the instructions given?  

Is there another way to test?  Is there a way to perhaps build and APK that can write to a physical file on the phone and view that log file?

simon6162, 2018-04-17:
Sorry from writing on a closed issue. 

[weight_scale_trace.zip](https://github.com/oliexdev/openScale/files/1918995/weight_scale_trace.zip)

I've attached a frontline ble sniffer trace of a transaction from the stock app, you might / might not find it useful. For me the device is names "electronic scale" the name is in the scan response. The connect request is at frame 3037, the scales MAC is  0x010203040769. You are probably better looking at the advert with the manufacturer specific data to make the connection and not the name.

See https://github.com/ec1oud/qt-ble-weightscale/issues/1 for how the magic numbers work. 

You must write the control charismatic, and enable notifications to get the data, the notification are send twice before the unit powers off (no disconnection is seen in the sniffer logs ever!)

erijo, 2018-04-17:
Reopen so it can be looked at.

erijo, 2018-04-17:
What happens if you, using the [dev version](https://github.com/oliexdev/openScale/releases/tag/travis-dev-build), try to connect to the scale @simon6162?

simon6162, 2018-04-17:
It connects just fine for me and pulls back the weight OK - FYI the scales only advertise when you stand on them (which is what most do that i've seen) so you need to do that to find the device. 

The body fat is way off though, which is to be expected given  what i worked out in https://github.com/ec1oud/qt-ble-weightscale/issues/1

From looking at BluetoothOneByone.java - where did magicBytes come from, i've not noticed that in the air trace only magicBytes2

erijo, 2018-04-17:
I only have a Beurer BF710 and that scale can be found without having to stand on it.

Is it possible to post the sniffer logs in some other format? I don't have Frontline ble sniffer.

simon6162, 2018-04-17:
It doesn't look like it does, you can get it from http://www.fte.com/support/download.aspx?demo=ComProbe+NFC&iid=1z&Product=16.5.10527.12085 assuming you are on windows - no need for the drivers if you don't have the sniffer. The web form It doesn't validate details ;)

Just looking at your code and the visfat should be /1 not /10. 
`float visfat = (float)(((weightBytes[11] & 0xFF) & 0xFF)) / 1.0f; // %`

This is my working python if you want to compare....
`weight = int(value[4:6].hex(),16)/10.0
fat    = int(value[6:8].hex(),16)/10.0
bone   = int(value[8:9].hex(),16)/1.0
muscle = int(value[9:11].hex(),16)/10.0
v_fat  = int(value[11:12].hex(),16)/1.0
water  =  int(value[12:14].hex(),16)/10.0
bmr    =  int(value[14:16].hex(),16)`

erijo, 2018-04-18:
@simon6162: the magicBytes seems to come from the btsnoop logs posted by @dinki.

But I'm a little confused now as it seems there are two bluetooth drivers in openScale that look similar: OneByone and ExcelvanCF369BLE. Which one is used for your device @simon6162, i.e. what does it say when you scan for the scale in the settings?

If you test with the latest dev version, that it work for you then?

erijo, 2018-04-18:
@dinki: can you try the latest [dev version](https://github.com/oliexdev/openScale/releases/tag/travis-dev-build) as well? I'm interesting in knowing what the name the device has, which can be seen after scanning for devices in the settings. If it isn't found you may have to stand on it.

Also, if it is found, can you get a measurement from it?

simon6162, 2018-04-19:
Interesting. It shows up as ExcelvanCF369BLE

erijo, 2018-04-19:
Does it work? Do you get the expected values? If not, what do you get?

simon6162, 2018-04-19:
The body fat is 10% under what it should be, i haven't checked the other
measurements. I'll try and grab a sniffer trace over the weekend and work
out the issue.

dinki, 2018-04-19:
@erijo Mine shows up as health scale.  It says connecting but I do not get any readings.  Tried uninstalling and reinstalling and it simply does not work.

erijo, 2018-04-20:
@dinki : please try again with the latest dev version and let me know how it goes.

erijo, 2018-04-21:
Also, @dinki and @simon6162, could you please post the model number/name as written under your scale (if there is a note there) so we can get some clarity in the naming confusion?

dinki, 2018-04-21:
@erijo Good news and bad news.  Good news is that the OS is getting weights now with latest dev version, ya!!  Bad news is it isn't receiving any of the other data (fat %, muscle mass, etc).

I'll try and get the model number for you soon.   Thanks again for working on this.

erijo, 2018-04-21:
@dinki I only added weight support as I didn't know what the expected values for the other measurements were in the logs you posted earlier. To get the other values I need logs where you also include the values for all measurements.

Also note the age, sex, activity level etc. that you enter in the original app. Would be good with logs where you change these settings so we can figure out what the values that the app send means.

dinki, 2018-04-23:
@erijo I'm not sure I follow what you're asking for.  The logs I grabbed originally were independent of the app.

Also, I took a look for the model number but there was no tag on the scale outside of an FTC number:

https://photos.app.goo.gl/pmBbywCVkebhCxsp1

erijo, 2018-04-23:
I need similar logs as you posted earlier but with all the other measurements noted as well (i.e. not just the weight as in the first logs).

I assumed you got the logs while using the original app as some data is sent to the scale to start measuring. To figure this part out you need to use the original app and change settings such as your age, sex, unit, etc. and do a measurement. Then attach logs together with the settings you used so we can figure out what all the bytes mean. Preferably only change one setting at a time.

erijo, 2018-04-24:
You can now take the latest openScale dev version, go to settings, about and enable debug log. Then connect to the scale with openScale and weight yourself. Write down the numbers seen on the scale and provide them together with the log from openScale in this issue.

Might be easier than getting Bluetooth snoop logs from the original app (even if we'll still need them later to reverse engineer the data sent to the scale).

dinki, 2018-04-24:
@erijo This certainly does seem easier.  My scale only shows the weight on it's display.  Don't you need to know body fat/bone density/etc?  How will I get those?  Do I do the debug in OpenScale then weigh again and grab the numbers from the scale's app?

erijo, 2018-04-24:
I have some bad news @dinki. Unfortunately it seems like the scale doesn't provide us with fat, water, etc., but only with weight and an impedance reading. This is then converted by the app to fat, water, etc. and reverse engineer that will be a challenge as it's done by a native (i.e. not java) library in the app.

oliexdev, 2018-04-25:
@erijo I am a bit confused do we talk about two different scale devices? I thought @ec1coud have been already done the reverse engineer part for the other measurements like described in his [notes](https://github.com/ec1oud/qt-ble-weightscale/blob/master/doc/notes.txt)?

erijo, 2018-04-25:
@oliexdev 
It looks like @dinki has a scale with the device name "Health Scale" and @itsmegb, @simon6162 and I guess @ec1oud has one with the device name "Electronic Scale".

The former seems to only send the impedance value which is then converted to body fat etc. by a library from holtek.com (libBodyfat). Which by the way seems to be the same library as is used in the Mi app. The latter on the other hand seems to send all values.

@dinki: it would be good if you could download the latest [dev version](https://github.com/oliexdev/openScale/releases/tag/travis-dev-build), start it and go to settings -> about and enable debug log. Then go to settings -> bluetooth and search for your device and click on it once found. Then do a measurement in openScale. Directly afterwards do the measurement in the original app and post all the values here together with the log from openScale. I'm interesting in seeing how the impedance values map to body fat etc.

@simon6162: if you could do the same that would be great.

simon6162, 2018-04-25:
Will do. Btw the label on the bottom of my scales has no fcc number or
model number

dinki, 2018-04-26:
@erijo Here's the info:

https://www.dropbox.com/s/n0r4eorhmy0lwti/openScale_2018-04-25_20-36.txt?dl=0
https://photos.app.goo.gl/3a0w73WaXoboMCtn1

Thanks for looking at this.

erijo, 2018-04-27:
@dinki: can you also post your age, sex, height and activity level (as entered in the app)? They are all used in the calculations.

dinki, 2018-05-01:
@erijo was that the information you needed?  Thanks

erijo, 2018-05-01:
Yes it was.

I've been trying to reverse engineer the calculations done by the app, but the problem is that they are done by a native library (libBodyfat.so) and my assembler reading skills are not as good as they would need to be to decompile the formula.

I have a small native C++ program that can call the functions in the library and get e.g. the fat measurement. Perhaps the way forward would be to get an existing formula for fat calculation and try to compare the result for different input?

Is there any good formula available?

oliexdev, 2018-05-05:
> Is there any good formula available?

For general formula see https://www.biodyncorp.com/tools/450/calculations.html

Condorello, 2018-05-15:
Hi, i've the same scale with the infamous oem app.... i'm happy to post my data for help the reverse engineering. I'll be back ASAP.

[87 more replies]

closed 2018-10-07 by oliexdev