oliexdev/openScale #162: Is posible change the theme to ligth color?

Nadivar, 2018-01-26:
Hi, I saw that the app showed in ligth color, but in my case is dark.... It´s very difficult reads anything
 could anybody tell me the way to change it?
this image is my actual theme....
[image]
this one is like I want...
[image]

oliexdev, 2018-01-27:
You are probably using another App which change global wide the App theme of your installed Apps. It has nothing to do with openScale. In openScale you can't change the theme it use a default one.

closed 2018-01-27 by oliexdev

Nadivar, 2018-01-27:
Upsss, I don´t know, because it´s the only app shows in dark.
Best regards.
Thanks for answering.

oliexdev, 2018-01-28:
@Nadivar I implemented now a theme selector for the app. Try out the latest development version of openScale.