oliexdev/openScale #164: Change data entry

natrius, 2018-01-27:
Hello, 

thanks for the app. I'm using **Version:** 1.7 (23) via the f-droid store.
**Android:** 7.1.2
Just some workflow changes to make manual entries easier.

- Choose in what order the data has to be entried.
- Add option to automatically open the next entrie field.

**Explanation:**
My scale does show the data pretty fast and has now bluetooth. So i have to look for the data, press the "Add" button, entry the data, press "Ok", search the next "Add" button, and so on.
It would be much easier if i could just "Start Entry" - write data - "OK" - write next data - "OK" - finished

(Additional question: How can i help with translations?)

Greetings

closed 2018-01-27 by oliexdev

erijo, 2018-01-27:
I had the same idea in #81 and I plan on implementing it eventually.

erijo, 2018-02-14:
@natrius: if you haven't found it already, translations can now be done here: https://hosted.weblate.org/projects/openscale/

natrius, 2018-02-16:
@erijo
Thanks, maybe you want to add this information into the readme to the "Contribute" section.

erijo, 2018-02-16:
Good idea, perhaps @oliexdev can add something?