oliexdev/openScale #169: [Suggestion] Import/Export from Google Account

Skyrid3r, 2018-01-30:
Is there anyway that there can be a feature where the data can be imported and exported (optional) from a google account? This would be helpful to easily share the data between multiple devices running the same google account. Also for backup reasons which may be useful.

erijo, 2018-01-30:
The backup reason might be a reason to increase API level to >= 23 as that gives apps the auto backup feature. That one the other hand has the downside of requiring location for Bluetooth to work.

Another idea I've been having is to use the overflow menu on the table view to hold export and import to CSV options. This could then also have options to "share" the CSV file to other applications such as Gmail, Google drive, etc. as is seen in many other apps.

oliexdev, 2018-01-31:
@Skyrid3r openScale won't have any integrated Gmail support in openScale, read the FAQ.

But @erijo second idea could be an option or you could also export the backup files into a Google-Drive or Dropbox folder which is automatically sync. 

An automatic local backup of the database (e.g. once a month) would be also a good idea for openScale.

oliexdev, 2018-02-01:
I implemented a share button for the CSV export. But I removed the abort button for that because it looks strange with three buttons or does anybody have a better idea where to place the share button?

erijo, 2018-02-01:
I think it would fit nicely in an options menu (i.e. the three dots menu in the top right corner). Personally I think export and import should go there as well to leave more room for measurements in the table view.

erijo, 2018-02-14:
As of #188 it's possible to both import from Google drive and export (by using the share option).

So I would say that this can be closed.

erijo, 2018-02-22:
@Skyrid3r: can you test the latest development version (https://github.com/oliexdev/openScale/releases/tag/travis-dev-build) and see if it does what you want. If yes, please close this issue, if no please let us know what's missing.

Skyrid3r, 2018-02-22:
Mind sharing where this export function is located? Latest development build is from Jan, 2. I can't find it anywhere. Not even in the 3 dot/stripe menu.

erijo, 2018-02-22:
I should also mention that for export to e.g. google drive you have to select "Share" in the menu.

Skyrid3r, 2018-02-22:
I don't have any share option in the menu though. Using 1.7 dev. Is it because I'm using a different language?

erijo, 2018-02-22:
In the Table view there is an icon with two arrows, one pointing up and the other down. If you click this, do you mean that you only see Import and Export, but not Share?

Skyrid3r, 2018-02-23:
I found it, but I have to admit this is one weird location to put a share button. If I couldn't find it easily, I doubt others will. The arrow buttons merge me think of a ascending and descending filter. 

Can't the share function be in the backup menu?

erijo, 2018-02-23:
I actually agree with you that it's not the best place for the menu. Personally I think they should be in the toolbar, just to the left of the + button (in a three dots menu commonly seen in Android apps). It could then be visible on all views.

What do you think @oliexdev?

oliexdev, 2018-02-23:
hmmm ok then change it but this icon was the import/export [material icon](https://material.io/icons/#ic_import_export) from Android.

closed 2018-02-24 by oliexdev