oliexdev/openScale #171: Python port for Mi Scale 2

pfiol, 2018-01-31:
Hi,

I would like to use your code as a reference to develop a Mi Scale 2 python client. 
There's already a similar client for the Mi Scale v1 here (https://github.com/chaeplin/Xiaomi_scale_scan/blob/master/Xiaomi_scale_scan.py), but it doesn't work with Mi Scale v2.

I was wondering, if you have any documentation about how to read the weight and other information out of the Mi Scale 2. My knowledge on JAVA is limited, and is not easy to me trace what you're doing on the BluetoothMiScale2.java file. All the information you could give me would be very helpful! 

Thanks a lot! 

Pedro

oliexdev, 2018-01-31:
The source code is really self explaining. 
Also it nearly the same as the Mi scale v1. There I wrote some information https://github.com/oliexdev/openScale/wiki/Xiaomi-Bluetooth-Mi-Scale

Currently we can only read the weight and the impedance value out of the mi scale v2. Other information like body water, muscle percentage and so on are intransparent calculated on the vendors app.

closed 2018-01-31 by oliexdev

Wingjam, 2018-11-16:
Hi, 

Maybe you can port https://github.com/Wingjam/ReverseMiScale written in Python from v1 to v2?

Jonathan