oliexdev/openScale #173: Crash when searching for Bluetooth device with latest dev

erijo, 2018-02-01:
Build version: 1.7-dev_531d38be 
Current date: 2018-02-01 18:33:58 
Device: LGE Nexus 5X 
 
Stack trace:  
java.lang.RuntimeException: Error receiving broadcast Intent { act=android.bluetooth.device.action.FOUND flg=0x10 (has extras) } in com.health.openscale.gui.preferences.BluetoothPreferences$1@3b850a4
	at android.app.LoadedApk$ReceiverDispatcher$Args.lambda$-android_app_LoadedApk$ReceiverDispatcher$Args_52497(LoadedApk.java:1323)
	at android.app.-$Lambda$aS31cHIhRx41653CMnd4gZqshIQ.$m$7(Unknown Source:4)
	at android.app.-$Lambda$aS31cHIhRx41653CMnd4gZqshIQ.run(Unknown Source:39)
	at android.os.Handler.handleCallback(Handler.java:790)
	at android.os.Handler.dispatchMessage(Handler.java:99)
	at android.os.Looper.loop(Looper.java:164)
	at android.app.ActivityThread.main(ActivityThread.java:6494)
	at java.lang.reflect.Method.invoke(Native Method)
	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:438)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:807)
Caused by: java.lang.NullPointerException: Attempt to invoke virtual method 'java.lang.String java.lang.String.toLowerCase()' on a null object reference
	at com.health.openscale.core.bluetooth.BluetoothCommunication.checkDeviceName(BluetoothCommunication.java:174)
	at com.health.openscale.gui.preferences.BluetoothPreferences$1.onReceive(BluetoothPreferences.java:97)
	at android.app.LoadedApk$ReceiverDispatcher$Args.lambda$-android_app_LoadedApk$ReceiverDispatcher$Args_52497(LoadedApk.java:1313)
	... 9 more

User actions: 
2018-02-01 18:31:39: MainActivity resumed
2018-02-01 18:31:56: MainActivity paused
2018-02-01 18:31:56: DataEntryActivity created
2018-02-01 18:31:57: DataEntryActivity resumed
2018-02-01 18:32:24: DataEntryActivity paused
2018-02-01 18:32:24: MainActivity resumed
2018-02-01 18:32:25: DataEntryActivity destroyed
2018-02-01 18:32:28: MainActivity paused
2018-02-01 18:32:28: DataEntryActivity created
2018-02-01 18:32:28: DataEntryActivity resumed
2018-02-01 18:32:30: DataEntryActivity paused
2018-02-01 18:32:30: MainActivity resumed
2018-02-01 18:32:30: DataEntryActivity destroyed
2018-02-01 18:33:07: MainActivity paused
2018-02-01 18:33:08: MainActivity destroyed
2018-02-01 18:33:11: MainActivity created
2018-02-01 18:33:12: MainActivity resumed
2018-02-01 18:33:16: MainActivity paused
2018-02-01 18:33:16: SettingsActivity created
2018-02-01 18:33:16: SettingsActivity resumed

oliexdev, 2018-02-01:
should be fixed now. Please test it.

erijo, 2018-02-01:
b21430d9 seems to have fixed the problem.

closed 2018-02-01 by erijo