oliexdev/openScale #175: memory leak occures on a screen orientation change

oliexdev, 2018-02-04:
A memory leak occurs when the screen orientation is changed. Some fragments or variables are not cleaned up on destroying the activity. Right now, I didn't find out who was causing the trouble.

It would be great If anybody can look into it.

erijo, 2018-02-08:
Could it be the fragmentList in OpenScale that holds on to fragments after they have been destroyed?

oliexdev, 2018-02-09:
I doubt that because the fragment list is inside a singleton class.
Maybe the FragmentManager are not cleared up correctly?

erijo, 2018-02-09:
As the list holds a reference to the fragment the GC can't free the memory even if the fragment is released by the manager. So even if not technically a memory leak, it will still consume memory each time a fragment is created.

oliexdev, 2018-02-09:
but even if I clear the fragmentList on MainActivity destroy the memory usage is increasing on rotation.

commit 5c1ef0cf (2018-02-26): Fix fragment leak

By forcing transactions to run before creating new fragment we don't
end up with two fragments due to selectDrawerItem being called twice
during creation.

Fixes #175
closed 2018-02-26 by oliexdev