oliexdev/openScale #183: Permission request problem

erijo, 2018-02-10:
I tried the latest dev version: first I removed the storage permission, then I pressed export and picked deny => no toast message

Then I clicked export again and the toast message was shown a short while before the permission dialog appeared.

If I then press allow I don't get any toast message about the export being done. I have to press export again to get that.

oliexdev, 2018-02-10:
You are right somehow the method `onRequestPermissionsResult` in the fragment are not being called. I tried [this](https://stackoverflow.com/questions/35989288/onrequestpermissionsresult-not-being-called-in-fragment-if-defined-in-both-fragm) but it didn't work.

oliexdev, 2018-02-11:
resolved with commit https://github.com/oliexdev/openScale/commit/8ec7eac2887baf19623b442477f2d0722bed1729 but unfortunately due a hack otherwise an API level > 23 is required.

closed 2018-02-11 by oliexdev