oliexdev/openScale #185: Add support for iHealth HS3 Wireless Scale

JohnLines, 2018-02-12:
I recently purchased this scale, which is a fairly basic scale. it claims to measure Weight and BMI, but the BMI is calculated on their app, using height and age plus the weight.

I have taken some btsnoop_hci.log captures, but can not find my weight in kg in hex in the captures, or weight in lb or st, which are the other options on the scales.

I have also looked for the strings AT and OK (and 0x6584 and 0x4F4B), without success.

I will keep working on it - my current results are at [https://johnlines.github.io/openScale/](https://johnlines.github.io/openScale/)

oliexdev, 2018-02-12:
With a quick peek into your logs your scale use the [L2CAP](https://en.wikipedia.org/wiki/List_of_Bluetooth_protocols#Logical_link_control_and_adaptation_protocol_(L2CAP)) protocol which is really special use for a scale.

JohnLines, 2018-02-13:
I had hoped that a simple (weight only)  and early version scale would be easier to decode than a more modern and complex one with more parameters - looks as if I will need to be looking deeper into the RFCOMM packets

JohnLines, 2018-02-14:
Have found time packets, but not weight data yet - updated [https://johnlines.github.io/openScale/](https://johnlines.github.io/openScale/) with updated information. The data seems to be in SPP packets

oliexdev, 2018-02-17:
I looked into your log files. Unfortunately, I also didn't find any data bytes about the weight. The bytes for 77.1 kg and the 77.2 kg should differ only in 1 decimal but comparing these log files I also didn't find any close values to each other.

JohnLines, 2018-02-17:
Looks as if I have some research into Bluetooth on Android to do !

JohnLines, 2018-03-04:
Getting there - I am returning valid data to addScaleData, but currently getting 3 results per weighing. I will look into suppressing. Also, although the scales know the date I am returning the date from the phone.

erijo, 2018-03-08:
For reference: PR #221 implements this.

closed 2018-03-11 by oliexdev