oliexdev/openScale #187: Finally working with Hesley scale

pkscwc, 2018-02-13:
Dear Olie,

Debugging took some time. 

Strange error!! The toast does not work with Marshmallow!!! 

The working java file is attached alongwith screenshots. A number of them. 

We can add it to our list of scales. It is a good one. 
[image]
[image]
[image]
[image]
[image]
[BluetoothHesley_java_13Feb2018.java.txt](https://github.com/oliexdev/openScale/files/1720995/BluetoothHesley_java_13Feb2018.java.txt)

Thanks.

oliexdev, 2018-02-14:
great I change the code with commit https://github.com/oliexdev/openScale/commit/0050ec46ff9e24ca500afa79f487afc2ef6a0aaa

closed 2018-02-14 by oliexdev

pkscwc, 2018-02-14:
> Thanks a lot. I
> great I change the code with commit 0050ec4

Do we have APK also updated with Healey scale.

pkscwc, 2018-02-14:
Thanks a lot.

erijo, 2018-02-14:
See https://github.com/oliexdev/openScale/releases/tag/travis-dev-build for an APK built from latest master.

pkscwc, 2018-02-21:
Dear Olie,
Can we have only one reading for every step on scale. I am attaching a csv
file for clarifying the issue. I am getting 100s of reading for each time I
step on the scale.

oliexdev, 2018-02-21:
Sure modify the BluetoothHesley.java file so that only the last send data is valid.

pkscwc, 2018-02-22:
:-)
Is it through BluetoothHesley.java file?
I cannot read it in this file. Isn't there any separate java file for data?

pkscwc, 2018-02-24:
Dear Olie,

Any hint!
I would work on that further.

Thanks.

oliexdev, 2018-02-24:
try to avoid that the method `parseBytes(...)` is always called in the BluetoothHesley.java file call it just once time if you receive a new weight.

pkscwc, 2018-02-26:
OK.
Will try and let you know.