oliexdev/openScale #196: Carry over additional measurements not present in all records

lenucksi, 2018-02-19:
# Use Case:
As a user I would like to be able to realize the following use case:
- Daily use of one of the supported BTLE scales using them to measure whatever they provide. (Schedule A)
- Weekly or less than daily measurement of other manual metrics such as e.g. waist size (Schedule B) 
- I would like to have the current measurement of schedule B to be carried over / applied to measurements of Schedule A until the next measurement of schedule B is available.
- Thus, all records would have data for automatic and manual measurements present. 

# Rationale / Current state
- Currently I have to generate a new measurement that gets the latest automatic / schedule A measurements and add my manual measurements.
- Upon looking at the diagrams this leads to wrong / weird display of data as it appears that the 0 entries for the additional / manual measurements on schedule A records are included in the diagramming / calculation process leading to wrong / undesirable results.

oliexdev, 2018-02-20:
ok good idea but I need to figure out how to I distinguish between the values that comes from the scale and that are carried over from the last measurement because it is always different depends on the Bluetooth scale.

erijo, 2018-02-20:
Instead of using an empty ScaleMeasurement as the base in the Bluetooth class, perhaps simply use the latest measurement. Then it would be similar to the data entry.

oliexdev, 2018-02-21:
Do you mean to use the latest measurements in the ScaleMeasurement constructor?
But this class doesn't know the latest measurement and also shouldn't know it. But maybe we find another place to set the latest measurements.

erijo, 2018-02-22:
No, not that. But the Bluetooth classes create a new ScaleMeasurement that they fill with data from the scale. My idea was that they instead could use a clone of the last measurement. One way to do it would be if the one calling the Bluetooth classes would provide the ScaleMeasurement instance that they should use as a base.

oliexdev, 2018-02-22:
hmm... I am not still not sure what you mean sorry. Do you mean to have a getter method in BluetoothCommunication.java to get the latest measurement?
Maybe you can create a example PR what you have in mind?

erijo, 2018-02-24:
I thought some more about this and I don't think my initial suggestion is all that good. If we copy over e.g. fat from the previous measurement we will hide a problem if e.g. the scale for some reason stop providing the fat measurement (or if the user switches to a different scale).

Instead we should just copy over waist and hip since these are unlikely to come from the bluetooth scale.

lenucksi, 2018-02-27:
Validated the feature that was provided with PR #205 using a SBF-75. From what I can see, it works as expected, thanks. 
I guess this issue can be closed then? @erijo  @oliexdev

oliexdev, 2018-02-27:
thanks for your testing. Yes if it working this issue can be closed :)

closed 2018-02-27 by oliexdev