oliexdev/openScale #202: Crash while clicking around in graph

erijo, 2018-02-24:
Got the following crash while clicking around in the graph with regression line active (degree 10). Don't have time just now to figure out exactly what I did, but perhaps it can be fixed anyway.

Build version: 1.7-dev_0cba7961 
Current date: 2018-02-24 09:53:51 
Device: LGE Nexus 5X 
 
Stack trace:  
java.lang.ArrayIndexOutOfBoundsException: length=11; index=11
	at com.health.openscale.core.utils.PolynomialFitter.gj_swap(PolynomialFitter.java:186)
	at com.health.openscale.core.utils.PolynomialFitter.gj_echelonize(PolynomialFitter.java:156)
	at com.health.openscale.core.utils.PolynomialFitter.getBestFit(PolynomialFitter.java:118)
	at com.health.openscale.gui.fragments.GraphFragment.generateLineData(GraphFragment.java:425)
	at com.health.openscale.gui.fragments.GraphFragment.generateGraphs(GraphFragment.java:526)
	at com.health.openscale.gui.fragments.GraphFragment.access$700(GraphFragment.java:73)
	at com.health.openscale.gui.fragments.GraphFragment$2.onClick(GraphFragment.java:158)
	at android.view.View.performClick(View.java:6294)
	at android.view.View$PerformClick.run(View.java:24770)
	at android.os.Handler.handleCallback(Handler.java:790)
	at android.os.Handler.dispatchMessage(Handler.java:99)
	at android.os.Looper.loop(Looper.java:164)
	at android.app.ActivityThread.main(ActivityThread.java:6494)
	at java.lang.reflect.Method.invoke(Native Method)
	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:438)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:807)

User actions: 
2018-02-24 09:05:18: MainActivity resumed
2018-02-24 09:05:31: MainActivity paused
2018-02-24 09:05:36: MainActivity resumed
2018-02-24 09:05:43: MainActivity paused
2018-02-24 09:05:51: MainActivity resumed
2018-02-24 09:06:10: MainActivity paused
2018-02-24 09:06:10: DataEntryActivity created
2018-02-24 09:06:11: DataEntryActivity resumed
2018-02-24 09:06:24: DataEntryActivity paused
2018-02-24 09:06:47: DataEntryActivity resumed
2018-02-24 09:07:02: DataEntryActivity paused
2018-02-24 09:48:27: DataEntryActivity resumed
2018-02-24 09:48:29: DataEntryActivity paused
2018-02-24 09:48:29: MainActivity resumed
2018-02-24 09:48:30: DataEntryActivity destroyed
2018-02-24 09:48:31: MainActivity paused
2018-02-24 09:48:31: SettingsActivity created
2018-02-24 09:48:31: SettingsActivity resumed
2018-02-24 09:51:07: SettingsActivity paused
2018-02-24 09:51:07: MainActivity destroyed
2018-02-24 09:51:07: MainActivity created
2018-02-24 09:51:07: MainActivity resumed
2018-02-24 09:51:08: SettingsActivity destroyed

erijo, 2018-02-24:
The crash happens with both week and month view unchecked and the going from 2018 to 2017 with the following data:

"bone","comment","dateTime","fat","hip","lbw","muscle","waist","water","weight"
0.0,"","13.12.2018 13:20",19.3,0.0,0.0,42.5,0.0,59.4,85.2
0.0,"","13.12.2018 13:19",19.3,0.0,0.0,42.5,0.0,59.4,85.2
0.0,"","13.12.2018 13:18",19.3,0.0,0.0,42.5,0.0,59.4,85.2
0.0,"","13.12.2018 13:17",19.3,0.0,0.0,42.5,0.0,59.4,85.2
0.0,"","13.12.2018 13:16",19.3,0.0,0.0,42.5,0.0,59.4,85.2
0.0,"","09.12.2018 19:40",20.3,0.0,0.0,42.2,0.0,58.7,86.9
0.0,"","09.12.2018 19:39",20.3,0.0,0.0,42.2,0.0,58.7,86.9
0.0,"","09.12.2018 19:38",20.3,0.0,0.0,42.2,0.0,58.7,86.9
0.0,"","09.12.2018 19:37",20.3,0.0,0.0,42.2,0.0,58.7,86.9
0.0,"","09.12.2018 19:36",20.3,0.0,0.0,42.2,0.0,58.7,86.9
0.0,"","08.12.2018 20:32",19.9,0.0,0.0,42.2,0.0,59.0,86.8
0.0,"","08.12.2018 20:31",19.9,0.0,0.0,42.2,0.0,59.0,86.8
0.0,"","08.12.2018 20:30",19.9,0.0,0.0,42.2,0.0,59.0,86.8
0.0,"","08.12.2018 20:29",19.9,0.0,0.0,42.2,0.0,59.0,86.8
0.0,"","08.12.2018 20:28",19.9,0.0,0.0,42.2,0.0,59.0,86.8
0.0,"","05.12.2018 18:25",20.3,0.0,0.0,42.2,0.0,58.7,86.7
0.0,"","05.12.2018 18:24",20.3,0.0,0.0,42.2,0.0,58.7,86.7
0.0,"","05.12.2018 18:23",20.3,0.0,0.0,42.2,0.0,58.7,86.7
0.0,"","05.12.2018 18:22",20.3,0.0,0.0,42.2,0.0,58.7,86.7
0.0,"","05.12.2018 18:21",20.3,0.0,0.0,42.2,0.0,58.7,86.7
0.0,"","04.08.2018 08:12",21.2,0.0,0.0,41.5,0.0,58.0,89.7
0.0,"","04.08.2018 08:11",21.2,0.0,0.0,41.5,0.0,58.0,89.7
0.0,"","04.08.2018 08:10",21.2,0.0,0.0,41.5,0.0,58.0,89.7
0.0,"","04.08.2018 08:09",21.2,0.0,0.0,41.5,0.0,58.0,89.7
0.0,"","04.08.2018 08:08",21.2,0.0,0.0,41.5,0.0,58.0,89.7
0.0,"","03.08.2018 05:21",26.4,0.0,0.0,41.6,0.0,54.6,89.0
0.0,"","03.08.2018 05:20",26.4,0.0,0.0,41.6,0.0,54.6,89.0
0.0,"","03.08.2018 05:19",26.4,0.0,0.0,41.6,0.0,54.6,89.0
0.0,"","03.08.2018 05:18",26.4,0.0,0.0,41.6,0.0,54.6,89.0
0.0,"","03.08.2018 05:17",26.4,0.0,0.0,41.6,0.0,54.6,89.0
0.0,"","02.08.2018 07:36",25.0,0.0,0.0,41.7,0.0,55.6,88.8
0.0,"","02.08.2018 07:35",25.0,0.0,0.0,41.7,0.0,55.6,88.8
0.0,"","02.08.2018 07:34",25.0,0.0,0.0,41.7,0.0,55.6,88.8
0.0,"","02.08.2018 07:33",25.0,0.0,0.0,41.7,0.0,55.6,88.8
0.0,"","02.08.2018 07:32",25.0,0.0,0.0,41.7,0.0,55.6,88.8
0.0,"","31.07.2018 04:43",29.2,0.0,0.0,41.6,0.0,52.8,89.1
0.0,"","31.07.2018 04:42",29.2,0.0,0.0,41.6,0.0,52.8,89.1
0.0,"","31.07.2018 04:41",29.2,0.0,0.0,41.6,0.0,52.8,89.1
0.0,"","31.07.2018 04:40",29.2,0.0,0.0,41.6,0.0,52.8,89.1
0.0,"","31.07.2018 04:39",29.2,0.0,0.0,41.6,0.0,52.8,89.1
0.0,"","18.07.2018 07:58",22.1,0.0,0.0,41.2,0.0,57.4,91.3
0.0,"","18.07.2018 07:57",22.1,0.0,0.0,41.2,0.0,57.4,91.3
0.0,"","18.07.2018 07:56",22.1,0.0,0.0,41.2,0.0,57.4,91.3
0.0,"","18.07.2018 07:55",22.1,0.0,0.0,41.2,0.0,57.4,91.3
0.0,"","18.07.2018 07:54",22.1,0.0,0.0,41.2,0.0,57.4,91.3
0.0,"","12.07.2018 07:18",21.9,0.0,0.0,41.3,0.0,57.6,91.1
0.0,"","12.07.2018 07:17",21.9,0.0,0.0,41.3,0.0,57.6,91.1
0.0,"","12.07.2018 07:16",21.9,0.0,0.0,41.3,0.0,57.6,91.1
0.0,"","12.07.2018 07:15",21.9,0.0,0.0,41.3,0.0,57.6,91.1
0.0,"","12.07.2018 07:14",21.9,0.0,0.0,41.3,0.0,57.6,91.1
0.0,"","16.06.2018 05:20",25.3,0.0,0.0,41.5,0.0,55.4,89.5
0.0,"","16.06.2018 05:19",25.3,0.0,0.0,41.5,0.0,55.4,89.5
0.0,"","16.06.2018 05:18",25.3,0.0,0.0,41.5,0.0,55.4,89.5
0.0,"","16.06.2018 05:17",25.3,0.0,0.0,41.5,0.0,55.4,89.5
0.0,"","16.06.2018 05:16",25.3,0.0,0.0,41.5,0.0,55.4,89.5
0.0,"","15.06.2018 05:38",26.3,0.0,0.0,41.4,0.0,54.7,90.1
0.0,"","15.06.2018 05:37",26.3,0.0,0.0,41.4,0.0,54.7,90.1
0.0,"","15.06.2018 05:36",26.3,0.0,0.0,41.4,0.0,54.7,90.1
0.0,"","15.06.2018 05:35",26.3,0.0,0.0,41.4,0.0,54.7,90.1
0.0,"","15.06.2018 05:34",26.3,0.0,0.0,41.4,0.0,54.7,90.1
0.0,"","12.06.2018 05:40",26.4,0.0,0.0,41.4,0.0,54.6,90.3
0.0,"","12.06.2018 05:39",26.4,0.0,0.0,41.4,0.0,54.6,90.3
0.0,"","12.06.2018 05:38",26.4,0.0,0.0,41.4,0.0,54.6,90.3
0.0,"","12.06.2018 05:37",26.4,0.0,0.0,41.4,0.0,54.6,90.3
0.0,"","12.06.2018 05:36",26.4,0.0,0.0,41.4,0.0,54.6,90.3
0.0,"","10.06.2018 04:26",22.3,0.0,0.0,41.3,0.0,57.3,90.8
0.0,"","10.06.2018 04:25",22.3,0.0,0.0,41.3,0.0,57.3,90.8
[... 164 lines cut ...]
0.0,"","06.01.2018 10:36",19.7,0.0,0.0,42.4,0.0,59.1,85.5
0.0,"","06.01.2018 10:35",19.7,0.0,0.0,42.4,0.0,59.1,85.5
0.0,"","06.01.2018 10:34",19.7,0.0,0.0,42.4,0.0,59.1,85.5
0.0,"","05.01.2018 08:29",26.1,0.0,0.0,42.4,0.0,54.9,85.6
0.0,"","05.01.2018 08:28",26.1,0.0,0.0,42.4,0.0,54.9,85.6
0.0,"","05.01.2018 08:27",26.1,0.0,0.0,42.4,0.0,54.9,85.6
0.0,"","05.01.2018 08:26",26.1,0.0,0.0,42.4,0.0,54.9,85.6
0.0,"","05.01.2018 08:25",26.1,0.0,0.0,42.4,0.0,54.9,85.6
15.1,"","05.01.2018 00:26",23.3,0.0,0.0,39.1,0.0,56.1,102.0
0.0,"","02.01.2018 18:10",19.8,0.0,0.0,42.3,0.0,59.1,86.3
0.0,"","02.01.2018 18:09",19.8,0.0,0.0,42.3,0.0,59.1,86.3
0.0,"","02.01.2018 18:08",19.8,0.0,0.0,42.3,0.0,59.1,86.3
0.0,"","02.01.2018 18:07",19.8,0.0,0.0,42.3,0.0,59.1,86.3
0.0,"","02.01.2018 18:06",19.8,0.0,0.0,42.3,0.0,59.1,86.3
15.1,"","30.12.2017 00:27",23.1,0.0,0.0,39.2,0.0,56.2,101.9
15.1,"","20.12.2017 21:08",23.0,0.0,0.0,39.3,0.0,56.3,101.4
14.8,"","06.12.2017 21:13",22.6,0.0,0.0,39.5,0.0,56.6,100.5
14.9,"","29.11.2017 22:27",22.8,0.0,0.0,39.5,0.0,56.5,100.9
14.9,"","26.11.2017 09:47",22.9,0.0,0.0,39.4,0.0,56.3,101.2
15.1,"","15.11.2017 22:15",23.5,0.0,0.0,39.1,0.0,55.9,102.9
15.1,"","12.11.2017 23:09",23.8,0.0,0.0,39.0,0.0,55.7,103.6
15.2,"","06.11.2017 22:34",24.0,0.0,0.0,38.9,0.0,55.6,104.0
15.0,"","23.10.2017 21:46",23.4,0.0,0.0,39.2,0.0,56.0,102.6
15.0,"","13.10.2017 20:19",23.4,0.0,0.0,39.2,0.0,56.0,102.6
15.1,"","09.10.2017 20:33",23.4,0.0,0.0,39.2,0.0,56.0,102.7

oliexdev, 2018-02-24:
thx should be fixed with commit https://github.com/oliexdev/openScale/commit/6854b9704e93c862657f99528077de203f21525c
After refactoring this class the regression line was not probably calculated because the time was not set and I fixed also some other side effects.

closed 2018-02-24 by oliexdev