oliexdev/openScale #206: Crash when looking for Bluetooth device

Skyrid3r, 2018-02-25:
Build version: 1.7-dev_a8859200 
Current date: 2018-02-25 11:51:31 
Device: Samsung SM-T580 
 
Stack trace:  
java.lang.IllegalArgumentException: Receiver not registered: com.health.openscale.core.bluetooth.BluetoothCommunication$1@7c66b8d
	at android.app.LoadedApk.forgetReceiverDispatcher(LoadedApk.java:1054)
	at android.app.ContextImpl.unregisterReceiver(ContextImpl.java:1376)
	at android.content.ContextWrapper.unregisterReceiver(ContextWrapper.java:668)
	at com.health.openscale.core.bluetooth.BluetoothCommunication.stopSearching(BluetoothCommunication.java:466)
	at com.health.openscale.core.OpenScale.stopSearchingForBluetooth(OpenScale.java:449)
	at com.health.openscale.gui.MainActivity.invokeSearchBluetoothDevice(MainActivity.java:456)
	at com.health.openscale.gui.MainActivity.onOptionsItemSelected(MainActivity.java:388)
	at android.app.Activity.onMenuItemSelected(Activity.java:3383)
	at android.support.v4.app.FragmentActivity.onMenuItemSelected(FragmentActivity.java:380)
	at android.support.v7.app.AppCompatActivity.onMenuItemSelected(AppCompatActivity.java:195)
	at android.support.v7.view.WindowCallbackWrapper.onMenuItemSelected(WindowCallbackWrapper.java:108)
	at android.support.v7.view.WindowCallbackWrapper.onMenuItemSelected(WindowCallbackWrapper.java:108)
	at android.support.v7.app.ToolbarActionBar$2.onMenuItemClick(ToolbarActionBar.java:63)
	at android.support.v7.widget.Toolbar$1.onMenuItemClick(Toolbar.java:203)
	at android.support.v7.widget.ActionMenuView$MenuBuilderCallback.onMenuItemSelected(ActionMenuView.java:780)
	at android.support.v7.view.menu.MenuBuilder.dispatchMenuItemSelected(MenuBuilder.java:822)
	at android.support.v7.view.menu.MenuItemImpl.invoke(MenuItemImpl.java:171)
	at android.support.v7.view.menu.MenuBuilder.performItemAction(MenuBuilder.java:973)
	at android.support.v7.view.menu.MenuBuilder.performItemAction(MenuBuilder.java:963)
	at android.support.v7.widget.ActionMenuView.invokeItem(ActionMenuView.java:624)
	at android.support.v7.view.menu.ActionMenuItemView.onClick(ActionMenuItemView.java:150)
	at android.view.View.performClick(View.java:6261)
	at android.widget.TextView.performClick(TextView.java:11159)
	at android.view.View$PerformClick.run(View.java:23752)
	at android.os.Handler.handleCallback(Handler.java:751)
	at android.os.Handler.dispatchMessage(Handler.java:95)
	at android.os.Looper.loop(Looper.java:154)
	at android.app.ActivityThread.main(ActivityThread.java:6776)
	at java.lang.reflect.Method.invoke(Native Method)
	at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:1496)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1386)

User actions: 
2018-02-25 11:50:23: MainActivity resumed
2018-02-25 11:50:30: MainActivity paused
2018-02-25 11:50:30: SettingsActivity created
2018-02-25 11:50:30: SettingsActivity resumed
2018-02-25 11:50:44: SettingsActivity paused
2018-02-25 11:50:44: MainActivity resumed
2018-02-25 11:50:44: SettingsActivity destroyed
2018-02-25 11:50:45: MainActivity paused
2018-02-25 11:50:46: MainActivity destroyed
2018-02-25 11:51:02: MainActivity created
2018-02-25 11:51:03: MainActivity resumed
2018-02-25 11:51:11: MainActivity paused
2018-02-25 11:51:11: DataEntryActivity created
2018-02-25 11:51:11: DataEntryActivity resumed
2018-02-25 11:51:13: DataEntryActivity paused
2018-02-25 11:51:13: MainActivity resumed
2018-02-25 11:51:13: DataEntryActivity destroyed

-------

Happens quite often lately. Health device is SBF75

oliexdev, 2018-02-25:
I can't reproduce this error but I implemented a more robust function with commit https://github.com/oliexdev/openScale/commit/e1d74bf94f9bae9b778a797eb25d2763a26371a7
Please try it out.

oliexdev, 2018-03-07:
I guess it is fixed...

closed 2018-03-07 by oliexdev