oliexdev/openScale #212: Bug: Endless Connection to device

Teyro, 2018-03-05:
Hi there,

First of all i want to thank you for your great work! Got this app from f-droid and i am very happy with it! Keep going!

But i had one small Bug since the lastest version of openscale (on a older version it didnt happen far as a know). I am useing openscale with the medisana bs 444 works great but every time a got new data via bluetooth it loops .... Got a toast massage "There a data with the same date and time existing" and it loops and loops need to force close openscale every time....

Sorry for my bad english my native language is german... :(

oliexdev, 2018-03-05:
Did you receive these messages if you stand on the scale?
Try to weight yourself and then after you are done connect to the scale with openScale

hirnsushi, 2018-03-06:
I can confirm this behaviour for my bs 444 as well.
Has been happening since 1.7.
If you wait long enough (1min or so) the toasts will stop, no need to fc it.
Already tried it with a clean db, that worked for the first sync, but as soon as the old measurements are synced, the toasts reappear. Then I got used to it… 😃

erijo, 2018-03-06:
Could it be that openScale always fetches all measurements from the scale?

oliexdev, 2018-03-06:
As @erijo already assumed the scales fetches all measurements from the scale. I also didn't change anything in the bs 444 source code but I added that a toast message is displayed. With commit https://github.com/oliexdev/openScale/commit/ebafa65becbd5949ee3992f98b7f253db911d140 I set the silent mode on for new Bluetooth messages (so no toast messages are displayed) but still as like in the previous version all measurements are fetched from the scale.

closed 2018-03-06 by oliexdev