oliexdev/openScale #216: Bodyfat / -water in kg uses the wrong scale

hirnsushi, 2018-03-07:
Not sure if this can be easily solved, the different values would need to be calculated for every new measurement.
[image]

erijo, 2018-03-07:
I don't think I understand the issue. Could you please elaborate?

hirnsushi, 2018-03-07:
If you look at body fat for example the arrow suggest I'm on the lower end of the green zone.
That would be correct if my weight was exactly 100kg, it isn't.
For comparison the same measurement in percent:
[image]

erijo, 2018-03-07:
Ah, now I see.

I can look into this if you want @oliexdev.

oliexdev, 2018-03-07:
@erijo sure go ahead.

closed 2018-03-10 by oliexdev

erijo, 2018-03-11:
@hirnsushi: please try the latest [dev build](https://github.com/oliexdev/openScale/releases/tag/travis-dev-build) and see if it looks better now.

hirnsushi, 2018-03-13:
Works great, thank you 👍

commit 5e058f4e (2018-03-10): Convert evaulation result to absolute weight if needed

Fixes #216