oliexdev/openScale #218: Crash report

alekksander, 2018-03-08:
Build version: 1.7.1 
Current date: 2018-03-08 12:07:33 
Device: BullittGroupLimited S40 
 
Stack trace:  
java.lang.RuntimeException: Unable to start activity ComponentInfo{com.health.openscale/com.health.openscale.gui.MainActivity}: java.lang.NullPointerException: Attempt to invoke virtual method 'int com.health.openscale.core.datatypes.ScaleUser.getId()' on a null object reference
	at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:2325)
	at android.app.ActivityThread.handleLaunchActivity(ActivityThread.java:2387)
	at android.app.ActivityThread.access$800(ActivityThread.java:151)
	at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1303)
	at android.os.Handler.dispatchMessage(Handler.java:102)
	at android.os.Looper.loop(Looper.java:135)
	at android.app.ActivityThread.main(ActivityThread.java:5255)
	at java.lang.reflect.Method.invoke(Native Method)
	at java.lang.reflect.Method.invoke(Method.java:372)
	at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:905)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:700)
Caused by: java.lang.NullPointerException: Attempt to invoke virtual method 'int com.health.openscale.core.datatypes.ScaleUser.getId()' on a null object reference
	at com.health.openscale.core.OpenScale.getTupleScaleData(OpenScale.java:226)
	at com.health.openscale.gui.fragments.OverviewFragment.updateOnView(OverviewFragment.java:167)
	at com.health.openscale.core.OpenScale.registerFragment(OpenScale.java:475)
	at com.health.openscale.gui.fragments.OverviewFragment.onCreateView(OverviewFragment.java:146)
	at android.support.v4.app.Fragment.performCreateView(Fragment.java:2261)
	at android.support.v4.app.FragmentManagerImpl.moveToState(FragmentManager.java:1419)
	at android.support.v4.app.FragmentManagerImpl.moveFragmentToExpectedState(FragmentManager.java:1750)
	at android.support.v4.app.FragmentManagerImpl.moveToState(FragmentManager.java:1819)
	at android.support.v4.app.FragmentManagerImpl.dispatchStateChange(FragmentManager.java:3227)
	at android.support.v4.app.FragmentManagerImpl.dispatchActivityCreated(FragmentManager.java:3181)
	at android.support.v4.app.FragmentController.dispatchActivityCreated(FragmentController.java:192)
	at android.support.v4.app.FragmentActivity.onStart(FragmentActivity.java:572)
	at android.support.v7.app.AppCompatActivity.onStart(AppCompatActivity.java:177)
	at android.app.Instrumentation.callActivityOnStart(Instrumentation.java:1236)
	at android.app.Activity.performStart(Activity.java:6052)
	at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:2288)
	... 10 more

User actions:

erijo, 2018-03-08:
I did an attempt at fixing this. But to know if this is the real fix, could you please provide some context on what you did prior to the crash? Do you have user(s) registered? Did the previous version work?

alekksander, 2018-03-09:
it happens when i'm trying to start application. i have one user registered (or two, i can't remember if i put another version 1.7 (not downwards compatible)). 1.7 worked if i recall correctly, but again – i', not 100% sure, sorry.

erijo, 2018-03-09:
Would you be able to use adb to connect to your phone? We could then both verify that the fix I did is the right one and make the app work for you again.

Otherwise you'll have to wait until the next version, or if you have an backup, uninstall and reinstall the app (but note that all settings and measurements will be lost).

alekksander, 2018-03-09:
i'll wait for the next version, and then go with adb for linux if the error still occurs. Thank You for Your interest

PR #220 (2018-03-10): Handle invalid user being selected in config
Fixed #218 (at least I think so).
commit 3014ceb2 (2018-03-10): Handle invalid user being selected in config

Fixed #218
closed 2018-03-10 by oliexdev