oliexdev/openScale #219: Beurer BF 600 not supported.

riadhrom, 2018-03-08:
Hi everyone,

Could you help me about the connection of Beurer BF 600.

[image]

Thank you.

riadhrom, 2018-04-10:
@erijo : Hi erijo. Of course, I'm interested in support for that scale but I'm novice in computer programming so I don't have any idea about how I do it. I read the post and it's a chineese for me :( But I can test your code if you want :)

erijo, 2018-04-10:
Please try the dev version and let me know how it goes: https://github.com/oliexdev/openScale/releases/tag/travis-dev-build

riadhrom, 2018-04-10:
Thank you very much. The connection is established but the data is empty and no records.

[image]

erijo, 2018-04-10:
Please reset the scale and then connect with the app again. Do you then see a message requesting you to step on the scale?

erijo, 2018-04-10:
I think we need the bluetooth trafic log to figure this out. Can you perform part 1 from https://github.com/oliexdev/openScale/wiki/How-to-reverse-engineer-a-Bluetooth-4.x-scale and post the log here?

Dododappere, 2018-04-12:
The BF600 is different than the other Beurer scales. This one is based on the standard bluetooth profiles for User, Weight and BodyComposition. So you have to implement those to make it work.

See:
[User Data Service](https://www.bluetooth.com/specifications/gatt/viewer?attributeXmlFile=org.bluetooth.service.user_data.xml)
[Weight Scale Service](https://www.bluetooth.com/specifications/gatt/viewer?attributeXmlFile=org.bluetooth.service.weight_scale.xml)
[Body Composition Service](https://www.bluetooth.com/specifications/gatt/viewer?attributeXmlFile=org.bluetooth.service.body_composition.xml)

erijo, 2018-04-29:
@riadhrom : will you be able to provide the bluetooth logs? Otherwise I don't think we can do much about this...

dglozano, 2018-05-05:
Hello, I am Diego from Argentina. I am trying build an app that connects to a BF600 for a project at University, but I am struggling a lot to get the weight data from the scale.

Looking at the logs (I am attaching 3 different logs, all of them with data that traveled between the scale and the Beurer App) it seems that you need to set the indicate flag in the following characteristics:
-Weight Measurement
-Body Composition Measurement
-User Control Point
And the notification flag in:
-Database change Increment
-Current Time

There are some other custom services and characteristics that I don't know what they are for.

Afterwards, it creates an user if no user exists already. If an user exists, you can write to the User Control Point a Consen operation with the user index and the user pin (a number that you can see in the scale for each user)

I have tried to do that but It is not working at all. It is my very first attempt to develop something connecting to a BLE device so I am very lost... I have tried to follow the tutorial in the Android API and I tried to get just the Weight Measurement for an already created user but I couldn't :(

I would be very grateful if any of you could help me, and if I can get it working I will be able to contribute adding support for the BF600 for this project too.

[btsnoop_hci4-5-4396PIN-CREACION-CONEXION.log](https://github.com/oliexdev/openScale/files/1977097/btsnoop_hci4-5-4396PIN-CREACION-CONEXION.log)
[btsnoop_hciLAST7277.log](https://github.com/oliexdev/openScale/files/1977098/btsnoop_hciLAST7277.log)
[btsnoop_hci75.log](https://github.com/oliexdev/openScale/files/1977099/btsnoop_hci75.log)

(if you need more logs or anything let me know)

dglozano, 2018-06-19:
Erijo, sorry I have been very busy with uni (the semester ends next week here). After that I will continue trying with this and I will upload the log @erijo

erijo, 2018-11-27:
Long time to no action, but I've created a dev version with some initial code that might be able to set the time on the scale. Can someone please test [this version](https://github.com/oliexdev/openScale/releases/tag/travis-dev_bf600-build) and attach a log here?

erijo, 2018-12-05:
Anyone willing to try the dev version? We can't bring this forward unless someone is willing to help run the dev version and post logs.

sirkevinlevin, 2018-12-29:
@erijo I've tested your version on my BF600, here's the log: [openScale_2018-12-29_23-26.txt](https://github.com/oliexdev/openScale/files/2716764/openScale_2018-12-29_23-26.txt)

fabianpie, 2019-01-09:
Hi @Dododappere I haven't bought and tried a Beurer BF600 yet. Could you confirm it implements the standard profiles ? Have you developed an app and you are sure ?
Thanks,
Fabian

oliexdev, 2019-03-17:
first skeleton by @erijo for reference

```
/* Copyright (C) 2018 Erik Johansson <erik@ejohansson.se>
 *
 *    This program is free software: you can redistribute it and/or modify
 *    it under the terms of the GNU General Public License as published by
 *    the Free Software Foundation, either version 3 of the License, or
 *    (at your option) any later version.
 *
 *    This program is distributed in the hope that it will be useful,
 *    but WITHOUT ANY WARRANTY; without even the implied warranty of
 *    MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
 *    GNU General Public License for more details.
 *
 *    You should have received a copy of the GNU General Public License
 *    along with this program.  If not, see <http://www.gnu.org/licenses/>
 */

package com.health.openscale.core.bluetooth;

import android.bluetooth.BluetoothGatt;
import android.bluetooth.BluetoothGattCharacteristic;
import android.content.Context;

import com.health.openscale.core.utils.Converters;

import java.util.Calendar;

public class BluetoothBf600 extends BluetoothCommunication {

    public BluetoothBf600(Context context) {
        super(context);
    }

    @Override
    public String driverName() {
        return "BF600";
    }

    @Override
    protected boolean nextInitCmd(int stateNr) {
        switch (stateNr) {
            case 0:
                setIndicationOn(BluetoothGattUuid.SERVICE_WEIGHT_SCALE,
                        BluetoothGattUuid.CHARACTERISTIC_WEIGHT_MEASUREMENT,
                        BluetoothGattUuid.DESCRIPTOR_CLIENT_CHARACTERISTIC_CONFIGURATION);
                break;
            case 1:
                setIndicationOn(BluetoothGattUuid.SERVICE_BODY_COMPOSITION,
                        BluetoothGattUuid.CHARACTERISTIC_BODY_COMPOSITION_MEASUREMENT,
                        BluetoothGattUuid.DESCRIPTOR_CLIENT_CHARACTERISTIC_CONFIGURATION);
                break;
            case 2:
                byte[] time = new byte[10];
                Calendar cal = Calendar.getInstance();

                Converters.toInt16Le(time, 0, cal.get(Calendar.YEAR));
                time[2] = (byte) (cal.get(Calendar.MONTH) + 1);
                time[3] = (byte) cal.get(Calendar.DAY_OF_MONTH);
                time[4] = (byte) cal.get(Calendar.HOUR_OF_DAY);
                time[5] = (byte) cal.get(Calendar.MINUTE);
                time[6] = (byte) cal.get(Calendar.SECOND);
                time[7] = 0; // day of week (0 = unknown)
                time[8] = 0; // fractions of a second
                time[9] = 0; // adjust reason

                writeBytes(BluetoothGattUuid.SERVICE_CURRENT_TIME,
                        BluetoothGattUuid.CHARACTERISTIC_CURRENT_TIME, time);
                break;
            default:
                return false;
        }

        return true;
    }

    @Override
    protected boolean nextBluetoothCmd(int stateNr) {
        return false;
    }

    @Override
    protected boolean nextCleanUpCmd(int stateNr) {
        return false;
    }

    @Override
    public void onBluetoothDataChange(BluetoothGatt bluetoothGatt, BluetoothGattCharacteristic gattCharacteristic) {

    }
}
```

Dododappere, 2019-04-14:
[https://github.com/weliem/beurer-BF600-BF850/blob/master/app/src/main/java/com/welie/bf600/BluetoothHandler.java](https://github.com/weliem/beurer-BF600-BF850/blob/master/app/src/main/java/com/welie/bf600/BluetoothHandler.java)

oliexdev, 2019-04-22:
@Dododappere thanks for the link

oliexdev, 2019-05-25:
please try out the latest blessed-android branch version at https://github.com/oliexdev/openScale/releases/tag/travis-dev_blessed-android-build
I implemented a first version based on the source-code by @weliem (many thanks to him!)
As I don't own this scale, I can't test it, so please test it while the debug log is on (enable it under settings-about)

mohdumarj, 2019-06-23:
the above file is not available by @weliem ...  :(

shurakr, 2019-09-30:
@oliexdev the above file is not available

shurakr, 2019-10-04:
I found out the description of the bf600 and bf850 protocol

[bleapi.zip](https://github.com/oliexdev/openScale/files/3690478/bleapi.zip)

oliexdev, 2019-10-12:
@shurakr I don't own this scale without testing the code it will be hard to support it. We need someone to own this scale and to to test the source code. I implemented already some untested version. Do you have programmer skills?

shurakr, 2019-10-12:
Yes, i have

oliexdev, 2019-10-12:
Then you could test the implementation, you have only uncomment the line
`if (name.startsWith("BEURER BF600".toLowerCase(Locale.US))`
in the BluetoothFactory.java file.

shurakr, 2019-10-12:
I do not have a development environment, please build a test version.

oliexdev, 2019-10-12:
@shurakr I thought you have programmer skills? If not it will be difficult.

shurakr, 2019-10-12:
experience is, but I'm not programming for android.

shackra, 2020-06-08:
got this scale lol. Uh, how can I help?

schoeller, 2020-11-09:
> Then you could test the implementation, you have only uncomment the line
> `if (name.startsWith("BEURER BF600".toLowerCase(Locale.US))`
> in the BluetoothFactory.java file.

Dear all,

thanks for all the work. I am running LineageOS 10. After uncommenting the lines in `BluetoothFactory.java` I compiled using

```
./gradlew compileDebugSources
./gradlew installDebug
```
copied the `openScale-2.3.2-debug.apk` onto my phone, installed, executed and connected to my BF600 scale. This is the error output I have received:

```
Build version: 2.3.2 
Build date: 1981-01-01 01:01:02 
Current date: 2020-11-09 10:04:54 
Device: Motorola moto g(7) 
OS version: Android 10 (SDK 29) 
 
Stack trace:  
java.lang.IllegalArgumentException: ID does not reference a View inside this Activity
	at android.app.Activity.requireViewById(Activity.java:3237)
	at androidx.core.app.ActivityCompat.requireViewById(ActivityCompat.java:363)
	at androidx.navigation.Navigation.findNavController(Navigation.java:58)
	at com.health.openscale.gui.preferences.BluetoothSettingsFragment$BluetoothDeviceView.onClick(BluetoothSettingsFragment.java:365)
	at android.view.View.performClick(View.java:7259)
	at android.view.View.performClickInternal(View.java:7236)
	at android.view.View.access$3600(View.java:801)
	at android.view.View$PerformClick.run(View.java:27892)
	at android.os.Handler.handleCallback(Handler.java:883)
	at android.os.Handler.dispatchMessage(Handler.java:100)
	at android.os.Looper.loop(Looper.java:214)
	at android.app.ActivityThread.main(ActivityThread.java:7356)
	at java.lang.reflect.Method.invoke(Native Method)
	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:491)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:940)

```

I would be glad to assist further with my limited capabilites.

Kind regards

Sebastian

krisjans, 2021-07-08:
@oliexdev 
> Then you could test the implementation, you have only uncomment the line
> `if (name.startsWith("BEURER BF600".toLowerCase(Locale.US))`
> in the BluetoothFactory.java file.

I uncommented the code block thats add BF600 support and here is application log I got:
[openScale_2021-07-08_12-00.txt](https://github.com/oliexdev/openScale/files/6783267/openScale_2021-07-08_12-00.txt)

Something went wrong with user creation:
`2021-07-08 12:36:24.124 Error [2] BluetoothStandardWeightProfile: Not authorized`
`2021-07-08 12:36:25.595 Debug [2] BluetoothStandardWeightProfile: Created user 13`

Another log after I did factory reset on scale:
[openScale_2021-07-08_12-34.txt](https://github.com/oliexdev/openScale/files/6783383/openScale_2021-07-08_12-34.txt)

krisjans, 2021-07-08:
@oliexdev 
I managed to store user config from app to Beurer BF600 scale and read weight and body composition  measurements.
Please take a look at pull request: https://github.com/oliexdev/openScale/pull/753
Would be nice if someone could test this also on BF850.

krisjans, 2021-07-11:
This is hci log when Beurer "Health manager" connects to scale BF600 (factory reset has been executed on the scale) for the first time: [btsnoop_hci.log](https://github.com/oliexdev/openScale/files/6796523/btsnoop_hci.log).
@jensMF , https://github.com/oliexdev/openScale/pull/735

krisjans, 2021-07-15:
Anyone wiling to test my latest support for BF600? Pinging people who tested BF600 before: @schoeller @shurakr @heikoschwarz @inkiiin @dglozano @riadhrom ?

This is my private build of #753 : [openScale-2.3.5-debug.zip](https://github.com/oliexdev/openScale/files/6822054/openScale-2.3.5-debug.zip):
- Weight and body composition measurements seems to be working OK (tested only in metrics/SI system);
- App will create new user on scale for active app user and remember the binding;
- Works with several users, but syncing only actively selected user in app;
- Unfortunately there is no support yet to bind existing scale user to app user.
- this version will recognize BF600 and BF105 only, if you want to test on BF850, BF950 then ping me and I might be able to build version that treats those scales as BF600 or BF105.

krisjans, 2021-07-22:
I made some more progress: added existing scale user selection (and pin/consent code dialog).
If anyone interested to test - this is updated build of #753: [openScale-2.3.5-debug_ui_change.zip](https://github.com/oliexdev/openScale/files/6862874/openScale-2.3.5-debug_ui_change.zip)

UI changes works only on BF600. Since I have no BF105 I did not update it's support code right now (@jensMF maybe you could check my UI solution and re-use it for BF105? It seems that some-kind of Beurer middle-layer class will be needed to re-use Beurer specific code for BF600, BF850, BF950 and BF105.)

[image] [image] [image]

shackra, 2021-07-22:
> UI changes works only on BF600. Since I have no BF105 I did not update it's support code right now ***@***.***MF](https://github.com/jensMF) maybe you could check my UI solution and re-use it for BF105? It seems that some-kind of Beurer middle-layer class will be needed to re-use Beurer specific code for BF600, BF850, BF950 and BF105.)

Do I need to factory reset my scale before pairing it with this APK?

‐‐‐‐‐‐‐ Original Message ‐‐‐‐‐‐‐

krisjans, 2021-07-22:
> Do I need to factory reset my scale before pairing it with this APK?

@shackra : No, factory reset is not needed.  You should be able to connect to BF600 and either use existing user or create a new one.

shackra, 2021-07-22:
I was able to connect with my BF600 and use the same user on the scale, I will weight myself this Friday

‐‐‐‐‐‐‐ Original Message ‐‐‐‐‐‐‐

shackra, 2021-07-23:
weighted myself, and for my surprise, things went smooth, like using the Beurer's HealthManager but 1000 times better. I'll manually move my data from there to OpenScale. Looking forward for its next release!

shackra, 2021-07-23:
oh, btw, I noticed that the bone weight is not getting registered.

I see the BMI getting recorded on the app but I don't see it exported in the CSV, I do have these numbers from HealthManager in my exported file

krisjans, 2021-07-23:
> oh, btw, I noticed that the bone weight is not getting registered.
> 
> I see the BMI getting recorded on the app but I don't see it exported in the CSV, I do have these numbers from HealthManager in my exported file

BF600 scale does not provide bone-mass neither in weight or body composition packet.
But it does transmit body "impedance", fat %, water mass, soft-lean-mass, muscle mass, fat-free-mass -  therefore I think bone mass could be calculated from those. I see that for other scales bone-mass is often calculated from other measurements (total weight, impedance, fat % and others).
@shackra could you check your exported data from HealthManager and see if your bone mass could be calculated from other provided values?

shackra, 2021-07-23:
> BF600 scale does not provide bone-mass neither in weight or body composition packet.
> But it does transmit body "impedance", fat %, water mass, soft-lean-mass, muscle mass, fat-free-mass - therefore [...]

my exported file only provides IBM, % of muscle mass, % of water, % of body fat and the bone density which is display in kg

krisjans, 2021-07-23:
@shackra Maybe you would like to help to figure out formula to calculate bone-mass from other measurments? 
If you do, then grab this debug version: [openScale-2.3.5-debug-bone-mass-quest.zip](https://github.com/oliexdev/openScale/files/6870619/openScale-2.3.5-debug-bone-mass-quest.zip)
This version will dump all measurements provided by scale in debug log.
If you already know your bone mass, then maybe you can figure out how to calculate it from other values (for example subtract all other known weight values from total body mass)?
Example:
```
D/BluetoothStandardWeightProfile: ################################################## Step Nr 15
D/BluetoothStandardWeightProfile: handleWeightMeasurement -------------------------------------------- value: 0E F2 2F E5 07 07 17 15 21 1F 02 D4 00 A4 06
D/BluetoothStandardWeightProfile: handleWeightMeasurement -------------------------------------------- flags: 0x0e 
D/BluetoothStandardWeightProfile: handleWeightMeasurement -------------------------------------------- Time: Fri Jul 23 21:33:31 GMT+03:00 2021
D/BluetoothStandardWeightProfile: handleWeightMeasurement -------------------------------------------- User id: 1, scale user index: 2
D/BluetoothStandardWeightProfile: handleWeightMeasurement -------------------------------------------- BMI: 21.2
D/BluetoothStandardWeightProfile: handleWeightMeasurement -------------------------------------------- heightInMeters: 1.7
D/BluetoothStandardWeightProfile: Got weight: 61.37
D/BluetoothStandardWeightProfile: ################################################## Step Nr 15
D/BluetoothStandardWeightProfile: handleBodyCompositionMeasurement ---------------------------------- value: 98 03 B6 00 13 18 A3 01 30 25 F4 1A 79 0B
D/BluetoothStandardWeightProfile: handleBodyCompositionMeasurement ---------------------------------- flags: 0x0398
D/BluetoothStandardWeightProfile: handleBodyCompositionMeasurement ---------------------------------- bodyFatPercentage: 18.2
D/BluetoothStandardWeightProfile: handleBodyCompositionMeasurement ---------------------------------- bmrInJoules: 6163 bmrInKcal: 1472
D/BluetoothStandardWeightProfile: handleBodyCompositionMeasurement ---------------------------------- musclePercentage: 41.9
D/BluetoothStandardWeightProfile: handleBodyCompositionMeasurement ---------------------------------- softLeanMass: 47.6
D/BluetoothStandardWeightProfile: handleBodyCompositionMeasurement ---------------------------------- bodyWaterMass: 34.5
D/BluetoothStandardWeightProfile: handleBodyCompositionMeasurement ---------------------------------- impedance: 293.7
D/BluetoothStandardWeightProfile: Got body composition: 98 03 B6 00 13 18 A3 01 30 25 F4 1A 79 0B
D/OpenScale: Added measurement: ID: 0, USER_ID: 1, DATE_TIME: Fri Jul 23 21:33:31 GMT+03:00 2021, WEIGHT: 61.37, FAT: 18.20, WATER: 34.50, MUSCLE: 41.90, LBM: 47.60, WAIST: 0.00, HIP: 0.00, BONE: 0.00, CHEST: 0.00, THIGH: 0.00, ARM: 0.00, NECK: 0.00, CALIPER1: 0.00, CALIPER2: 0.00, CALIPER3: 0.00, COMMENT: 
D/BluetoothStandardWeightProfile: ################################################## Step Nr 15

```

shackra, 2021-07-23:
I rather give you my logs, lol. I'm awful at math.

my bone weight has been always between 3.2 KG, 3.3 KG and 3.4 KG, I don't discard that is still fluctuating around those numbers (probably I'm at 3.3 KG right now)

[openScale_2021-07-23_13-23.txt](https://github.com/oliexdev/openScale/files/6870868/openScale_2021-07-23_13-23.txt)

dror3go, 2021-07-25:
Hey @krisjans - many thanks for the progress you're making here :clap: 
I have BF 950, and I would like to try out your latest version. Any chance you can create a relevant APK?

> It seems that some-kind of Beurer middle-layer class will be needed to re-use Beurer specific code for BF600, BF850, BF950 and BF105.

Should it be "Bluetooth standard protocol"-specific instead of "Beurer"-specific?

krisjans, 2021-07-26:
A bit more progress:
- fixed the crash that happened when user spent a lot of time in scale-user-selection or pin/consent dialog and scale got disconnected by time-out;
- tested BF600 support on BF850 and it works (except no support for user initials). Added BF850 to supported devices;
- experiment: treat BF950 as BF600 (request by @dror3go );

If anyone interested to test - this is updated build of #753: [openScale-2.3.5-debug_bf600_bf850_bf950.zip](https://github.com/oliexdev/openScale/files/6877239/openScale-2.3.5-debug_bf600_bf850_bf950.zip)

krisjans, 2021-07-26:
> > It seems that some-kind of Beurer middle-layer class will be needed to re-use Beurer specific code for BF600, BF850, BF950 and BF105.
> 
> Should it be "Bluetooth standard protocol"-specific instead of "Beurer"-specific?

It should be in between  "Bluetooth standard protocol" and specific Beurer scale classes (see this [comment](https://github.com/oliexdev/openScale/pull/753#issuecomment-885063965) in pull-request).
BluetoothStandardWeightProfile -> BluetoothStandardWeightProfileBeurer -> BluetoothBeurerBF105
BluetoothStandardWeightProfile -> BluetoothStandardWeightProfileBeurer -> BluetoothBeurerBF600
BluetoothStandardWeightProfile -> BluetoothStandardWeightProfileBeurer -> BluetoothBeurerBF850
BluetoothStandardWeightProfile -> BluetoothStandardWeightProfileBeurer -> BluetoothBeurerBF950

krisjans, 2021-07-27:
A bit more progress:

- Added calculation of "lean body mass" and "bone mass" (Thanks to ASerbinski for providing formulas) @shackra ;
- experiment: treat BF950 as BF600 (#502, @dror3go );
- super-experimental support for SBF77 (#681, @PeterHuewe @rapht0r)

If anyone interested to test - this is updated build of #753[openScale-2.3.5-debug_bone_mass_bf600_bf850_bf950_sbf77.zip](https://github.com/oliexdev/openScale/files/6887456/openScale-2.3.5-debug_bone_mass_bf600_bf850_bf950_sbf77.zip)

Comment here and attach openScale debug logs (and if possible also btsnoop_hci.log) if something (or nothing) works! Especially if you test on BF950 or SBF77.

shackra, 2021-07-27:
gonna install and tomorrow will weight myself and report back anything worth reporting

thanks!

‐‐‐‐‐‐‐ Original Message ‐‐‐‐‐‐‐

krisjans, 2021-07-28:
If someone is interested in testing my latest changes related to [BF600](https://github.com/oliexdev/openScale/issues/219), [BF850](https://github.com/oliexdev/openScale/issues/531), [BF950](https://github.com/oliexdev/openScale/issues/502) and [SBF77](https://github.com/oliexdev/openScale/issues/681), then latest automatic debug build is availabe here: [openScale-2.3.5-debug.apk](https://github.com/krisjans/openScale/releases/download/debug_latest/openScale-2.3.5-debug.apk), change log in #753.

If something does not work or you encounter a crash - I would highly appreciate if you leave me a comment with attached openScale debug log (if possible also  btsnoop_hci.log. [Instructions how to get logs.](https://github.com/oliexdev/openScale/wiki/How-to-reverse-engineer-a-Bluetooth-4.x-scale#1-acquiring-some-bluetooth-traffic))

shackra, 2021-07-28:
@krisjans so I weight myself this morning, my lean mass has increase by 3.14 kg, lost 0.10% of body fat and 0.06 kg of bone density along with 0.30 kg of weight and 0.14% of water.

I think so far so good?

krisjans, 2021-07-28:
@shackra Thanks for your feedback!
> my lean mass has increase by 3.14 kg

Yesterday I updated lean-body-mass calculation formula for BF600 (thanks to ASerbinski for providing formulas)  - maybe this caused such dramatic change in this measurement.
Previously I saved soft-lean-body-mass (which excludes bone mass) as lean-body-mass (which should includes bone mass). And you previously wrote that you bone-mass is around 3.1-3.4 kg.
I guess you need to adjust few of your previous measurements manually...

shackra, 2021-07-28:
Those previous measurements came from HealthManager, IIRC, but I cannot recall if in my last measurements for that slot was 0 kg (thus, having a correct lecture introduces the dramatic increase)

‐‐‐‐‐‐‐ Original Message ‐‐‐‐‐‐‐

krisjans, 2021-07-28:
How did you imported your date from Health manager? You exported your data
from Beurer Health manager as CSV and imported it in openScale as backuP?

shackra, 2021-07-28:
I exported data from OpenScale after my first measurement, then exported all my data from HealthManager, open both files in LibreOffice, copied data exported from HealthManager to the exported file from OpenScale and imported that back without the first row of data

‐‐‐‐‐‐‐ Original Message ‐‐‐‐‐‐‐

krisjans, 2021-07-28:
Do you still have that csv file that you imported into openScale? Could you
provide fragment from it?
There might be some kind of error in openScale formula that calculates
lean-body-mass from other measurements when lean-body-mass is not saved in
database directly.

shackra, 2021-07-28:
Here may be a confusion, but the data exported had lean mass as 0 kg IIRC and the APK that was installed was the first one you released with support for my scale

shackra, 2021-08-07:
today I woke up, measured myself on the scale and somehow I went up 21% in body fat for a total of 48.60%. I know I've been eating sweets here and there but that increase of body fat is impossible.

my APK is version 2.3.5 (54)

krisjans, 2021-08-07:
@shackra Could you enable openScale debug log, repeat measurement and attach log to a comment? It is hard to tell what went wrong without additional logs.

shackra, 2021-08-08:
@krisjans turns out my scale is actually reporting those outrageous numbers lol, so never mind!

oliexdev, 2021-10-10:
see #753

closed 2021-10-10 by oliexdev

soloinovator, 2022-08-13:
https://docs.github.com/en/codespaces/overview#using-github-codespaces