oliexdev/openScale #223: Separate axes for measurements with different units and ranges

cosminribo, 2018-03-09:
I've been using Open Scale for the past year and it's been a great application. However, recently, the graph showing weight got mixed with other measurements, which are all shown on the same y axis. This is very confusing. While I understand that you want to show as many trends as possible, the use of a single axis for measurements that have widely different ranges and units seem unfortunate.

Would it be possible to have separate graphs for the measurements that users want to see ?

(my usage involves only weight measurements, and the ability to export and import measurement values is absolutely fantastic).

Thank you.

erijo, 2018-03-09:
Do you know that you can disable measurements you don't use in the settings and that you can hide measurements in the graph by clicking on the icons on the graph page?

oliexdev, 2018-03-18:
as @erijo already said just disable the lines that you don't need.

closed 2018-03-18 by oliexdev

cosminribo, 2018-03-31:
Thank you for your solution, it works well for the graph page. However, it is currently impossible to hide the BMI on the overview page.

Having different measures, with different units on the same graph remains highly confusing. Not a critical problem though. I like the app anyway :-)

erijo, 2018-03-31:
In the next version you can turn off measurements in the overview graph (from the settings menu).