oliexdev/openScale #226: BMI can't be disabled

woj-tek, 2018-03-11:
As the title says - BMI calculation and display can't be disabled (and thus hidden in the overview graph; somewhat related to  #130).

momo42, 2018-03-13:
I use this app almost every day and I really like how it evolved over time but I just monitor my weight. So I disabled all the other graphs. In the recent version disabling the BMI is not possible. In the graph section this is not a big deal because you can disable it here "on the fly" but the overview graph degenerates to two straight lines in my case ... the bmi on the bottom and the weight on the top. I'd really like to have the option to hide the bmi graph. :-)

erijo, 2018-03-18:
I'm working on restructuring the preferences a bit. This will be included in that work.

closed 2018-03-24 by oliexdev

erijo, 2018-03-24:
Please test the latest dev build and let me know what you think about the new preferences and if you have any suggestion.

https://github.com/oliexdev/openScale/releases/tag/travis-dev-build

woj-tek, 2018-03-24:
It looks great now (very very! and I love the new weight input box!)

There is a slight issue with the settings - sometimes tapping on the category (left part, with the name) doesn't open the new screen - at first I thought it was linked to the toggle (right part) but it's possible to navigate to the next screen with toggle off. Works always with circumference. 

Also (possibly related with above) - it's possible to disable metric and enable it in overview screen.

erijo, 2018-03-24:
Hmm. You're not supposed to be able to navigate to the next screen when the switch is off. Can you provide step-by-step instructions on how to reproduce this?

erijo, 2018-03-24:
Think I might have found the bug. Please test with latest dev version once 57daf44 has been built.

woj-tek, 2018-03-26:
OK, it seems to work ok now. Thank you for the quick fix

PR #233 (2018-03-24): Refactor preferences
- Make it possible to disable all measurements (except weight). Fixes #226 and fixes #224.

- Add separate preference screens to all measurements to configure e.g. if the measurement should be displayed in the overview graph (this fixes #225 in a different way compared to previously), if relative or absolute weight should be used etc.

- Include reordering of measurements in the main measurement preference screen.

+ some other fixes.

@oliexdev: if you have time, please take a look and let me know what you think.
commit 34e6856e (2018-03-24): Refactor preferences (#233)

* Remove unused code

* Make measurement key available as a static field as well

* Move enable logic to base class

* State dependency explicitly in all measurement views

* Add new class to keep track of settings for a measurement view

* Use new settings class to get hold of estimation settings

* Move handling of estimation/percentage enabled to base class

* Remove unused imports

* Update to c29dea32

* Refactor measurement settings

- Make it possible to disable all measurements (except weight). Fixes
  #226 and fixes #224.

- Add separate preference screens [... cut]
commit ebe2d6df (2018-03-24): Refactor measurement settings

- Make it possible to disable all measurements (except weight). Fixes
  #226 and fixes #224.

- Add separate preference screens to all measurements to configure
  e.g. if the measurement should be displayed in the overview graph
  (this fixes #225 in a different way compared to previously), if
  relative or absolute weight should be used etc.

- Include reordering of measurements in the main measurement
  preference screen.