oliexdev/openScale #228: New SBF-75 instability on measurements

lenucksi, 2018-03-14:
## Issue
- SBF-75 does show user name in scale (--- instead) anymore.
- Also, history is not stored anymore. The scale needs to be actively connected to the app while measuring or the measurement will be lost.
- If two users with similar weight ranges are present, measurements might be wrongly associated to a user.
- Registering a profile with a name with the same prefix on two devices appears to cause havoc on the scale.

## Observed behaviour leading to the issue:
- I've tested #205 on a different device (Also Android 7.1.2) using the dev-build given then. For that purpose I've created and connected the app to the scale, that time using a username with a similar 4 letter prefix.
- Measurement as illustrated worked fine on both scales.
- I guess starting with the upgrade 1.7.1 the issues illustrated above appeared. This time on both devices. Maybe the scale has issues handling multiple devices, or the user creation process has issues? Any ideas how to debug this?
  - I've tried to fully reset the scale, however this appeared to have no effect.
  - I've tried to create a new, different, non-prefix similar user and have it registered, and shown on measurements, however to no avail.
- This might also include issues with non-weight-only measurements.

I'd be happy to get any insight or ideas on how to debug this, as the state before this was very nice and stable.

Dododappere, 2018-03-14:
@lenucksi when you see '---' appear on the scale, it means that the scale doesn't know which user to assign it to. Probably because the two users you are using are too similar in weight.

As a consequence, the measurement is internally stored as 'unknown measurement'. OpenScale currently doesn't handle the 'unknown measurement'....

So I wouldn't call it a bug in openScale....

lenucksi, 2018-03-15:
Well, it obviously leads to unexpected behavior and thus irritation of users. This, just like other things a UX issue and thus a bug.

## Questions
- As far as it appears, this can also be triggered by two devices with a similar user being linked to the scale.
- Is it known which weight difference is needed by the scale's measurement association function to identify a user?
- Also, do we know any procedure to fix this dissociation once it happened? As in, delete / recreate, magic rain dance, ...?

## Possible action points, fix approaches
- Upon creation of multiple users warn or disable creation if the resulting users would trigger this behavior in the scale due to too similar weight
- Add information that coupling the scale to multiple devices might trigger such a bug
- Are the approaches that work around the scales limitations?

Dododappere, 2018-03-15:
I can understand the fact it leads to UX issues.  But to me a bug is a mistake in the software and in this case it is not so much a mistake in openScale but simply a consequence of how the scale works. It would also happen with the official app 

But to be constructive:
- It is technically possible to influence the range for detecting a user. I think by default it is +/- 2 Kg of the last weight. However, even if you would set it to 3 kg, the chances of having overlapping range with the other user is high. If you would make it smaller it will make it even more hard to detect the correct user because of the natural up/downs of weight on a day-to-day basis. So this doesn't really help much.
- It is also technically possible to manually set the reference weight for a user on the scale. Openscale does not implement this yet but it could be used to get out of this situation if it occurs.
- The best way is probably to start supporting 'unknown measurements'. There are commands to retrieve those and then assign them to the right user.....something for the TO-DO list!
- But in general there is not really a good solution for detecting users if the users are of nearly the same weight....most scales have this issue, not just this one

P.S. If you want to solve it quickly. Let one of the users hold some like a book to (temporarily) make the user heavier and see if the detection works again....

oliexdev, 2018-04-08:
if you have ranges that are overlapping in weight then as @Dododappere wrote it is not clear which user should be assign. 

So if you enable the smartUserAssignment in openScale. It would use the user with the closest weight but also checks if the weight is in range +-15. If they are out of range openScale will ignore this measurement. You can disable this range check in the settings! Then the out of range measurements are assign to the current user. That is the best automatic solution which come to my mind to handle multiple user with close weight measurements. But this is a general problem and cannot be 100% solved. As you see your scale manufacture seems to ignore it or they made it easy to assign it to an unknown user.

Other solution would be to ask every time the user with a dialog to assign the measurement to the correct user but that might be annoying!?

As @Dododappere wrote the unknown measurements are not received currently by openScale, so yeah that should be implemented. The assign process could be done by openScale with the smartUserAssign described above but I don't know these commands for retrieving the unknown measurements.

lenucksi, 2018-04-10:
Thanks @oliexdev, I'll check back with the scale later and update this. I've found another weird effect with regards to user assignment on the scale and coupling, too.

oliexdev, 2018-04-21:
I will close this issue for the time being. If you found something similar please wrote it in this issue.

closed 2018-04-21 by oliexdev