oliexdev/openScale #230: How I fixed strange issues with a Sanitas SBF 70

sebmate, 2018-03-20:
Hi all,

first of all, thanks to all developers of openScale! It's great to see that you are "freeing" these devices!

This is not an "issue" report, but rather some comments in the hope that they could be useful for those who encounter problems with their Sanitas SBF 70. I had some major connection issues and was about to return the nice scale. Here's the rather long story:

After picking the scale up at my parents' house (where I also tested it), I manged to get the full measurements working after some fiddling. At my home, it stopped working. OpenScale found the SBF 70, but it did not receive any measurements (not even weight). Also the official App from Sanitas found the scale, but could not connect to it. Then I noticed that my SD card somehow got corrupted - just out of coincidence. I'm not sure if this is what caused the problems.

To make the long story short, here's what I did:

First, I reset ALL network settings in Android (which is version 7.0), which also includes the bluetooth settings. This can be found under in "Settings => Backup & Restore => Reset Network Connections" (not sure about that, my mobile is set to German).

Second, I booted a Windows 10 tablet (Trekstor Wintron 7) with an Android-x86 CM 14.1-r1 "Live" image, connected a USB mouse and after a lot of patience I managed to get openScale running. To my surprise it worked right out of the box, and I was able to retrieve complete measurements. I then reverted back to my normal mobile and it worked.

Note that I also reset the scale via the button between all steps.

I'm not sure if it was the reset of the network settings in Android or the assignment to another bluetooth device (the tablet PC). I think it was the latter. My assumption is that the reset button on the scale probably does not reset the bonding of the scale's internal bluetooth module (any opinions on that?).

I will keep an eye on this and will have to think about what I will do with the Medisana BS 444, which arrives tomorrow (rolling eyes).

Anyway, keep up the good work!
Best, Sebastian

Dododappere, 2018-03-21:
Difficult to say what the issue is exactly. But most likely the issue is that both devices are trying to connect to the scale and 'fight' for the connection. The scale can only have 1 active connection at a time. If one device 'takes' the connection, another device will not be able to connect to it...

So if you are standing on the scale AND you see the Bluetooth icon on, a device is connected. Find that device by turn off the devices one by one and checking again on the sale. Once all devices are off there should be no Bluetooth icon visible anymore as you stand on the scale. Then try again with the device you want to use....measurements should come in again!

sebmate, 2018-03-21:
Hi Dododappere,

thanks for your tips. I observed this "fighting" for the connection, but this does not seem to be the problem. I can manage to create a stable connection between the phone and the scale (the scale then shows the Bluetooth symbol and openScale the symbol with the two dots on the left and right). But both do not appear to exchange data.

I'm not even sure if both devices create a permanent bond. Using the "reset" button on the scale also does not seem to affect which (openScale) device can connect to it.

I even traveled down into the basement of my 400-apartment building to make sure that there are no other Bluetooth devices. Maybe 

I don't know what's going on. If someone could tell me how I can access the logs (in the source code I can see that there's logging), I could dig deeper. I think that the phone's Bluetooth or Android is somehow messed up. It's strange that it WAS working.

Luckily I've found a temporary workaround: I've managed to permanently install Android x86 on that Windows tablet, and it does work with the scale (just fine). I can then export the data and transfer the CSV file via KDE Connect. Luckily the scale saves up to 30 measurements, so I don't have to do this every day.

I will continue to experiment with that and keep you updated. But I've had enough of that for now. :-)

Best, Sebastian

oliexdev, 2018-04-08:
Well if you select the correct device in the openScale Bluetooth settings then openScale will only search for this MAC address, so an interfering can't be the problem.
Am I understand right you are you using two devices to connect to the scale?
Do you have the same user names set for the two devices?

sebmate, 2018-04-08:
Hi oliexdev,

thanks for your feedback. It's good to know that the connection is bound to the MAC address. Yes, I was using two devices (the Wileyfox smartphone and the x86 Android tablet). Of course I've tried all various settings, including same user names. I've literally spent days trying. But it was only working on the tablet. Even a factory-reset of the phone did not help.

One interesting observation I've made: When the scale is connected via bluetooth and one steps on the scale, the weight values (on the LCD) are updated slower (one each second), compared to when the scale is NOT connected (there the LCD's refresh rate is much faster). So I think the scale was in fact sending the data, but the smartphone/openScale was not receiving anything or could not process it.

So I think it's problem with the Android version of this phone. I've returned the scale and got a Medisana BS444, which is working just fine (but the Sanitas is the nicer scale).

If this is a one-time issue, I guess we can close it ...

Keep up the good work!
Sebastian

closed 2018-04-08 by sebmate

erijo, 2018-04-14:
Please try the [dev version](https://github.com/oliexdev/openScale/releases/tag/travis-dev-build) @sebmate and see if it works better for you. I've done some refactoring of the Bluetooth connection code.