oliexdev/openScale #231: The last n days, with a fixed max and min?

rangzen, 2018-03-21:
Hello,
First thanks for openScale!
I can't find a way to see the last 30 or 90 days with a fixed maximum and a fixed minimum weight. I always have to zoom, to translate the diagram, etc. And when an objective is filled I can't zoom around, it's like a fixed minimum (yeah, I'm above :) )
Is it possible actually?

oliexdev, 2018-03-24:
Sorry I don't really know what you mean. Can you do some screenshots?

oliexdev, 2018-04-08:
Seems not a real problem so I will close this issue.

closed 2018-04-08 by oliexdev

rangzen, 2018-04-09:
Sorry, updates was in spam folder...
I want to fix a minimum and/or a maximum for the Y axis, actually the diagram take the max and min value.
And a number of days of history, actually it's the start of the year. In january I don't have any history visible.

For example, with 70kg min, 80kg min and 90 days of history, the diagram will look like that :
[image]

oliexdev, 2018-04-13:
and what exactly is the problem with that behaviour?

rangzen, 2018-04-15:
The actual behaviour?