oliexdev/openScale #232: 1.7.1 can't work without location / Add configure for disable location

Olangu, 2018-03-23:
Since 1.7.1 Open Scale requires location data. Denying location makes it stop working.
However, in earlier version this was not needed, so why is it now added at all?

If any of the scales needs this to function properly it should be a configurable option to enable it.

Please do not gather data not needed!

I will stay on 1.7 for now!

Olangu, 2018-03-23:
Ok, so I found the reason here:
https://github.com/oliexdev/openScale/issues/74#issuecomment-334922013

But the link provided also states the following now:
https://stackoverflow.com/a/39892691

Is this possible. It really sucks that you can't use OpenScale without giving it location data.

erijo, 2018-03-23:
Do you use Bluetooth? Which Android version?

Olangu, 2018-03-23:
Yes, bluetooth.
Sanitas SBF70 scale.
Android 7.0 (EMUI 5.0.1)

erijo, 2018-03-23:
Going back to targetSdk 22 is unfortunately not a good option. At the end of the year, Google will start requiring that updated apps use targetSdk >= 26. So we are stuck with needing the COARSE location permission to be able to search for devices.

One option might be to add support for specifying the scale's MAC address manually, or remember it during an initial discovery. The permission would then (presumably) only be needed for the initial discovery (or not at all if the MAC address is entered manually). Unfortunately I don't have any Bluetooth scale myself so I can't test if this works or not.

It should of course be added that openScale never uses the location in any way. It is only required by Android to be able to do Bluetooth scans. As the source is available for examination you can feel confident that enabling location access doesn't bring along any privacy problem. For extra security you can use the F-droid builds where you can be sure which source is actually used for a given build.

oliexdev, 2018-03-24:
@Olangu please read the FAQ!
@erijo You need the permission anyway. I thought the information text before enabling the Bluetooth permission was enough but I guess nobody reads that or maybe it is better to link to the FAQ for further information.

closed 2018-03-24 by oliexdev

erijo, 2018-04-15:
@Olangu
In the latest [dev version](https://github.com/oliexdev/openScale/releases/tag/travis-dev-build) you only need the location permission during the initial scan. Once that is done the permission can be revoked. Is that acceptable to you?

I thought about adding support for manual entry of the Bluetooth address, but at least on my scale it doesn't seem to be printed on the scale so a search had to be done anyway.