oliexdev/openScale #234: Crash

Auctoritas-Dirk, 2018-03-24:
Build version: 1.7.1 
Current date: 2018-03-25 00:16:09 
Device: Motorola Moto G4 Plus 
 
Stack trace:  
java.lang.RuntimeException: Failure delivering result ResultInfo{who=null, request=100, result=-1, data=Intent { dat=content://com.estrongs.files/storage/emulated/0/Download/openscale/openScale Dirk113.csv }} to activity {com.health.openscale/com.health.openscale.gui.MainActivity}: java.lang.NullPointerException: Attempt to invoke interface method 'boolean android.database.Cursor.moveToFirst()' on a null object reference
	at android.app.ActivityThread.deliverResults(ActivityThread.java:4126)
	at android.app.ActivityThread.handleSendResult(ActivityThread.java:4169)
	at android.app.ActivityThread.-wrap20(ActivityThread.java)
	at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1552)
	at android.os.Handler.dispatchMessage(Handler.java:102)
	at android.os.Looper.loop(Looper.java:154)
	at android.app.ActivityThread.main(ActivityThread.java:6186)
	at java.lang.reflect.Method.invoke(Native Method)
	at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:889)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:779)
Caused by: java.lang.NullPointerException: Attempt to invoke interface method 'boolean android.database.Cursor.moveToFirst()' on a null object reference
	at com.health.openscale.core.OpenScale.getFilenameFromUri(OpenScale.java:351)
	at com.health.openscale.core.OpenScale.importData(OpenScale.java:357)
	at com.health.openscale.gui.MainActivity.onActivityResult(MainActivity.java:685)
	at android.app.Activity.dispatchActivityResult(Activity.java:6937)
	at android.app.ActivityThread.deliverResults(ActivityThread.java:4122)
	... 9 more

User actions: 
2018-03-25 00:12:53: MainActivity resumed
2018-03-25 00:12:53: MainActivity paused
2018-03-25 00:12:53: UserSettingsActivity created
2018-03-25 00:12:53: UserSettingsActivity resumed
2018-03-25 00:13:02: UserSettingsActivity paused
2018-03-25 00:13:02: UserSettingsActivity resumed
2018-03-25 00:13:39: UserSettingsActivity paused
2018-03-25 00:13:39: MainActivity resumed
2018-03-25 00:13:40: UserSettingsActivity destroyed
2018-03-25 00:13:44: MainActivity paused

closed 2018-03-25 by erijo

erijo, 2018-03-25:
Thanks for your bug report. I think I have fixed this crash now. Could you please verify that it is fixed by testing the [dev build](https://github.com/oliexdev/openScale/releases/tag/travis-dev-build)?

Auctoritas-Dirk, 2018-03-25:
Ich bin noch am Testen:
Eine exportierte Datei kann jetzt fehlerfrei wieder importiert werden.
Etwas schwieriger ist es bei einer anderen früher  mal exportierten Datei:
Hier tauchen Fehler mit der Permission auf, ich werde dies aber noch
genauer testen.

Besten Dank für die schnelle Reaktion

erijo, 2018-03-25:
If you are ok with sharing the file here or with me via mail I can have a look and see if I can reproduce the wrote.

Auctoritas-Dirk, 2018-03-26:
Besten Dank noch mal für die Hilfe,
Ich habe den Fehler beim Import gefunden:
Ich hatte versucht eine Datei zu importieren die ich vorher am PC
bearbeitet hatte und dabei hatte sich wohl ein Fehler eingeschlichen. Der
Import der unbearbeiteten Datei verlief ohne Fehler.

Mit freundlichen Grüßen Dirk stolte

erijo, 2018-03-26:
Ok, glad to hear you sorted it out.

commit cbf3c983 (2018-03-25): Handle ContentResolver.query() returning null result

Fixes #234