oliexdev/openScale #235: Complete Pain to pair with scales

Lewiscowles1986, 2018-03-26:
Here's the ritual I had to perform to connect with scales (may be manufacturer problem).

1. Open settings, turn on Bluetooth, turn on discoverability, rename device
2. Restart phone
3. Open app, goto settings, bluetooth.
4. Power on device, and whilst standing on scales, click device to search for what looks like a mac address.
5. Step off of scales, go back to home screen in app, and click pair
6. Step on the scales, if it doesn't work, repeat stepping on and off and clicking pair, only after a toast message appears

It's a fantastic app and has saved me from having to send my data to the scale manufacturer app of choice (seemed to include telemetry, a user account requiring online access etc). 

Anything that can be done to improve scanning for Bluetooth devices would be awesome.

Allowing my phone to disconnect from the scales would likewise be great as right now I cannot use bluetooth headphones if my wife wants to weigh herself.

Tested on Redmi 5 & x5 pro. Android 6 & 7 (respectively) using Xenta Smart Fitness Scale

timoc, 2018-04-07:
Is it possible to permenantly pair a scale with a device?

Lewiscowles1986, 2018-04-07:
Certainly, my device all I now do is turn on Bluetooth and open OpenScale before getting on the scales so if people don't change devices often, it seems to be working for me once initially paired. The initial device pairing. Well android devices and even the iPhone 4s I've worked with seem to have terrible connectivity, not just to the smart-scales. It's a real area I wish we could scan a QR code before hand, or generate one from a pre-connected device to say this is the mac address or device ID I'd like to connect to.

erijo, 2018-04-07:
Please test the latest dev version: https://github.com/oliexdev/openScale/releases/tag/travis-dev-build

You should now be able to close down the Bluetooth connection. The pairing process may also be improved. Please report back your result.

Lewiscowles1986, 2018-04-08:
> @timoc a constant Bluetooth pairing is not necessary. openScale only searches for the MAC address which you have set in the openScale Bluetooth settings.

Right, this is about using the openScale bluetooth settings though.

There's no need for you to do anything, I merely stated it would be awesome if something could be done. The issue is for others to be aware of the process I carried out, that's all.

oliexdev, 2018-04-08:
> There's no need for you to do anything, I merely stated it would be awesome if something could be done. The issue is for others to be aware of the process I carried out, that's all.

To avoid your two first steps we need the manufacture Bluetooth device name, so we can add this in our "supported Bluetooth device name" list.

closed 2018-04-08 by oliexdev

Lewiscowles1986, 2018-04-08:
Thanks, the device name is `SWAN`, it's MAC is `04:B3:EC:50:9D:6E`

oliexdev, 2018-04-08:
@Lewiscowles1986 thanks did you change the device name to SWAN? Otherwise it is the same like the MGB scale.
Can you please post a photo of the scale for our scale gallery on the wiki?
And is [this](https://www.amazon.co.uk/Xenta-Smart-Fitness-Scales/dp/B077H234H2) your scale, right?

Lewiscowles1986, 2018-04-08:
Yeah it's my scale, 
[image]

erijo, 2018-04-15:
Can you test the dev version @Lewiscowles1986 and let us know how it works for you?