oliexdev/openScale #237: It doesn't work on Beurer BF710

zachrdlapetr, 2018-04-02:
I have tried connect application to BF710 but it doesn't work. I can see a transfer informations between application and the scale but next there is an information that it isn't possible to connect. Please, could you fix it? Thank you so much.

erijo, 2018-04-02:
Please see this: https://github.com/oliexdev/openScale/wiki/Frequently-Asked-Questions-(FAQ)#my-scale-xyz-is-not-supported-why

Especially this link: https://github.com/oliexdev/openScale/wiki/How-to-reverse-engineer-a-Bluetooth-4.x-scale

Some work was done in #151, but as we don't own that scale we need your help.

zachrdlapetr, 2018-04-02:
Erijo, thank you for the explanation. I have opened this issue because oliexdev wrote a comment on Google Play. He responded to the negative evaluation where a man wrote that his scale BF710 doesn't work with openScale. And oliexdev recommended him to input new issue on github.

erijo, 2018-04-03:
I've ordered a BF710 and we'll see if I manage to make it work once I get it.

zachrdlapetr, 2018-04-04:
Wow. Thank you so much. I appreciate it. I believe there will be a huge amount grateful people because the orginal app from Beurer is really crap.

weliem, 2018-04-04:
The BF710 is identical to the BF700. Just add it to the list of advertisement names....and it will probably work

erijo, 2018-04-04:
Thanks for the tips @weliem. I'll test that once I get the scale.

magaiverpr, 2018-04-05:
I have this scale. I could test for you. How can I do that?

erijo, 2018-04-05:
Just adding an entry to the list is not enough. Other changes are needed as well. But I have gotten it to work and will try to put something together for test.

closed 2018-04-05 by erijo

erijo, 2018-04-06:
Please try the dev version and let me know how it works for you: https://github.com/oliexdev/openScale/releases/tag/travis-dev-build

zachrdlapetr, 2018-04-06:
Yes, It does work!! Great job. Thank you. I appreciate it.

weliem, 2018-04-06:
@erijo , nice! So apparently it is then using the same protocol as the Sanitas SBF70. Didn't expect that...

So now you could remove the Sanitas handler code and use the BF700 handler code.....if you would add their advertisement names...Nice cleanup...

erijo, 2018-04-07:
Will do that, thanks for the hint @weliem.

oliexdev, 2018-04-08:
@erijo could you add a photo of the BF710 scale for our wiki gallery?

erijo, 2018-04-14:
I've done the merge now @weliem. Could you please try the [dev version](https://github.com/oliexdev/openScale/releases/tag/travis-dev-build) and see if it still works with SBF70?

commit ee5ea9dd (2018-04-05): Initial support for Beurer BF710

Fixes #237