oliexdev/openScale #238: App crash on start

ilazgo, 2018-04-04:
Build version: 1.7.1 
Build date: 1980-01-01 00:00:00 
Current date: 2018-04-04 08:25:06 
Device: Fairphone FP1 
 
Stack trace:  
java.lang.RuntimeException: Unable to start activity ComponentInfo{com.health.openscale/com.health.openscale.gui.MainActivity}: android.view.InflateException: Binary XML file line #0: Error inflating class ImageView
	at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:2338)
	at android.app.ActivityThread.handleLaunchActivity(ActivityThread.java:2390)
	at android.app.ActivityThread.access$800(ActivityThread.java:151)
	at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1321)
	at android.os.Handler.dispatchMessage(Handler.java:110)
	at android.os.Looper.loop(Looper.java:193)
	at android.app.ActivityThread.main(ActivityThread.java:5299)
	at java.lang.reflect.Method.invokeNative(Native Method)
	at java.lang.reflect.Method.invoke(Method.java:515)
	at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:825)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:641)
	at dalvik.system.NativeStart.main(Native Method)
Caused by: android.view.InflateException: Binary XML file line #0: Error inflating class ImageView
	at android.view.LayoutInflater.createViewFromTag(LayoutInflater.java:713)
	at android.view.LayoutInflater.rInflate(LayoutInflater.java:755)
	at android.view.LayoutInflater.rInflate(LayoutInflater.java:758)
	at android.view.LayoutInflater.rInflate(LayoutInflater.java:758)
	at android.view.LayoutInflater.rInflate(LayoutInflater.java:758)
	at android.view.LayoutInflater.rInflate(LayoutInflater.java:758)
	at android.view.LayoutInflater.inflate(LayoutInflater.java:492)
	at android.view.LayoutInflater.inflate(LayoutInflater.java:397)
	at com.health.openscale.gui.fragments.StatisticsFragment.onCreateView(StatisticsFragment.java:79)
	at android.support.v4.app.Fragment.performCreateView(Fragment.java:2261)
	at android.support.v4.app.FragmentManagerImpl.moveToState(FragmentManager.java:1419)
	at android.support.v4.app.FragmentManagerImpl.moveFragmentToExpectedState(FragmentManager.java:1750)
	at android.support.v4.app.FragmentManagerImpl.moveToState(FragmentManager.java:1819)
	at android.support.v4.app.FragmentManagerImpl.dispatchStateChange(FragmentManager.java:3227)
	at android.support.v4.app.FragmentManagerImpl.dispatchActivityCreated(FragmentManager.java:3181)
	at android.support.v4.app.FragmentController.dispatchActivityCreated(FragmentController.java:192)
	at android.support.v4.app.FragmentActivity.onStart(FragmentActivity.java:572)
	at android.support.v7.app.AppCompatActivity.onStart(AppCompatActivity.java:177)
	at android.app.Instrumentation.callActivityOnStart(Instrumentation.java:1174)
	at android.app.Activity.performStart(Activity.java:5274)
	at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:2311)
	... 11 more
Caused by: java.lang.NumberFormatException: Invalid int: "res/color/abc_primary_text_material_light.xml"
	at java.lang.Integer.invalidInt(Integer.java:137)
	at java.lang.Integer.parse(Integer.java:374)
	at java.lang.Integer.parseInt(Integer.java:365)
	at com.android.internal.util.XmlUtils.convertValueToInt(XmlUtils.java:122)
	at android.content.res.TypedArray.getInt(TypedArray.java:255)
	at android.widget.ImageView.<init>(ImageView.java:155)
	at android.support.v7.widget.AppCompatImageView.<init>(AppCompatImageView.java:71)
	at android.support.v7.widget.AppCompatImageView.<init>(AppCompatImageView.java:67)
	at android.support.v7.app.AppCompatViewInflater.createView(AppCompatViewInflater.java:106)
	at android.support.v7.app.AppCompatDelegateImplV9.createView(AppCompatDelegateImplV9.java:1016)
	at android.support.v7.app.AppCompatDelegateImplV9.onCreateView(AppCompatDelegateImplV9.java:1073)
	at android.view.LayoutInflater$FactoryMerger.onCreateView(LayoutInflater.java:172)
	at android.view.LayoutInflater.createViewFromTag(LayoutInflater.java:684)
	... 31 more

User actions:

erijo, 2018-04-04:
Thanks for your report. I think this crash was fixed in 23183ab18.

If you want, I would appreciate if you can test [the dev version](https://github.com/oliexdev/openScale/releases/tag/travis-dev-build) and report back if it works or not.

closed 2018-04-04 by erijo

erijo, 2018-04-10:
The dev version is a separate app and doesn't have access to the data in the release version (to not mess that up). But if you backup the data from the release version (in the settings) you can import it in the dev version.

Thanks for verifying the fix!

ilazgo, 2018-04-10:
Ok. I don't have a recent backup and I can't do it now (the app crash on startup). So I will wait to a new version with this issue fixed.