oliexdev/openScale #239: [feature request] Autosave when app closes

natrius, 2018-04-04:
Implement an option for an opt-in autosave when the app-closes. I set the backup to a folder that is synced automatically via nextcloud.
Now i have to do it manually, i would love an option like that - no clue how easy it is to implement.

erijo, 2018-04-04:
In #236 we discussed adding an option to automatically export the measurements when a new one is added. Would that work for you as well, i.e. are you ok with CSV format or do you really want the sqlite database?

natrius, 2018-04-05:
Thanks for the fast answer. I did not connect the other isse with that :D 

Actually i just want a kind of a backup in case somethings happenes to my cellphone. 
.csv would be perfectly fine, because its accessible through excel or libreoffice calc. It should be possible to import the file again (=restore)

Further comments in #236

closed 2018-04-05 by natrius

erijo, 2018-04-07:
If you have backup enabled for Android I think it automatically backs up the database. See under system settings -> system -> backup -> app data.

natrius, 2018-04-11:
Well, i found the backup dialogue, but its empty and i don't know how to add something there. Still, thanks for the help, will research a bitmore :D