oliexdev/openScale #240: App unable to locate exported data

fabianmani, 2018-04-08:
#### Expected behavior
I expected after exporting my details to the default folder created by the app, I should be able to import the file from the folder back to the app for further usage. 

#### Actual behavior
When I tried to import the exported file I get an error message that the file isn't found but when I check my file manager the file is there hence making me unable to import my exported file. 

#### How to reproduce
* Download and open the app.
* create a user and input your details.
* export to the default folder created by the app
* click on import. 

* Device: Tecno K7
* Operating system: Android 7.0
* App Version: 1.7.1

#### Recording Of The Bug
https://youtu.be/qrhZuOwJ52o

<br /><hr/><em>Posted on <a href="https://utopian.io/utopian-io/@fabianmani/app-unable-to-locate-exported-data">Utopian.io -  Rewarding Open Source Contributors</a></em><hr/>

erijo, 2018-04-09:
This is very strange. Could you try removing the backup directory first and retry the steps?

fabianmani, 2018-04-09:
I'll try that and let you know the result.

oliexdev, 2018-04-09:
strange but what makes me wonder that the write/read permission was not asked? 
Did you already grant these permission? Can you check if these permission are granted to openScale?

fabianmani, 2018-04-09:
I just checked and discovered the permissions were not granted and that's
because it was not asked from the start. I granted the permission and its
now working fine.

erijo, 2018-04-09:
It seems like a bug if it didn't ask for permission.

Can you try installing the [dev version](https://github.com/oliexdev/openScale/releases/tag/travis-dev-build) and see if it asks for permission when you try to backup?

oliexdev, 2018-04-13:
could be possible a problem with the Tecno’s HiOS 2.0 on his smartphone which is a custom made Android OS.

oliexdev, 2018-04-21:
I hope the problem doesn't happen anymore with the implemented file chooser dialog.

closed 2018-04-21 by oliexdev