oliexdev/openScale #241: Logo/Icon Proposal

jbeguna04, 2018-04-09:
Hi good day! I would like to ask for your permission to propose a new designs of logo in your application/project. This issue is not saying that your current logo is unsatisfactory, I would just like to propose new designs. Hoping for your positive response Thank You!

erijo, 2018-04-09:
I think this is something that @oliexdev should comment on, but here are my 2 cents.

Replacing the logo with something (radically) different is not something I think should be done without a lot of consideration, specially since I think the current one looks rather good. On the other hand it's hard to do this consideration without knowing what the alternatives are.

Do you have some more details on why you think a change would be good and/or a design that we can see?

If you would like to contribute something graphical (something we of course welcome) perhaps it would be easier to start with some in-app icon or ideas on how the look and feel could be improved?

jbeguna04, 2018-04-09:
Thanks ill be proposing it today

oliexdev, 2018-04-21:
@jbeguna04 did you change you mind?

jbeguna04, 2018-04-26:
No sir, i still making designs for your project. I want to make you satisfied as i can

jbeguna04, 2018-05-08:
[image]
Sorry for the wait
this is my proposal for the logo design. Im open to your suggestions
[image]

jbeguna04, 2018-05-10:
I will have some improvement on that. Please allow me to send the designs for a moment

jbeguna04, 2018-05-10:
[image]

[image]<br />This work is licensed under a <a rel="license" href="http://creativecommons.org/licenses/by/4.0/">Creative Commons Attribution 4.0 International License</a>.

oliexdev, 2018-05-10:
@jbeguna04 I think your first version looks better. Maybe if you rotate the needle a bit then it wouldn't look a bit better and not like a off button!?

Btw it reminds a bit of the [Arduino icon](https://www.arduino.cc/).

jbeguna04, 2018-05-10:
Thank you for your suggestion.

jbeguna04, 2018-05-11:
[image]
What do you think sir?

oliexdev, 2018-05-11:
I think the new version looks better. Thanks! :+1: 

Only two small remarks. The icon with the shadow (2nd row 1st icon on your first icon version) is missing and could you write openScale with a small "o"? That would be great :smiley:  

Would you mind to post the svg files for each icon? I will try out this new icon in the next version. We will see how the reaction of the users will be.

Do you have also an idea to have a nice [feature graphic ](https://developer.android.com/distribute/best-practices/launch/feature-graphic) (1024x500px banner) for the GooglePlay store? 
See [here](https://incipia.co/post/app-marketing/google-play-feature-graphic-ideas/) or [here](https://medium.com/@incipiagabe/google-play-feature-graphic-examples-43ac8a473a2a) for some examples.

My current feature graphic looks like the following:
[image]

jbeguna04, 2018-05-11:
(2nd row 1st icon on your first icon version) is missing. Im sorry sir, I cant truly understand this.
[image]

oliexdev, 2018-05-11:
The following version is missing:

[image]

jbeguna04, 2018-05-11:
Now i got it, here sir
[image]

oliexdev, 2018-05-11:
Thanks do you have a SVG version of the icons?

jbeguna04, 2018-05-11:
[New folder.zip](https://github.com/oliexdev/openScale/files/1995979/New.folder.zip)

jbeguna04, 2018-05-11:
sorry sir but i didnt mean to closed this. It was accident due to hanging up my conputer

jbeguna04, 2018-05-11:
can  i create a PR for my designs sir?

oliexdev, 2018-05-11:
sure

jbeguna04, 2018-05-11:
thank you sir. Here is the link
https://github.com/oliexdev/openScale/pull/263

oliexdev, 2018-05-11:
@jbeguna04 Thanks in the branch [icon_design](https://github.com/oliexdev/openScale/tree/icon_redesign) I added a slightly different icon.
It looks like the following:

[image]

I close the gap in the middle and lighten up the shadow a bit. The problem is that if the icon is really small, the details of the icon will really hard to see.

Currently, I am a bit unsure if the original icon fits better for small resolutions!? I need to think a bit more. :thinking: 

Anyway, thank your really much for your effort @jbeguna04 You did a great job! :1st_place_medal:

jbeguna04, 2018-05-12:
I can adjust the sizes sir.

erijo, 2018-05-12:
One potential issue I see with the new design is that to me it isn't directly obvious that the app has anything to do with a scale in the same way that it was with the old version. Is it possible to achieve that in some way?

jbeguna04, 2018-05-12:
Sir. That proposal doesnt mean that you have to change the old icon. As what i have in mind is, my proposal can be used by an open source project that i can extend my help through this. Its up to you sir if you want to use my proposed logo someday. Thank you for your time But the  Letters o and s stands for open sclae and the mini scale illustrated on letter o

jbeguna04, 2018-05-12:
The Project Owner by the way is performing a testing on the new icon design. Hope it can help the project

oliexdev, 2018-05-13:
I have archived your icon design into the [repository](https://github.com/oliexdev/openScale/blob/master/doc/raw_icons/launcher_design_proposal_%20jbeguna04.zip)
Thanks again for your effort and contribution @jbeguna04 :+1:

closed 2018-05-13 by oliexdev