oliexdev/openScale #245: Crash with new version

ercoleg, 2018-04-18:
Please find below the crash report

Build version: 1.7.2 
Build date: 1980-01-01 00:00:00 
Current date: 2018-04-18 07:51:29 
Device: HUAWEI EVA-L09 
 
Stack trace:  
java.lang.RuntimeException: Unable to destroy activity {com.health.openscale/com.health.openscale.gui.MainActivity}: java.lang.NullPointerException: Null reference used for synchronization (monitor-enter)
	at android.app.ActivityThread.performDestroyActivity(ActivityThread.java:4398)
	at android.app.ActivityThread.handleDestroyActivity(ActivityThread.java:4417)
	at android.app.ActivityThread.-wrap6(ActivityThread.java)
	at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1629)
	at android.os.Handler.dispatchMessage(Handler.java:105)
	at android.os.Looper.loop(Looper.java:156)
	at android.app.ActivityThread.main(ActivityThread.java:6523)
	at java.lang.reflect.Method.invoke(Native Method)
	at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:942)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:832)
Caused by: java.lang.NullPointerException: Null reference used for synchronization (monitor-enter)
	at com.health.openscale.core.bluetooth.BluetoothCommunication.handleRequests(BluetoothCommunication.java:384)
	at com.health.openscale.core.bluetooth.BluetoothCommunication.setBtMachineState(BluetoothCommunication.java:197)
	at com.health.openscale.core.bluetooth.BluetoothCommunication.disconnect(BluetoothCommunication.java:348)
	at com.health.openscale.core.OpenScale.disconnectFromBluetoothDevice(OpenScale.java:499)
	at com.health.openscale.gui.MainActivity.onDestroy(MainActivity.java:200)
	at android.app.Activity.performDestroy(Activity.java:7141)
	at android.app.Instrumentation.callActivityOnDestroy(Instrumentation.java:1158)
	at android.app.ActivityThread.performDestroyActivity(ActivityThread.java:4385)
	... 9 more

User actions: 
2018-04-18 07:50:59: MainActivity resumed
2018-04-18 07:51:16: MainActivity paused

erijo, 2018-04-18:
What did you do when it crashed?

closed 2018-04-18 by erijo

oliexdev, 2018-04-18:
Hope it is not a major bug otherwise we have to release another version during the ongoing 1.7.2 release.

erijo, 2018-04-18:
@ercoleg: does it crash on every start?

Can you please test the [dev version](https://github.com/oliexdev/openScale/releases/tag/travis-dev-build) and report if it works or not?

ercoleg, 2018-04-19:
@erijo I'm testing it tonight and I'll let you know.
the crash happens every time you exit the application during BT connection (not completed).
I think you catched the right root cause.

erijo, 2018-04-19:
Aborting the current roll out and preparing. 1.7.3 release seems like a good idea. But there are a few minor things I'd like to get done first.

erijo, 2018-04-19:
Thanks for verifying the fix @ercoleg!

@oliexdev: should we get a 1.7.3 release ready?

commit 0965f67a (2018-04-18): Don't do any cleanup if disconnected before fully connected

Hopefully this fixes #245.