oliexdev/openScale #248: add support for Accuway scale

en1389, 2018-04-26:
I am use an Accuway scale 
http://images.hktvmall.com/h5680001/0c4204ced13f013b5d9854684d7b7271e527ccb4/h5680001_sl0916006a_171123082124_01_1200.jpg

It will shows the following data in its app.
[image]

The openScale can connect with the scale with v1.7.2
but it lack of showing the visceral index
I would like the openScale to add the support for this.

I capture the bt hci snoop log file, and it is attached. 
[btsnoop_hci_0426.log](https://github.com/oliexdev/openScale/files/1949533/btsnoop_hci_0426.log)

I check it with wireshark and found the corresponding data
00:00:35:ab:02:ae:00:bd:02:29:00:1c:01:2c:08:06:29:00:ea:00
00:00:
35:		53	age
ab:		171	height (cm)
02:ae:	686	weight (68.6 kg)
00:bd:	189	Body Fat (18.9%)
02:29:	553 Body water (55.3%)
00:1c:       28  bone mass (2.8 kg)
01:2c:       300 muscle (30.0%)
08:		8	visceral
06:29:	1577	BMR
00:ea:	234		BMI	(23.4602)
00

This data agree with the captured screen above.
However, on the openScale the data shown are,
 
[image]
There are some slightly different in the BMI and body fat, but I am not too worry about that.
For the BMR the different is bigger (1501 vs 1578 Kcal)
The biggest different is in Bone mass (4.08 vs 2.8 kg)

Please, check if some data is mis-calculated.

erijo, 2018-04-26:
The calculation for bone mass can be seen here: https://github.com/oliexdev/openScale/blob/master/android_app/app/src/main/java/com/health/openscale/core/bluetooth/BluetoothExingtechY1.java

Maybe @oliexdev can shed some light on why the calculation is done as it is?

en1389, 2018-04-26:
From the java script, the bone parameter is calculated like this,

float bone = (float)(((weightBytes[10] & 0xFF) << 8) | (weightBytes[11] & 0xFF)) / weight * 10.0f; // kg

We can see that the bone data is divided by 'weight'
So the result is in % not in kg.

oliexdev, 2018-04-26:
@en1389 fixed it with commit https://github.com/oliexdev/openScale/commit/3e3ff403e61f1e0f12c7f615064b37a367bfae93
Please try out the last openScale dev version.

It seems I received some false specification of the scale. :/

Anyway the difference between BMR is not a big deal.

oliexdev, 2018-04-26:
@en1389 btw can you take a photo of your scale please for our wiki gallery?

closed 2018-04-26 by oliexdev

erijo, 2018-04-26:
@en1389: if you add a second user in openScale and select that user, can you still get readings from the scale then? I suspect not... Please enable debug log under settings -> about and attach that here if it doesn't work.

en1389, 2018-04-27:
picture of accuway scale

http://images.hktvmall.com/h5680001/128093/h5680001_sl0916006b_171011050140_01_1200.jpg

erijo, 2018-04-27:
@en1389: we need an image taken by you, not an official one (due to copyright reasons).

erijo, 2018-04-30:
User id starts on 1 in openScale so 2 is expected. I've removed the check that made it only work for the first user. Please verify that it works with the dev version after this fix.

en1389, 2018-05-28:
I tried out the new v1.8.1(28) with my Accuway scale.

Frist, I am happy that the Visceral fat is added.
But I found that when I stepped on the scale, the app always made 2 entries each time.
And the first entry has incorrect data, i.e. body fat, body water, muscle are all at 80%.
And the Visceral for both entries are 0.

Please check it.
[image]