oliexdev/openScale #249: graph axis

en1389, 2018-04-27:
One thing I like the openScale app is the graph.
It it interesting to see my body weight changing over time.
[image]

But if I add other measurements (e.g. body fat) to the graph, they became flat lines.
And that is no fun.
[image]

It is because the Y axis has to spread across all the data points. And weight and body fat are in different data range.
I suggest the graph support a secondary Y axis, So that, the primary and secondary Y axis can have different data range.
User can have option to choose which measurement data reference to which Y axis.
[image]

oliexdev, 2018-04-27:
you can simple hide and show diagram lines if you click on the circles in the graph page.
What do you do if you want to show your weight, your BMI and your waist in one graph?
Well, you could present a three dimensional graph but what if you want to show a 4th diagram line?
Then it will reach the limits. I think with the current option you are flexible enough.

en1389, 2018-04-30:
I know I can hide the lines in the chart in the second page.
I am interested in the overview chart in the home page.
I am not talking about making some multi-dimensional graph. 
I would like 2 Y axis to take care low number range data and high number range data.
e.g. body weight, lean body mass can be reference to the primary Y axis with range 50-70 kg
and body fat, water, muscle, bone % can be on the secondary Y axis with 0-60% range

oliexdev, 2018-04-30:
You can also disable the percent of some values and show it in the unit format. So it will complicated everything and hard to handle. If we add more and more options, openScale won't be easy to use anymore.

closed 2018-04-30 by oliexdev