oliexdev/openScale #250: java.lang.IllegalStateException in com.health.openscale.core.database.ScaleMeasurementDAO_Impl.getAll

erijo, 2018-04-27:
Seen on Google Play Console:
Huawei LISZT (hwliszt), 3072MB RAM, Android 5.1

java.lang.RuntimeException: 
  at android.app.ActivityThread.handleBindApplication (ActivityThread.java:4867)
  at de.robv.android.xposed.XposedBridge.invokeOriginalMethodNative (Native Method)
  at de.robv.android.xposed.XposedBridge.handleHookedMethod (XposedBridge.java:334)
  at android.app.ActivityThread.handleBindApplication (<Xposed>)
  at android.app.ActivityThread.access$1900 (ActivityThread.java:163)
  at android.app.ActivityThread$H.handleMessage (ActivityThread.java:1443)
  at android.os.Handler.dispatchMessage (Handler.java:102)
  at android.os.Looper.loop (Looper.java:135)
  at android.app.ActivityThread.main (ActivityThread.java:5601)
  at java.lang.reflect.Method.invoke (Native Method)
  at java.lang.reflect.Method.invoke (Method.java:372)
  at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run (ZygoteInit.java:964)
  at com.android.internal.os.ZygoteInit.main (ZygoteInit.java:759)
  at de.robv.android.xposed.XposedBridge.main (XposedBridge.java:102)

Caused by: java.lang.IllegalStateException: 
  at android.arch.persistence.room.RoomOpenHelper.onUpgrade (RoomOpenHelper.java:82)
  at android.arch.persistence.room.RoomOpenHelper.onDowngrade (RoomOpenHelper.java:94)
  at android.arch.persistence.db.framework.FrameworkSQLiteOpenHelper$OpenHelper.onDowngrade (FrameworkSQLiteOpenHelper.java:128)
  at android.database.sqlite.SQLiteOpenHelper.getDatabaseLocked (SQLiteOpenHelper.java:254)
  at android.database.sqlite.SQLiteOpenHelper.getWritableDatabase (SQLiteOpenHelper.java:163)
  at android.arch.persistence.db.framework.FrameworkSQLiteOpenHelper$OpenHelper.getWritableSupportDatabase (FrameworkSQLiteOpenHelper.java:93)
  at android.arch.persistence.db.framework.FrameworkSQLiteOpenHelper.getWritableDatabase (FrameworkSQLiteOpenHelper.java:54)
  at android.arch.persistence.room.RoomDatabase.query (RoomDatabase.java:193)
  at com.health.openscale.core.database.ScaleMeasurementDAO_Impl.getAll (ScaleMeasurementDAO_Impl.java:525)
  at com.health.openscale.core.OpenScale.updateScaleData (OpenScale.java:601)
  at com.health.openscale.core.OpenScale.<init> (OpenScale.java:103)
  at com.health.openscale.core.OpenScale.createInstance (OpenScale.java:111)
  at com.health.openscale.core.Application.onCreate (Application.java:34)
  at android.app.Instrumentation.callApplicationOnCreate (Instrumentation.java:1014)
  at android.app.ActivityThread.handleBindApplication (ActivityThread.java:4858)

erijo, 2018-04-28:
According to the source at [RoomOpenHelper.java:82](https://android.googlesource.com/platform/frameworks/support/+/master/room/runtime/src/main/java/android/arch/persistence/room/RoomOpenHelper.java#82) the illegal state eception would the thrown if a migration is missing. Don't really understand how that could happen.

oliexdev, 2018-04-30:
I don't know what the user has done. It was only one user who had this problem. I will close this issue until we have more information.

closed 2018-04-30 by oliexdev