oliexdev/openScale #252: Problems with Sanitas SBF70 since version 1.7.2

nazco, 2018-04-28:
After updating openScale on my Android 7.1.1 I can't receive data from my Sanitas SBF70 anymore. OpenScale informs me about a successful connection and then after a few seconds it tells me that it lost the connection. No data is received in this time. Maybe it is liked to the bluetooth optimizations mentioned in the release notes of 1.7.2. I also just updated to 1.7.3 but the problem persists.

erijo, 2018-04-28:
I take it version 1.7.1 worked well for you?

nazco, 2018-04-28:
Yes 1.7.1 was working well.

erijo, 2018-04-28:
Please try the [dev version](https://github.com/oliexdev/openScale/releases/tag/travis-dev-build) and let me know if it works better for you.

nazco, 2018-04-28:
Just tested the new dev version but sorry, didn't solve the problem.

erijo, 2018-04-28:
Make sure you use the latest dev version when you do this.

nazco, 2018-05-02:
Sorry, has to wait until friday.

erijo, 2018-05-02:
@MartinNowack, @jflesch, @weliem: do anyone of you have this scale and if so, do you have issues with 1.7.3? Can you provide a debug log?

ReidarHH, 2018-05-03:
Hi.
I have the same problem. Sbf70 not receiving any data. 
Connects with Bluetooth for 5 seconds and then disconnects. 
Running 1.7.3-dev.
Also tried with 1.7.3(26)

Keep up the good work 👍 

Debug log
2018-05-03 13:03:57.501 Debug AboutPreferences: Debug log enabled, openScale (dev) v1.7.3-dev_8706cd58 (1525296613), SDK 24, motorola Moto G (5)
2018-05-03 13:04:12.637 Debug OpenScale: Disconnecting from bluetooth device
2018-05-03 13:04:14.746 Debug OpenScale: Trying to connect to bluetooth device [C8:FD:19:9D:0B:68] (SANITAS SBF70)
2018-05-03 13:04:14.750 Info BluetoothCommunication: Connecting to [C8:FD:19:9D:0B:68] (driver: Sanitas SBF70/SilverCrest SBF75)
2018-05-03 13:04:14.756 Debug BluetoothCommunication: Starting LE scan
2018-05-03 13:04:17.022 Debug BluetoothCommunication$GattCallback: onConnectionStateChange: status=0, newState=2
2018-05-03 13:04:17.040 Debug MainActivity: Bluetooth connection successful established
2018-05-03 13:04:37.837 Debug BluetoothCommunication$GattCallback: onConnectionStateChange: status=8, newState=0
2018-05-03 13:04:37.843 Info BluetoothCommunication: Disconnecting
2018-05-03 13:04:37.849 Debug MainActivity: Bluetooth connection lost
2018-05-03 13:04:44.609 Debug OpenScale: Disconnecting from bluetooth device
2018-05-03 13:04:45.320 Debug OpenScale: Trying to connect to bluetooth device [C8:FD:19:9D:0B:68] (SANITAS SBF70)
2018-05-03 13:04:45.323 Info BluetoothCommunication: Connecting to [C8:FD:19:9D:0B:68] (driver: Sanitas SBF70/SilverCrest SBF75)
2018-05-03 13:04:45.329 Debug BluetoothCommunication: Starting LE scan
2018-05-03 13:04:45.736 Debug BluetoothCommunication$GattCallback: onConnectionStateChange: status=0, newState=2
2018-05-03 13:04:45.755 Debug MainActivity: Bluetooth connection successful established
2018-05-03 13:05:06.469 Debug BluetoothCommunication$GattCallback: onConnectionStateChange: status=8, newState=0
2018-05-03 13:05:06.472 Info BluetoothCommunication: Disconnecting
2018-05-03 13:05:06.485 Debug MainActivity: Bluetooth connection lost
2018-05-03 13:05:10.642 Debug OpenScale: Disconnecting from bluetooth device
2018-05-03 13:05:11.335 Debug OpenScale: Trying to connect to bluetooth device [C8:FD:19:9D:0B:68] (SANITAS SBF70)
2018-05-03 13:05:11.339 Info BluetoothCommunication: Connecting to [C8:FD:19:9D:0B:68] (driver: Sanitas SBF70/SilverCrest SBF75)
2018-05-03 13:05:11.345 Debug BluetoothCommunication: Starting LE scan
2018-05-03 13:05:15.331 Debug BluetoothCommunication$GattCallback: onConnectionStateChange: status=0, newState=2
2018-05-03 13:05:15.352 Debug MainActivity: Bluetooth connection successful established
2018-05-03 13:05:36.079 Debug BluetoothCommunication$GattCallback: onConnectionStateChange: status=8, newState=0
2018-05-03 13:05:36.083 Info BluetoothCommunication: Disconnecting
2018-05-03 13:05:36.097 Debug MainActivity: Bluetooth connection lost

erijo, 2018-05-03:
Please try the latest dev version and attach the log.

ReidarHH, 2018-05-03:
It should be the latest dev version, downloaded today.

erijo, 2018-05-03:
Yes, but I did a change after reading your log so I wanted you to test again with the new version.

erijo, 2018-05-03:
Can you try searching for the scale in Bluetooth settings (in openScale) with the debug log enabled?

Also, can you enable the Android bt snoop log and attach it as well?

ReidarHH, 2018-05-03:
I just tried with developer mode and hci debugger.
First time the openscale app crashed. 
With following fault report 
Build version: 1.7.3-dev_d9b8c625 
Current date: 2018-05-03 20:56:22 
Device: Motorola Moto G (5) 
 
Stack trace:  
java.lang.NullPointerException: Attempt to invoke virtual method 'void android.bluetooth.BluetoothGatt.close()' on a null object reference
	at com.health.openscale.core.bluetooth.BluetoothCommunication.disconnect(BluetoothCommunication.java:443)
	at com.health.openscale.core.OpenScale.disconnectFromBluetoothDevice(OpenScale.java:579)
	at com.health.openscale.gui.preferences.BluetoothPreferences$9.handleMessage(BluetoothPreferences.java:345)
	at android.os.Handler.dispatchMessage(Handler.java:102)
	at android.os.Looper.loop(Looper.java:154)
	at android.app.ActivityThread.main(ActivityThread.java:6123)
	at java.lang.reflect.Method.invoke(Native Method)
	at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:867)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:757)

User actions: 
2018-05-03 20:34:16: MainActivity resumed
2018-05-03 20:34:20: MainActivity paused
2018-05-03 20:34:20: SettingsActivity created
2018-05-03 20:34:20: SettingsActivity resumed
2018-05-03 20:34:21: SettingsActivity paused
2018-05-03 20:34:21: SettingsActivity created
2018-05-03 20:34:21: SettingsActivity resumed
2018-05-03 20:34:25: SettingsActivity paused
2018-05-03 20:34:30: SettingsActivity resumed
2018-05-03 20:35:04: SettingsActivity paused
2018-05-03 20:35:04: SettingsActivity resumed
2018-05-03 20:35:04: SettingsActivity destroyed
2018-05-03 20:35:07: SettingsActivity paused
2018-05-03 20:36:17: SettingsActivity resumed
2018-05-03 20:36:19: SettingsActivity paused
2018-05-03 20:36:19: SettingsActivity created
2018-05-03 20:36:19: SettingsActivity resumed
2018-05-03 20:36:22: SettingsActivity paused
2018-05-03 20:36:22: SettingsActivity resumed
2018-05-03 20:36:22: SettingsActivity destroyed
2018-05-03 20:36:23: SettingsActivity paused
2018-05-03 20:36:23: SettingsActivity created
2018-05-03 20:36:23: SettingsActivity resumed
2018-05-03 20:36:24: SettingsActivity paused
2018-05-03 20:36:27: SettingsActivity resumed
2018-05-03 20:36:28: SettingsActivity paused
2018-05-03 20:36:28: SettingsActivity resumed
2018-05-03 20:36:28: SettingsActivity destroyed
2018-05-03 20:36:30: SettingsActivity paused
2018-05-03 20:36:30: SettingsActivity created
2018-05-03 20:36:30: SettingsActivity resumed
2018-05-03 20:37:21: SettingsActivity paused
2018-05-03 20:37:21: SettingsActivity resumed
2018-05-03 20:37:21: SettingsActivity destroyed
2018-05-03 20:37:23: SettingsActivity paused
2018-05-03 20:37:23: MainActivity resumed
2018-05-03 20:37:23: SettingsActivity destroyed
2018-05-03 20:38:27: MainActivity paused
2018-05-03 20:38:27: SettingsActivity created
2018-05-03 20:38:27: SettingsActivity resumed
2018-05-03 20:38:28: SettingsActivity paused
2018-05-03 20:38:28: SettingsActivity created
2018-05-03 20:38:28: SettingsActivity resumed
2018-05-03 20:38:32: SettingsActivity paused
2018-05-03 20:55:17: SettingsActivity resumed
2018-05-03 20:55:20: SettingsActivity paused
2018-05-03 20:55:23: SettingsActivity resumed
2018-05-03 20:55:25: SettingsActivity paused
2018-05-03 20:55:25: SettingsActivity resumed
2018-05-03 20:55:26: SettingsActivity destroyed
2018-05-03 20:55:27: SettingsActivity paused
2018-05-03 20:55:27: SettingsActivity created
2018-05-03 20:55:27: SettingsActivity resumed

ReidarHH, 2018-05-03:
Here is the openscale debug log

2018-05-03 20:57:55.970 Debug AboutPreferences: Debug log enabled, openScale (dev) v1.7.3-dev_d9b8c625 (1525359575), SDK 24, motorola Moto G (5)
2018-05-03 20:58:02.225 Debug OpenScale: Disconnecting from bluetooth device
2018-05-03 20:58:02.228 Debug BluetoothPreferences: Start discovery
2018-05-03 20:58:21.544 Debug BluetoothPreferences: Start LE scan
2018-05-03 20:58:31.544 Debug BluetoothPreferences: Stop discovery
2018-05-03 20:58:31.550 Debug BluetoothPreferences: Stop LE scan
2018-05-03 20:58:40.352 Debug BluetoothPreferences: Start discovery
2018-05-03 20:58:43.199 Debug BluetoothPreferences: Found supported device SANITAS SBF70 [C8:FD:19:9D:0B:68] (driver: Sanitas SBF70/SilverCrest SBF75, type: 2)
2018-05-03 20:58:45.041 Debug BluetoothPreferences: Stop discovery
2018-05-03 20:58:45.098 Debug OpenScale: Trying to connect to bluetooth device [C8:FD:19:9D:0B:68] in debug mode
2018-05-03 20:58:45.101 Info BluetoothCommunication: Connecting to [C8:FD:19:9D:0B:68] (driver: Debug)
2018-05-03 20:58:45.104 Debug BluetoothCommunication: Starting LE scan
2018-05-03 20:58:45.527 Debug BluetoothCommunication$GattCallback: onConnectionStateChange: status=0, newState=2
2018-05-03 20:59:06.346 Debug BluetoothCommunication$GattCallback: onConnectionStateChange: status=8, newState=0
2018-05-03 20:59:06.349 Debug OpenScale: Disconnecting from bluetooth device
2018-05-03 20:59:06.362 Info BluetoothCommunication: Disconnecting
2018-05-03 20:59:06.362 Info BluetoothCommunication: Disconnecting (with cleanup)
2018-05-03 20:59:51.322 Debug OpenScale: Trying to connect to bluetooth device [C8:FD:19:9D:0B:68] (SANITAS SBF70)
2018-05-03 20:59:51.325 Info BluetoothCommunication: Connecting to [C8:FD:19:9D:0B:68] (driver: Sanitas SBF70/SilverCrest SBF75)
2018-05-03 20:59:51.331 Debug BluetoothCommunication: Starting LE scan
2018-05-03 20:59:51.827 Debug BluetoothCommunication$GattCallback: onConnectionStateChange: status=0, newState=2
2018-05-03 20:59:51.842 Debug MainActivity: Bluetooth connection successful established
2018-05-03 21:00:12.556 Debug BluetoothCommunication$GattCallback: onConnectionStateChange: status=8, newState=0
2018-05-03 21:00:12.567 Info BluetoothCommunication: Disconnecting
2018-05-03 21:00:12.570 Debug MainActivity: Bluetooth connection lost
2018-05-03 21:00:20.931 Debug AboutPreferences: Debug log disabled

ReidarHH, 2018-05-03:
I can't share the results from the hci debugger. The app crashes every time I press share.... Sorry but my skills are a bit limited... ☺

erijo, 2018-05-03:
Please try again with the latest version and post the debug log.

ReidarHH, 2018-05-04:
Tried it. No fix
Bluetooth symbol shows on the scale, but no data is transferred to the app.

2018-05-04 11:37:05.381 Debug AboutPreferences: Debug log enabled, openScale (dev) v1.7.3-dev_a3f64be6 (1525385180), SDK 24, motorola Moto G (5)
2018-05-04 11:37:19.333 Debug BluetoothPreferences: Start discovery
2018-05-04 11:37:19.362 Debug BluetoothPreferences: Discovery: android.bluetooth.adapter.action.DISCOVERY_STARTED
2018-05-04 11:37:20.227 Debug BluetoothPreferences: Discovery: android.bluetooth.device.action.FOUND
2018-05-04 11:37:21.547 Debug BluetoothPreferences: Discovery: android.bluetooth.device.action.FOUND
2018-05-04 11:37:28.251 Debug BluetoothPreferences: Discovery: android.bluetooth.device.action.FOUND
2018-05-04 11:37:28.280 Debug BluetoothPreferences: Found supported device SANITAS SBF70 [C8:FD:19:9D:0B:68] (driver: Sanitas SBF70/SilverCrest SBF75, type: 2)
2018-05-04 11:37:28.290 Debug BluetoothPreferences: Discovery: android.bluetooth.device.action.FOUND
2018-05-04 11:37:28.481 Debug BluetoothPreferences: Discovery: android.bluetooth.device.action.FOUND
2018-05-04 11:37:28.728 Debug BluetoothPreferences: Discovery: android.bluetooth.device.action.FOUND
2018-05-04 11:37:28.996 Debug BluetoothPreferences: Discovery: android.bluetooth.device.action.FOUND
2018-05-04 11:37:31.761 Debug BluetoothPreferences: Discovery: android.bluetooth.device.action.FOUND
2018-05-04 11:37:33.270 Debug BluetoothPreferences: Discovery: android.bluetooth.device.action.FOUND
2018-05-04 11:37:34.105 Debug BluetoothPreferences: Stop discovery
2018-05-04 11:37:34.158 Debug OpenScale: Trying to connect to bluetooth device [C8:FD:19:9D:0B:68] in debug mode
2018-05-04 11:37:34.160 Info BluetoothCommunication: Connecting to [C8:FD:19:9D:0B:68] (driver: Debug)
2018-05-04 11:37:34.166 Debug BluetoothCommunication: Starting LE scan
2018-05-04 11:38:03.676 Debug BluetoothCommunication$GattCallback: onConnectionStateChange: status=0, newState=2
2018-05-04 11:38:24.526 Debug BluetoothCommunication$GattCallback: onConnectionStateChange: status=8, newState=0
2018-05-04 11:38:24.529 Debug OpenScale: Disconnecting from bluetooth device
2018-05-04 11:38:24.529 Info BluetoothCommunication: Disconnecting
2018-05-04 11:39:19.114 Debug OpenScale: Trying to connect to bluetooth device [C8:FD:19:9D:0B:68] (SANITAS SBF70)
2018-05-04 11:39:19.118 Info BluetoothCommunication: Connecting to [C8:FD:19:9D:0B:68] (driver: Sanitas SBF70/SilverCrest SBF75)
2018-05-04 11:39:19.124 Debug BluetoothCommunication: Starting LE scan
2018-05-04 11:39:19.776 Debug BluetoothCommunication$GattCallback: onConnectionStateChange: status=0, newState=2
2018-05-04 11:39:19.799 Debug MainActivity: Bluetooth connection successful established
2018-05-04 11:39:40.601 Debug BluetoothCommunication$GattCallback: onConnectionStateChange: status=8, newState=0
2018-05-04 11:39:40.605 Info BluetoothCommunication: Disconnecting
2018-05-04 11:39:40.614 Debug MainActivity: Bluetooth connection lost
2018-05-04 11:39:54.404 Debug OpenScale: Disconnecting from bluetooth device
2018-05-04 11:40:04.809 Debug OpenScale: Trying to connect to bluetooth device [C8:FD:19:9D:0B:68] (SANITAS SBF70)
2018-05-04 11:40:04.813 Info BluetoothCommunication: Connecting to [C8:FD:19:9D:0B:68] (driver: Sanitas SBF70/SilverCrest SBF75)
2018-05-04 11:40:04.819 Debug BluetoothCommunication: Starting LE scan
2018-05-04 11:40:07.700 Debug BluetoothCommunication$GattCallback: onConnectionStateChange: status=0, newState=2
2018-05-04 11:40:07.720 Debug MainActivity: Bluetooth connection successful established
2018-05-04 11:40:28.498 Debug BluetoothCommunication$GattCallback: onConnectionStateChange: status=8, newState=0
2018-05-04 11:40:28.562 Info BluetoothCommunication: Disconnecting
2018-05-04 11:40:28.570 Debug MainActivity: Bluetooth connection lost
2018-05-04 11:40:40.443 Debug AboutPreferences: Debug log disabled

erijo, 2018-05-04:
It would be nice to have a BT log. Can you try again to capture the log as [described here](https://forums.lenovo.com/t5/Moto-G5-Moto-G5-Plus/Bluetooth-BLE-problem-on-moto-G5/m-p/3961770/highlight/true#M18512)?

erijo, 2018-05-04:
Can you provide the same type of log when you connect with the vendor app?

erijo, 2018-05-04:
You can also check if the log is stored in /sdcard/btsnoop_hci.log on your phone.

erijo, 2018-05-19:
It's anyone able to provide the requested log?

oliexdev, 2018-05-25:
Since no one is answering, the problem doesn't seem so big. I'm gonna close this until somebody has some new input.

erijo, 2018-05-26:
I added some more debug logs. Could you please try the dev version again @ReidarHH and attach the log afterwards?

ReidarHH, 2018-05-28:
First of all, I'm sorry for the late reply. I've been very busy. This last update did the trick. ☺ Now it is importing all previous measurements. Excellent work Erik. ☺ 👍

erijo, 2018-05-28:
@nazco: does it work for you as well?

erijo, 2018-05-28:
For the record: problem was most likely fixed by  3f11e55

ReidarHH, 2018-05-29:
For info, my phone is running android 7
And it works for me ☺️👍

erijo, 2018-05-29:
@nazco: please restart the device and see if it works then. If not, please attach the openScale debug log from trying to connect with the dev version.

MartinNowack, 2018-05-30:
Sorry, I currently do not have access to the scale. @nazco I think increasing the sleep value is not the right solution. It covers hidden timing issues. For example, if the phone is under load, the value might not be enough. On the other hand, without load, the thread unnecessarily sleeps too long.

erijo, 2018-05-31:
@MartinNowack: from what I gather from reading a lot of comments on stackoverflow and other places, BLE on Android seems to suck big time. I'm not particularly found of sleeps neither, but it seems to be necessary on some devices to get a stable connection. But I'm open for any other ideas you or anyone else may have.

(But I would still like to see the debug log from the dev version when you have time @nazco.)

nazco, 2018-06-02:
I think what @MartinNowack pointed out with the timings is right. I tried to collect the logs yesterday and this time the problem with the connection occurs again with just a slightly higher system load. Here is the debug log.
```
2018-06-02 12:48:46.028 Debug AboutPreferences: Debug log enabled, openScale (dev) v1.8.1-dev_e4981f22 (1527585733), SDK 26, motorola moto g(6)
2018-06-02 12:48:53.031 Debug OpenScale: Trying to connect to bluetooth device [C4:BE:84:06:AA:75] (SANITAS SBF70)
2018-06-02 12:48:53.037 Info BluetoothCommunication: Connecting to [C4:BE:84:06:AA:75] (driver: Sanitas SBF70/SilverCrest SBF75)
2018-06-02 12:48:53.050 Debug BluetoothCommunication: BT is enabled, state=12, scan mode=21, is not discovering
2018-06-02 12:48:53.063 Debug BluetoothCommunication: Starting LE scan
2018-06-02 12:48:58.927 Debug BluetoothCommunication$GattCallback: onConnectionStateChange: status=0, newState=2
2018-06-02 12:48:58.952 Debug MainActivity: Bluetooth connection successful established
2018-06-02 12:49:00.757 Debug BluetoothCommunication$GattCallback: onServicesDiscovered: status=0
2018-06-02 12:49:01.768 Debug BluetoothCommunication: INIT STATE: 0
2018-06-02 12:49:01.776 Debug BluetoothCommunication: Set notification on for 0000ffe1-0000-1000-8000-00805f9b34fb
2018-06-02 12:49:01.788 Debug BluetoothCommunication: Write descriptor 00002901-0000-1000-8000-00805f9b34fb: 01 00 
2018-06-02 12:49:06.886 Debug BluetoothCommunication: INIT STATE: 1
2018-06-02 12:49:06.903 Debug BluetoothCommunication: Write characteristic 0000ffe1-0000-1000-8000-00805f9b34fb: E6 01 
2018-06-02 12:49:06.950 Debug BluetoothCommunication$GattCallback: onConnectionStateChange: status=8, newState=0
2018-06-02 12:49:06.955 Info BluetoothCommunication: Disconnecting
2018-06-02 12:49:06.965 Debug MainActivity: Bluetooth connection lost
```

nazco, 2018-06-17:
I'll try it the next few days but I have to mention that the previous version worked or didn't worked in a very non deterministic way so I need some time to check if it is really solved.

erijo, 2018-06-28:
Have you had time to test this yet @nazco?

nazco, 2018-06-28:
Not yet sorry. I'll try to start with some tests this weekend.

nazco, 2018-07-10:
Couldn't see any problems with my android 8 device in the last days with the latest development version. But could't test is with other android versions.

nazco, 2018-07-12:
Well, sorry it seems that with the latest dev build there is a new problem introduced. Now I need to have the app open and running to get values. Older values are ignored and not synced. I'll try to add a log tomorrow.

oliexdev, 2018-08-18:
I will close this issue due inactivity.

closed 2018-08-18 by oliexdev

nazco, 2018-08-20:
Well the problem persists.

commit f4c825c3 (2018-04-28): Run BLE scan when connecting to scale

Improves connectivity on e.g. Sony Xperia Z5 compact (Android 7.1.1)
and hopefully it will also fix #252.