oliexdev/openScale #253: Improve first user experience

erijo, 2018-04-28:
It would be good ease the setup for new users. Here are some ideas I've been thinking on:
* Remove initial weight. It's confusing to have to enter it when you want to connect to a scale to get the weight.
* Ask the user if he/she wants to connect a Bluetooth scale after adding the user.
* Change "device type" in Bluetooth settings to something that indicates that it starts a device search.
* Make goal date/weight optional.

oliexdev, 2018-04-30:
> Remove initial weight. It's confusing to have to enter it when you want to connect to a scale to get the weight.

see issue #43 for the reason.

> Ask the user if he/she wants to connect a Bluetooth scale after adding the user.

Not all users have a Bluetooth scale. It is not hard to find the Bluetooth settings. Maybe if no device is selected and a user pressed the Bluetooth icon, we could show the Bluetooth settings menu...

> Change "device type" in Bluetooth settings to something that indicates that it starts a device search.

Why? It will indicate the searching process if you click on it.

> Make goal date/weight optional.

In my opinion the whole statistic page needs a rework. :/ Currently we don't really do some evaluation with the goal date and goal weight. But these two items needs to be set otherwise we can't draw the goal line or calculate the remaining days. If we reworked the statistic page, we could change it completely into something different!? You are welcome to post some ideas or sketch ups what we can do with the statistic page and how it should be looked like.

erijo, 2018-05-02:
I'll have to think some more about the initial weight problem. Thanks for the pointer to the reason it was added.

After adding the first user, we could perhaps show some help that indicates that the user should press + to add manual measurements or the Bluetooth button to connect a scale (assuming we do as you suggested which sounded like a good idea).

"Device type" says nothing about it starting a search for a scale (at least not to me). I changed it to "Your Bluetooth scale" which at least gives a hint about what it is for.

Agree that the statistics page needs some work. But that will have to wait for another rainy day :)

oliexdev, 2018-05-05:
> After adding the first user, we could perhaps show some help that indicates that the user should press + to add manual measurements or the Bluetooth button to connect a scale (assuming we do as you suggested which sounded like a good idea).

Only a few users will read that and it will only annoy the rest of the users. I don't think it is really complicated to understand the UI concept. Sure some exceptions will always exist that doesn't understand how it works but these users are not willing to read neither. 

What should improve is the measurement settings because it is not so clear that the measurements has a sub menu. Maybe we should move the enable/disable button into the sub menu and have also a sub menu only for the ordering like we have it before. Then it is more separated and more clear to understand. 

> "Device type" says nothing about it starting a search for a scale (at least not to me). I changed it to "Your Bluetooth scale" which at least gives a hint about what it is for.

But you are in the Bluetooth settings, so you know that the device type is a Bluetooth device. Anyway, at least I will remove the word "your" we could stick with "Bluetooth scale".

> Agree that the statistics page needs some work. But that will have to wait for another rainy day :)

Yeah no hurry. I would recommend that we will collect some ideas or draft some mock ups before we code.

erijo, 2018-05-08:
> What should improve is the measurement settings because it is not so clear that the measurements has a sub menu. Maybe we should move the enable/disable button into the sub menu and have also a sub [...]

After having spent a lot of time to get it working they way it does now I'm a bit reluctant to change it again ;) I've instead improved some strings to guide the user.

oliexdev, 2018-05-10:
> I've recently started wondering if it should be "Smart" instead of "Bluetooth"? Even if Bluetooth is the correct technical term, what term is mostly used by the scale manufactures themselves? I.e. what do the end user expect to see?

Well, mostly you don't have to select a scale because you have only one scale for one app. 
Smart Bluetooth refers to Bluetooth 4.0 but the word Bluetooth scale is clear. Let's stick with it.

> After having spent a lot of time to get it working they way it does now I'm a bit reluctant to change it again ;) I've instead improved some strings to guide the user.

I can understand that. ;) You did a great job.

closed 2018-05-19 by oliexdev